arkq/bluez-alsa #4: Alsa config

francoismartin, 2016-10-28:
Hi @Arkq 

To begin, thank you very very much, your help my life with my alsa problem and bluez5.

I created systemd (debian) daemon and It's work. When I do `aplay -D bluealsa:HCI=hci0,DEV=00:0C:8A:6B:8E:BD,PROFILE=a2dp applause.wav` I listen my song ( but the first second is croped, but is not the problem here ;) )

Now, i want to config bluealsa with my alsa config.

Without bluealsa, my default config is 

```
# Adapted via http://stackoverflow.com/a/14398926/6268583
#
# Enceinte
pcm.dmixed {
    type dmix
    ipc_key 1024
    ipc_key_add_uid 0
    ipc_perm 0666     # mixing for all users
    slave.pcm "hw:1,0"
}
# Micro
pcm.dsnooped {
    type dsnoop
    ipc_key 1025
    slave.pcm "hw:1,0"
}

pcm.duplex {
    type asym
    playback.pcm "dmixed"
    capture.pcm "dsnooped"
}

# Instruct ALSA to use pcm.duplex as the default device
pcm.!default {
    type plug
    slave.pcm "duplex"
}

# tell ALSA to use hw:0 to control the default device (alsamixer and so on)
ctl.!default {
    type hw
    card 0
}
```

I use dmixed and duplex, because sometimes I have 2 song started together, all explication is [here](http://stackoverflow.com/a/14398926/6268583)

Now in my `~/.asoundrc` when I copy/past our explication code

```
cat ~/.asoundrc
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "00:0C:8A:6B:8E:BD"
defaults.bluealsa.profile "a2dp"
```

and start `aplay applause.wav` I don't have a error msg and another, but I doesn't listen my song in my speaker that listened with `aplay -D bluealsa:HCI=hci0,DEV=00:0C:8A:6B:8E:BD,PROFILE=a2dp applause.wav` 

And When I would like to change my `pcm.dmixed` to replace `slave.pcm "hw:1,0"` by 
```
defaults.bluealsa {
    interface "hci0"
    device "00:0C:8A:6B:8E:BD"
    profile "a2dp"
}
```

it isn't work, I have 

```
pi@raspberrypi:~ $ aplay applause.wav
ALSA lib pcm_direct.c:1699:(snd1_pcm_direct_parse_open_conf) Unknown field defaults
aplay: main:722: audio open error: Invalid argument
````

And to finish, I don't  config amixer with your explication

```
pi@raspberrypi:~ $ amixer -D bluealsa '00:0C:8A:6B:8E:BD' sset 70%
amixer: Unknown command '00:0C:8A:6B:8E:BD'...
pi@raspberrypi:~ $ amixer -D bluealsa '<00:0C:8A:6B:8E:BD>' sset 70%
amixer: Unknown command '<00:0C:8A:6B:8E:BD>'...
pi@raspberrypi:~ $ amixer -D bluealsa 00:0C:8A:6B:8E:BD sset 70%
amixer: Unknown command '00:0C:8A:6B:8E:BD'...
pi@raspberrypi:~ $ amixer -D bluealsa "00:0C:8A:6B:8E:BD" sset 70%
amixer: Unknown command '00:0C:8A:6B:8E:BD'...
pi@raspberrypi:~ $ amixer -D bluealsa "DEV=00:0C:8A:6B:8E:BD" sset 70%
amixer: Unknown command 'DEV=00:0C:8A:6B:8E:BD'...
```

Thanks again

francoismartin, 2016-10-28:
It's work if I did that 

```
# Adapt   via http://stackoverflow.com/a/14398926/6268583
#
# Enceinte
pcm.btspeaker {
 type plug
  slave {
    pcm {
      type bluealsa
      interface hci0
      device 00:0C:8A:6B:8E:BD
      profile "a2dp"
    }
  }
  hint {
    show on
    description "BT Headset"
  }
}
# Micro
pcm.dsnooped {
    type dsnoop
    ipc_key 1025
    # A adapter en fonction de la config de reconnaissance
    # par Pi, il se peut que l'ordre ne soit pas le m  me
    slave.pcm "hw:1,0"
}

pcm.duplex {
    type asym
    playback.pcm "btspeaker"
    capture.pcm "dsnooped"
}

# Instruct ALSA to use pcm.duplex as the default device
pcm.!default {
    type plug
    slave.pcm "duplex"
}

# tell ALSA to use hw:0 to control the default device (alsamixer and so on)
ctl.!default {
    type hw
    # A adapter en fonction de la config de reconnaissance
    # par Pi, il se peut que l'ordre ne soit pas le m  me
    card 0
}
```

But I can't stream 2 song in same time because I lost dmix plugin : 

```
pi@raspberrypi:~ $ aplay applause.wav 
../../../src/asound/bluealsa-pcm.c:101: Getting transport for 00:0C:8A:6B:8E:BD profile 1
../../../src/asound/bluealsa-pcm.c:502: Setting constraints
Playing WAVE 'applause.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
../../../src/asound/bluealsa-pcm.c:337: Initializing HW
../../../src/asound/bluealsa-pcm.c:137: Requesting PCM open for 00:0C:8A:6B:8E:BD
../../../src/asound/bluealsa-pcm.c:343: Couldn't open PCM FIFO: Device or resource busy
aplay: set_params:1297: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (23219 23220)
PERIOD_SIZE: (1023 1024)
PERIOD_BYTES: (4092 4093)
PERIODS: 22
BUFFER_TIME: (510362 510363)
BUFFER_SIZE: 22507
BUFFER_BYTES: 90028
TICK_TIME: 0
../../../src/asound/bluealsa-pcm.c:327: Closing plugin
```

Arkq, 2016-10-29:
Hi,

First of all, I'm glad you've made it - I mean, you've figured out how to configure asoundrc. This ALSA configuration file is hard to grasp at first. I should have provided more examples regarding this, however, I'm not sure if it is the right time to do this - there is still a lot of work with BlueALSA itself.

> And to finish, I don't config amixer with your explication

I've fixed the example in the README. I've had put amixer command into the README file without testing it... my bad. However, the sole purpose of the example is to steer reader in the right direction - ALSA documentation :) If you are unfamiliar with the amixer tool, you might try to use alsamixer (ncurse based, so it is easier to navigate) as follows:

    $ alsamixer -D bluealsa

> But I can't stream 2 song in same time because I lost dmix plugin

Yeeeap... I know about this. However, right now I don't know how to proceed. I will try to figure something out (hopefully without implementing stream mixer by myself), but right now I have to finish my work with HSP profile. I'll be back to this issue in a week or two.

Cheers!

francoismartin, 2016-10-31:
Thanks @Arkq.

closed 2016-10-31 by francoismartin