arkq/bluez-alsa #5: cannot set volume - no controls listed by amixer

ldanicel, 2016-11-01:
Hi!
first of all, thanks a lot for this, i am a true pulseaudio hater and am trying hard to keep it that way :)
amixer -D bluealsa controls
doesn't return anything except:
===
bluealsa: ctl.c:411: New client accepted: 8
bluealsa: ctl.c:414: +-+-
bluealsa: ctl.c:414: +-+-
bluealsa: ctl.c:414: +-+-
bluealsa: ctl.c:385: Client closed connection: 8
bluealsa: ctl.c:414: +-+-
===
in bluealsa daemon
also the volume setting command is not working as it can't find the control:
[liviud@devbox ~]$ amixer -D bluealsa sset 'Bose SoundLink Mobile speaker II - A2DP' 50%
amixer: Unable to find simple control 'Bose SoundLink Mobile speaker II - A2DP',0

Arkq, 2016-11-01:
Hi,

First of all, which version of code you are running (e.g master snapshot)? Secondly, could you past the whole bluealsa daemon logs - up from the startup?

If you are running master snapshot, try something like this:

    $ bluealsa --disable-hfp --disable-hsp

then connect speaker, and try to get the list of controls with `amixer -D bluealsa`.

I'm worried, that it might be something with your speaker name being long. Also, which version of bluez and alsa-lib you are using.

ldanicel, 2016-11-01:
I've installed it from this https://aur.archlinux.org/packages/bluez-alsa-git/ which is latest git as far as i figure.
local/bluez 5.43-1
local/alsa-lib 1.1.2-1

I ran it with --disable-hfp --disable-hsp and on speaker connect i got:
`bluealsa: Transport not available: /org/bluez/hci0/dev_00_0C_8A_7D_B4_69/fd2
bluealsa: Transport not available: /org/bluez/hci0/dev_00_0C_8A_7D_B4_69/fd2`

Arkq, 2016-11-01:
OK, I will have to try to run bluez-alsa using the same configuration. There might be some changes in both bluez and alsa-lib... Currently, I'm developing on bluez-5.39 and alsa-lib-1.0.29. However, it's strange that someone would submit package to the Arch repo without testing :/

Anyway, I will let you know tomorrow.

Arkq, 2016-11-02:
Hi,

I've upgraded my bluez and alsa-lib to 5.43 and 1.1.2 respectively, and it seems, that everything works as before. There is no major change in any of these upgrades. However, I've checked the possibility of long device name (which was my first guess) and something is not right :)

In the current master snapshot there is a constraint of 32 bytes for device name (which minus NULL character, it is 31 characters for name). The length of "Bose SoundLink Mobile speaker II" is 32 characters, which is one character more. So, the question is, from where you've got this name? If this is the name, which appears in the Bluez manager (or something) then OK, however not quite so...

Current master snapshot certainly has a bug/issue regarding ALSA control name (if the Bluetooth device name is very long), but it relevant only if one is using more than one device with the same name. I will fix it in the future. Controlling PCM volume should be possible, though, if one's got name from the amixer itself. The name should be listed via the command `amixer -D bluealsa`. If it is not, the issue is somewhere else.

First of all, are you able to stream music (regardless of the controller)? Because, I've assumed, that you are, which might be wrong. Could you provide full logs from the bluealsa daemon? If should looks like this (1st block - initialization; 2nd block - connecting device; 3rd block - running `amixer -D bluealsa`):

    $ bluealsa --disable-hfp --disable-hsp 2>&1 |cut -d: -f2-
    ../../src/ctl.c:353: Starting controller loop
    ../../src/bluez.c:677: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
    ../../src/bluez.c:677: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
    ../../src/bluez.c:677: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
    ../../src/bluez.c:677: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
    ../../src/main.c:202: Starting main dispatching loop

    ../../src/bluez.c:482: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
    ../../src/bluez.c:482: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
    ../../src/bluez.c:416: A2DP Source (AAC) configured for device 1C:48:F9:9D:81:5C
    ../../src/bluez.c:418: Configuration: channels: 2, sampling: 48000
    ../../src/transport.c:399: State transition: 0 -> 0

    ../../src/ctl.c:411: New client accepted: 8
    ../../src/ctl.c:414: +-+-
    ../../src/ctl.c:414: +-+-
    ../../src/ctl.c:414: +-+-
    ../../src/ctl.c:414: +-+-
    ../../src/ctl.c:414: +-+-
    ../../src/ctl.c:385: Client closed connection: 8
    ../../src/ctl.c:414: +-+-

The output from the amixer should be something along these lines:

    Simple mixer control 'Jabra MOVE v2.3.0 - A2DP',0
      Capabilities: pvolume pswitch
      Playback channels: Front Left - Front Right
      Limits: Playback 0 - 127
      Mono:
      Front Left: Playback 127 [100%] [on]
      Front Right: Playback 127 [100%] [on]

But before running, make sure that there is no bluealsa daemon in the background, e.g. `pgrep -a bluealsa`. If something pops up, kill it! :)

If these logs will not be enough, and you have a courage and time for some more thorough debugging :D, I'll try to add debugs in few more places, and we'll see what's happening.

infirit, 2016-11-02:
@Arkq are you using the ``alias`` or the ``name`` field from the bluez device? The former can contain more than the 31 limit the later not.

Arkq, 2016-11-02:
I'm using the Name property from the D-Bus (device API), which I think is a "friendly name" and can be up to 248 characters - it is described as Bluetooth remote name. However, thanks for pointing out, that there is other name which can be set by user - I will definitely use that.

The limitation of 31 character will not save me, thought. The problem is with ALSA. The maximal name length of a control element is 44 characters. This name HAS to include Switch/Volume and Playback/Capture suffixes (plus spaces), which is 16 characters. So it's 28 characters for the name. Then, I do not know if there is a bug in the ALSA lib or no, but it looks like potential crash if name is not \0 terminated - for safety 27 chars. Then, I have to mark somehow, that the control is for A2DP or for HFP (which will be suffixes with SCO) - another 7 characters. So finally, there is a room for 20 characters for the actual name, which is less then 31 :D. And the name has to be unique!

I thought about using MAC addresses, but is seems to be ugly as hell. If you've got any idea how the control element should be presented to the uses, write without any hesitation.

commit 579f49fd (2016-12-03): Annotate element name with the device number

Controller plug-in ALSA library can not handle elements with the same
name. However, in our case, it is possible to connect Bluetooth devices
with the same names. This commit ensures, that the element name will be
unique.

Fixes #5
closed 2016-12-03 by Arkq