arkq/bluez-alsa #6: Transport not available during connection phase, but amixer shows device

aufziehvogel, 2016-11-05:
Hi,

I seem to have some trouble getting my bluetooth headset fully connected with ALSA. It shows an error "Transport not available" when I want to play a media file and on connection to the device. However, unlike issue #5 I am able to get the device shown in the output of `amixer -D bluealsa:

```Simple mixer control 'PLT_Legend - HSP',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 127 Capture 0 - 127
  Mono: Playback 82 [65%] [on] Capture 82 [65%] [on]```

Beforehand I already got an error in the connection phase in the output of `bluetooth` (Transport not available).

I execute commands in this order:

1. `bluealsa` as root
2. `bluetoothctl` as root: `power on` and `connect E4:22:A5:08:7B:FF` (headset then tells me, it connected to PC)

When the command `connect E4:22:A5:08:7B:FF` executes, I get the error `Transport not available: /org/bluez/hci0/dev_E4_22_A5_08_7B_FF/fd9` in the output of `bluealsa` (where the fd9 is an increasing number with each try).
When the command `connect E4:22:A5:08:7B:FF` executes, I get the error `Transport not available: /org/bluez/hci0/dev_E4_22_A5_08_7B_FF/fd9` in the output of `bluealsa` (where the fd9 is a number increasing with each trial).

```bluealsa: ctl.c:353: Starting controller loop
bluealsa: bluez.c:677: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: bluez.c:677: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: bluez.c:677: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: bluez.c:677: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: bluez.c:893: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: bluez.c:893: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: main.c:202: Starting main dispatching loop
bluealsa: bluez.c:846: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: bluez.c:778: HSP Audio Gateway configured for device E4:22:A5:08:7B:FF
bluealsa: transport.c:399: State transition: 0 -> 2
bluealsa: io.c:960: Starting IO loop: HSP Audio Gateway
bluealsa: transport.c:91: Created new IO thread: HSP Audio Gateway
bluealsa: io.c:994: AT command: +VGS=08
bluealsa: io.c:994: AT command: +VGM=15
bluealsa: Transport not available: /org/bluez/hci0/dev_E4_22_A5_08_7B_FF/fd9
bluealsa: Transport not available: /org/bluez/hci0/dev_E4_22_A5_08_7B_FF/fd9```

That's also the error I then get, when I want to play a media file either by passing the device information via command line or using it from the `asound.conf` (like `amixer` does):

```[root@pc folder]# aplay -D bluealsa:HCI=hci0,DEV=E4:22:A5:08:7B:FF,PROFILE=a2dp applause.wav 
bluealsa-pcm.c:101: Getting transport for E4:22:A5:08:7B:FF profile 1
ALSA lib bluealsa-pcm.c:619:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:786: audio open error: No such device
[root@pc folder]# aplay -D bluealsa applause.wav 
bluealsa-pcm.c:101: Getting transport for E4:22:A5:08:7B:FF profile 1
ALSA lib bluealsa-pcm.c:619:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:786: audio open error: No such device```

I am executing all commands as root, to make sure it's not permission issues.

How can we further investigate this? Am I missing some libraries for HSP? `Transport not available` sounds like it could be such an issue.

aufziehvogel, 2016-11-05:
Solution: Yes, pulseaudio and bluealsa cannot both try to access the bluetooth device at the same time (whatever that means in lower level, probably registering handlers to it on bluez). I removed the pulseaudio bluetooth connection (in my case on arch linux `pulseaudio-bluetooth`, restarted, and now it works.

closed 2016-11-05 by aufziehvogel

Arkq, 2016-11-05:
Hi,

This issue is definitely related to #5, but you've provided full logs, which is really helpful. It seems, that the A2DP profile is not connected (at least bluealsa think it is not connected). The crucial log (from my explanation in #5) is this:

    Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()

And it seems, that you are right, it might be the conflict with PulseAudio (or any other application, which has registered A2DP profile). To be honest, I never though about that. I will try to figure something out (if it is possible).

Many thanks for reporting it!

george-ai, 2017-09-14:
Hi

Quick question. Is HSP included now?

Arkq, 2017-09-14:
Yes, it is in the master branch. However, I've tested HSP audio-gateway only.