arkq/bluez-alsa #8: plug ALSA plugin cannot convert to bluealsa

paolobolzoni, 2016-11-22:
I wanted to automatically convert to the sound format expected by my headset so added this lines to my asoundrc:

```
pcm.btspeaker {
 type plug
  slave {
    pcm {
      type bluealsa
      interface "hci0"
      device "20:74:CF:04:D6:52"
      profile "a2dp"
    }
    format "unchanged"
    channels "unchanged"
    rate "unchanged"
  }
  hint {
    show on
    description "BT Headset"
  }
}
```
I did not add the three lines "format, channels, rate" in the beginning, but without them programs crashes or fail cleanly. Still, if plug ALSA plugin cannot convert it is not very useful.

Example, the first execution works. The second crashes. The third freezes.
Disconnecting and connecting or restarting bluetoothd does not help at all, bluealsa appears frozen and reacts only to a kill -9 and when I restart bluealsa I get "Couldn't initialize controller thread: Address already in use." The only solution I found is a full reboot.

```
~ % aplay -D pcm.btspeaker ./Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav
bluealsa-pcm.c:102: Getting transport for 20:74:CF:04:D6:52 type 1
bluealsa-pcm.c:516: Setting constraints
Playing WAVE './Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Warning: rate is not accurate (requested = 44100Hz, got = 48000Hz)
         please, try the plug plugin (-Dplug:pcm.btspeaker)
bluealsa-pcm.c:351: Initializing HW
bluealsa-pcm.c:139: Requesting PCM open for 20:74:CF:04:D6:52
bluealsa-pcm.c:156: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-20:74:CF:04:D6:52-1-0
bluealsa-pcm.c:368: FIFO buffer size: 4096
bluealsa-pcm.c:373: Selected HW buffer: 94208 ?= 23 periods x 4096 bytes
bluealsa-pcm.c:404: Prepared
bluealsa-pcm.c:310: Starting
bluealsa-pcm.c:241: Starting IO loop
bluealsa-pcm.c:325: Stopping
bluealsa-pcm.c:325: Stopping
bluealsa-pcm.c:380: Freeing HW
bluealsa-pcm.c:185: Closing PCM for 20:74:CF:04:D6:52
bluealsa-pcm.c:341: Closing plugin

~ % aplay -Dplug:pcm.btspeaker ./Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav
bluealsa-pcm.c:102: Getting transport for 20:74:CF:04:D6:52 type 1
bluealsa-pcm.c:516: Setting constraints
Playing WAVE './Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
bluealsa-pcm.c:351: Initializing HW
bluealsa-pcm.c:139: Requesting PCM open for 20:74:CF:04:D6:52
bluealsa-pcm.c:156: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-20:74:CF:04:D6:52-1-0
bluealsa-pcm.c:368: FIFO buffer size: 4096
bluealsa-pcm.c:373: Selected HW buffer: 94208 ?= 23 periods x 4096 bytes
bluealsa-pcm.c:404: Prepared
bluealsa-pcm.c:310: Starting
bluealsa-pcm.c:241: Starting IO loop
[1]    9830 segmentation fault (core dumped)  aplay -Dplug:pcm.btspeaker ./Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav
~ %
aplay -D pcm.btspeaker ./Korg-TR-Rack-Standard-Kit-Crash-Cymbal.wav
bluealsa-pcm.c:102: Getting transport for 20:74:CF:04:D6:52 type 1
^C
```

Still, thanks a lot for bluealsa. Even with its problems seems better than pulseaudio to me.

Arkq, 2016-11-22:
Hi,

Thx for reporting this issue. Bluealsa is still in a development stage. It works in the most common cases, but all edge cases might need some workarounds and/or fixing :)

> Disconnecting and connecting or restarting bluetoothd does not help at all, bluealsa appears frozen and reacts only to a kill -9 and when I restart bluealsa I get "Couldn't initialize controller thread: Address already in use." The only solution I found is a full reboot.

I'll try to fix this right away. However, in a meantime there is a workaround. When bluealsa server hangs and kill -9 is required, before restating it, the `/var/run/bluealsa` directory needs to be cleaned - named sockets and FIFO files are stored here, which are not properly cleaned upon kill -9. Just run:

    rm /var/run/bluealsa/*

Reboot works as well, because on most distros this directory (/var/run) is mounted as a tmpfs.

paolobolzoni, 2016-11-22:
Is there anything I can do? Backtrace or something?

Arkq, 2016-11-22:
No, thanks. I've never tested plug ALSA plugin before, so this bug did not pop out. It is 100% reproducible :D.

paolobolzoni, 2016-11-23:
Meanwhile I updated the archlinux wiki to help future PulseAudio-less users: https://wiki.archlinux.org/index.php/Bluetooth_headset#Headset_via_Bluez5.2Fbluez-alsa

Arkq, 2016-11-23:
I thought about adding some guidelines on the wiki page, but I'm not a good editor :) So, many thanks for that!

Anyway, I've fixed (I hope so) the issue with the limbo-state bluealsa server. There shouldn't be any leftovers in the /var/run any more. I've seen this issue by myself from time to time, however I was not able to pinpoint it without a reliable way to reproduce it.

The issue with the plug ALSA plugin I'll try to fix maybe tomorrow, my brain is off for today :D

Arkq, 2016-11-24:
I've added small fix to the current master snapshot 8a240c9, which should help a little bit with this issue. Now, I'm able to use plug plugin without any extra parameters and it seems to work. However, I'm not able to make it work with anything except `aplay`.

My .asoundrc setup:

```
pcm.jabra {
	type bluealsa
	device "1C:48:F9:9D:81:5C"
	profile "a2dp"
}
pcm.btspeaker {
	type plug
	slave {
	pcm "jabra"
	}
}
```

Please, let me know if current master works for you (with aplay). However, to fix this issue in a full extent, I might need some more time.

paolobolzoni, 2016-11-25:
Sorry for the slow answer, I forgot my headset at work.

Now I tested it and it apparently works fine. I tried: aplay, mplayer, and quakespasm. It worked fine for the first two, but in the third the sound was crackling. (Just guess what program I was trying when my boss entered the room, :D )

However, I think it's not fault of bluealsa, because it has always problems unless you "plug" it directly to the hadware (hw:0 or so). That it is not really possible with bluetooth, isn't it?

Since in many programs is difficult to setup what is the output sound card I use this little trick:
```
pcm.!default {
  type plug
  slave.pcm {
    @func getenv
    vars [ ALSAPCM ]
    default "pcm.smix"
  }
  hint {
    show on
    description "Default ALSA Output"
  }
}

pcm.trekzt {
  type bluealsa
  interface "hci0"
  device "20:74:CF:04:D6:52"
  profile "a2dp"
}
```
and simply start programs with `ALSAPCM=pcm.trekzt prgname`.

It works great, it is just unfortunate I cannot, or I do not know a way to, setup an env variable system-wide and change it with ease.

Arkq, 2016-11-27:
After LOTS of digging I've found out, that alsa-lib >= 1.1.2 does not play well with BlueALSA (in general with any software ALSA plugin). The problem is with thread-(non)safety ALSA patch, which is shipped with alsa-lib >= 1.1.2. So now (after disabling thread-safety, or downgrading to alsa-lib 1.0.29), I'm able to play audio via the plug plugin too (aplay, mpv, cmus, they all work great). Also, I've tested BlueALSA with opera (which is based on chrome), but I was not able to crush it. Could you tell me which version of chrome and alsa-lib you are using. Also, if youtube was played via the HTML5 functionality or via the flash player?

paolobolzoni, 2016-11-28:
alsa-lib 1.1.2
Chromium 54.0.2840.100
I am not sure about the youtube video format, but by default chromium
uses the Html5 player so I assume it's that.

Vitalic66, 2017-07-08:
Hallo,

i'm trying to setup a Pi3 with jessie lite and bluealsa. I got a Hifiberry Dac installed and want to use the Pi as bluetooth speaker for my smartphone.

So far i managed to install bluez 5.43 and also your bluealsa. I can connect the phone via BT. Also in terminal i can see blueals running and listings commands when music is´stopped or next song is played, but i can't her music. With speaker-test i can her the test sound. So soundcard is working.

`defaults.bluealsa.interface "hci0"
#defaults.bluealsa.device "00:1A:7D:DA:71:13"   addess of the usb bt dongle
#defaults.bluealsa.device "8D:C4:E9:B6:5C:80"    address of the smartphone
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

pcm.!default  {
 type hw card 0
}
ctl.!default {
 type hw card 0
}
`
I tried several settings. I can connect BT without problems, but to soundoutput is going anywhere.
aplay - l alsa says

`pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
aplay: device_list:277: control open (0): Invalid argument
`

Arkq, 2019-11-25:
Closing stale issue.

closed 2019-11-25 by Arkq