arkq/bluez-alsa #9: Aesthetic issue with volume levels

paolobolzoni, 2016-11-24:
In this case an example is worth thousands of words.

The following is copied from my shell. I setup the volume with the first command, but strangely it get setup in a weird way (right channel goes to 100%) as shown in the output of the second command.
The third and forth are similar try.

However, it seems just an aesthetic problem as the two channels actually have the same volume.

I also tried to play around with volumes and apparently what is counts is in fact the "Front Right," that strangely is the one that appear messed up.

I am not sure bluealsa is at fault at all, it is well possible that is the hardware being weird?
```
% amixer -D bluealsa sset 'TREKZ Titanium by Af - A2DP' 10%
Simple mixer control 'TREKZ Titanium by Af - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 13 [10%] [on]
  Front Right: Playback 13 [10%] [on]

% amixer -D bluealsa sget 'TREKZ Titanium by Af - A2DP'
Simple mixer control 'TREKZ Titanium by Af - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 13 [10%] [on]
  Front Right: Playback 127 [100%] [on]

% amixer -D bluealsa sset 'TREKZ Titanium by Af - A2DP' 20%
Simple mixer control 'TREKZ Titanium by Af - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 26 [20%] [on]
  Front Right: Playback 26 [20%] [on]

% amixer -D bluealsa sget 'TREKZ Titanium by Af - A2DP'
Simple mixer control 'TREKZ Titanium by Af - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 26 [20%] [on]
  Front Right: Playback 127 [100%] [on]
```

Arkq, 2016-11-24:
I'm afraid it is my fault... It is not implemented, not yet. Mixer has two separate channels (for volume change and mute), however bluealsa uses value only from one channel. I thought, that it is not a critical feature, so I have postponed it until core functionality is mature and tested (e.g. #8). It is on my todo list, though :)

paolobolzoni, 2016-11-25:
As implied by the "Aesthetic" word I am aware it is critical, or even important. However, the "issues" section of github are here for a reason :) . Definitely #8, #7 and the ability to use dmix/dsnoop are more important.

commit 09e90425 (2016-11-28): Scale audio channels separately

Fixes #9
closed 2016-11-28 by Arkq