arkq/bluez-alsa #10: AAC payload fragmentation.

Arkq, 2016-11-27:
If the size of the RTP packet exceeds writing MTU, the RTP payload should be fragmented. According to the RFC 3016, fragmentation of the audioMuxElement requires no extra header - the payload should be fragmented and spread across multiple RTP packets. However, [such an implementation](https://github.com/Arkq/bluez-alsa/blob/master/src/io.c#L977) seems not to work correctly.

Any help will be much appreciated.

pyntofmyld, 2018-02-06:
This was annoying me, as AAC sounds great streamed from a Pi Zero W, but the (rare) dropouts messed it up.

I think I kind of worked around it by setting the AAC bitrate field to 160kbps and enabling CBR mode. VBR doesn't make much sense when you've got a fixed bitrate channel, anyway - just use all of the bandwidth available.

Doesn't fix the fragmentation code of course, but makes AAC much more usable, based on an evening of testing with a single pair of headphones.

olliewalsh, 2018-02-21:
(Doubt it's the root cause but...) should [this](https://github.com/Arkq/bluez-alsa/blob/3e600b874d4424e01fe3d803a993f4982f8f33df/src/io.c#L886) be:
`rtp_header->markbit = len <= payload_len_max;`

Arkq, 2018-02-22:
@olliewalsh You're right, it's not a cause :) But yeahh... this line is buggy :D Nice catch! It should definitely read:
    `rtp_header->markbit = payload_len <= payload_len_max;`
or:
    `rtp_header->markbit = len == payload_len;`
whatever one desires :).

@pyntofmyld It is indeed possible to disable VBR mode (just let other device know, that [we are not supporting it](https://github.com/Arkq/bluez-alsa/blob/3e600b874d4424e01fe3d803a993f4982f8f33df/src/bluez-a2dp.c#L102)), because sink IO thread is not prepared for fragmented frames as well... However, right now I'd rather have buggy version of AAC, just for the sake of reminding me of this issue. The more I'm frustrated, the more I'm determined to find a way to fix it :)

commit 4a81de1d (2018-06-15): Fix AAC payload fragmentation

Fixes #10
closed 2018-06-15 by Arkq