arkq/bluez-alsa #11: Using A2DP Sink

chacal, 2016-11-29:
I'm trying to set up my computer as a bluetooth speaker. I compiled & installed bluez-alsa and managed to connect my phone to it. Last three lines of `bluealsa`'s output are:

    bluealsa: ../../src/transport.c:548: State transition: 1 -> 2
    bluealsa: ../../src/transport.c:94: Created new IO thread: A2DP Sink (SBC)
    bluealsa: ../../src/io.c:330: Starting IO loop: A2DP Sink (SBC)`

Also `hcitop` shows constant flow of data - everything looks good.

The problem is that no sound is played back from the headphones connected to the computer (regular audio output e.g. with `aplay` or `mpg123` works just fine).

What should I do to get the audio data from bluealsa played back via computer's sound card?

chacal, 2016-11-30:
Figured it out. I needed to do the following:
1) add ~/.asoundrc file with correct interface, device & profile (just like in the Readme)
2) when a2dp connection is established "arecord -f cd | aplay" can be used to play back the stream using the default audio output

closed 2016-11-30 by chacal

Arkq, 2016-11-30:
Yeap, BlueALSA tires to act as an extra audio card. There is no direct link between separate audio devices. One has to use extra program for redirecting audio signal.

For testing purposes I'm using [ecasound](https://en.wikipedia.org/wiki/Ecasound) sound processor, as follows:

    $ ecasound -i alsa,phone -o alsa