arkq/bluez-alsa #12: How to use this with Firefox?

onli, 2016-12-03:
I want to directly apologize, I'm aware that this issue might be offtopic.

BlueALSA on its own is working great on my system, but to be really useful I would need a way to let it play html5 audio over Firefox. I was unable to find a alsa-configuration or tool support to get the sound of Firefox to the bluealsa plugin. Is there a known or coming solution for FF, or is that something that would need support in the browser?

Arkq, 2016-12-03:
To be honest I don't know if it is possible to specify which ALSA device FF should use. For example, there is an option for chrome-based browsers (e.g. chromium, opera) which allows to specify ALSA device:

    chromium --alsa-output-device=my-device

If there is no way to set such an option (via command line, about:config page, or add-on), you can set bluealsa as a default ALSA device, as follows:

```
pcm.!default {
	type plug
	slave.pcm {
		type bluealsa
		device "XX:XX:XX:XX:XX:XX"
		profile "a2dp"
	}
}
```

However, I'm not recommending this - at least not now. Currently, bluealsa is not able to handle more than one audio client and the `dmix` or `dshare` plugins won't help, because they require hardware device (`hw`) as a backend.

Regardless of the above, I've just tried to play HTML5 video using FF and.... there is no sound. I'm using FF 45.5.0. Connection with the bluealsa server works fine (FF uses the default ALSA device), however it seems there is a bug in the bluealsa PCM plugin. I will have to check it. It looks like a deadlock related to the alsa-lib >= 1.1.2 (mentioned in the readme - troubleshooting section), but I'm not sure, since I'm using alsa-lib-1.0.29. You may try this approach (setting bluealsa as a default device) by yourself, but you may hit the same issue as I did.

infirit, 2016-12-03:
There is the multi plugin, see http://alsa.opensrc.org/TwoCardsAsOne for an example. This would copy all audio to all slaves. Would that work @Arkq?

onli, 2016-12-03:
thanks for the answer.

> You may try this approach (setting bluealsa as a default device) by yourself, but you may hit the same issue as I did.

I did that already :) Sadly it did also not work for me. I was not sure though whether that was because someone was blocking the device or whether it does not work at all. `dmix` was my next try, I found out the hard way it does not work with bluealsa…

Arkq, 2016-12-03:
@infirit, it seems that the multi plugin is for quite a different purpose. It combines many devices into one, nothing more.

The problem with connecting more than one client is different. In order to play multiple streams through single device, one has to mix audio signals - hence the dmix plugin name. It is not a new problem. Every hardware sound card has to overcome it. So, ALSA guys have created mixing plugins for sounds cards which lack mixing, but these plugins are tailored for hardware devices. So, right now I'm not sure how should I proceed. The easiest way, it is to implement mixing within bluealsa PCM plugin (based on dmix). However, this will bond bluealsa server and PCM plugin very tightly and I don't want that. The other approach is to mix signals within bluealsa server, but it will require slight refactoring. So, I've decided to postpone this feature until bluealsa-1.2.0 will be released. For this to happen I have to fix #8, #10 (hopefully) and this FF issue :)

infirit, 2016-12-03:
>it seems that the multi plugin is for quite a different purpose. It combines many devices into one, nothing more.

Well, you have two output devices and a new pcm output that send all audio to both the normal audio card and to bluealsa. Sounds (no pun intended) like a reasonable solution to this issue.

Here is another example of someone with 2 audio cards one for normal speaker and another for his headphones. The only difference being for this issue is that we are dealing with a bluetooth headphone.

http://www.6by9.net/output-to-multiple-audio-devices-with-alsa/

Arkq, 2016-12-03:
Hmm... maybe it is indeed worth a try. However, my guts tells me, that connecting more than one client will still require mixing... I will check it maybe tomorrow, or during the next week.

Arkq, 2016-12-10:
@onli, I've managed to partially overcome the issue with Firefox. Now, audio should start correctly. However, there is still a problem - it seems that the audio is going out of sync, and Firefox is trying to resync, which produces audio glitches (samples are lost).

I've pushed this commit already, because right now it is "possible" to watch vlogs. Unfortunately, watching everything else is still problematic (especially music videos...).

onli, 2016-12-14:
Thanks for working on that.

I tried to test it. Downloaded the source current git master, compiled, replaced my bluealsa-binary in /usr/bin/ with the one that got compiled (so there might have been an issue introduced by compiling manually this time, and not via the ebuild). Restarted bluealsa, reconnected the headset, restarted firefox after in the .asoundrc bluealsa was set as default – and it does not work for me. I get this error when trying to play something with aplay (that worked before swapping the binaries):

```
onli@Fallout:~/bluez-alsa/build$ aplay Downloads/a2002011001-e02.wav 
ALSA lib bluealsa-pcm.c:594:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: Invalid argument
aplay: main:786: audio open error: Invalid argument
```

Just to report back. Of course also possible that just the bluetooth system is not working properly right now and it would work after a restart, wouldn't be the first time. If not getting suggested otherwise, I'll test again when this ends up on my system via portage.

Arkq, 2016-12-14:
> replaced my bluealsa-binary in /usr/bin/  
OK, but bluealsa server is only a part of the whole bluez-alsa. You have to overwrite all files, otherwise you might end up with API incompatibilities.

And for the record, the change I've made for Firefox was in a ALSA PCM plug-in :)

Simple `make install` might suffice. However, if you don't like to run make install (personally, I'm avoiding this as much as possible), you might make symlinks for crucial bluez-alsa components (bluealsa server and ALSA plug-ins). I'm using such an approach for developing.

    $ ls -l /usr/bin/bluealsa
    lrwxrwxrwx 1 root root   87 Oct 21 14:04 bluealsa -> /home/[...]/bluez-alsa/build/src/bluealsa

    $ ls -l /usr/lib/alsa-lib/
    lrwxrwxrwx 1 root root   87 Oct 21 14:04 libasound_module_ctl_bluealsa.so -> /home/[...]/bluez-alsa/build/src/asound/.libs/libasound_module_ctl_bluealsa.so
    lrwxrwxrwx 1 root root   87 Oct 21 14:04 libasound_module_pcm_bluealsa.so -> /home/[...]/bluez-alsa/build/src/asound/.libs/libasound_module_pcm_bluealsa.so

And then, restart bluealsa service.

PS.
I've sent this comment via email, but it did not appear here, so, I'm adding it manually. Sorry for notifying you twice.

closed 2016-12-18 by Arkq

onli, 2016-12-18:
Thanks again for your work :)

Arkq, 2016-12-18:
No problemo :)
Please let me know if this fix works for you. For the record, I've tested it on FF 45.5.1.

onli, 2016-12-18:
Sure, least I can do.

So, I uninstalled bluez-alsa, pulled the master and compiled, this time with make clean, make, make install. Then starting bluealsa, setting the pcm.!default to it and connecting the headset. My .asoundrc looks like this:

```
pcm.!default {
  type bluealsa
  interface "hci0"
  device "00:1E:7C:2A:B1:3B"
  profile "a2dp"
}
```

It does not work for me. The audio element in FF, like before, gets stuck when starting pressing play. Youtube videos play, but without sound. This is the console output after FF starts:

```
bluealsa: ../../src/ctl.c:544: New client accepted: 10
bluealsa: ../../src/ctl.c:547: +-+-
bluealsa: ../../src/ctl.c:547: +-+-
bluealsa: ../../src/ctl.c:516: Client closed connection: 10
bluealsa: ../../src/ctl.c:547: +-+-
bluealsa: ../../src/ctl.c:544: New client accepted: 10
bluealsa: ../../src/ctl.c:547: +-+-
bluealsa: ../../src/ctl.c:547: +-+-
bluealsa: ../../src/ctl.c:516: Client closed connection: 10
bluealsa: ../../src/ctl.c:547: +-+-
```
aplay does work (I can't properly control its volume, but that's another story).

I'm on FF 50.1.0. If I can help debug, just tell me how, I'll take the time.

Arkq, 2016-12-18:
Damn...

Your .asoundrc is same as mine, so it is OK.

The output from the bluealsa seems to be OK as well. However, it is not very important (I think). The important stuff in in the output from the firefox itself. I've just downloaded FF 50.1.0, and everything works fine:

```
$ /tmp/firefox/firefox --safe-mode --private-window  # and then playing some youtube video
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 1C:48:F9:9D:81:5C type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 1C:48:F9:9D:81:5C type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:366: Initializing HW
../../../src/asound/bluealsa-pcm.c:140: Requesting PCM open for 1C:48:F9:9D:81:5C
../../../src/asound/bluealsa-pcm.c:157: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-1C:48:F9:9D:81:5C-1-0
../../../src/asound/bluealsa-pcm.c:389: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:394: Selected HW buffer: 17640 ?= 5 periods x 3528 bytes
../../../src/asound/bluealsa-pcm.c:425: Prepared
../../../src/asound/bluealsa-pcm.c:319: Starting
../../../src/asound/bluealsa-pcm.c:243: Starting IO loop
../../../src/asound/bluealsa-pcm.c:212: Requesting PCM pause for 1C:48:F9:9D:81:5C
```

Try to use `--safe-mode`, maybe one of your add-an is doing funky stuff. If running in a safe mode doesn't help I'll prepare brunch with extra debugging.

onli, 2016-12-18:
I'm getting this (also just playing a Youtube video):

```
onli@Fallout:~$ firefox --safe-mode --private-window
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 00:1E:7C:2A:B1:3B type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 00:1E:7C:2A:B1:3B type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 00:1E:7C:2A:B1:3B type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
```

Will now try with the other .asoundrc.

onli, 2016-12-18:
Great, we're getting there! With your .asoundrc youtube (https://www.youtube.com/watch?v=7mXg9Vkd3OE) works.

What does not work is the audio element. I'm using a software of mine as media player, https://github.com/onli/music-streamer. Clicking on play there still lets the audio element stay stuck, it is using a normal html5 audio element, and normally works. If you need a test-case, I could make you an account on my installation.

The strange thing is that http://hpr.dogphilosophy.net/test/ *does* work.

Arkq, 2016-12-18:
> What does not work is the audio element.

There is a limitation in a current bluealsa version. It is not possible to stream more than one autio at a time. If something is not working, check for the debug output like this (from the audio application, e.g. firefox):

    ../../../src/asound/bluealsa-pcm.c:371: Couldn't open PCM FIFO: Device or resource busy

> I remove the PCM switch to bluealsa and restart firefox

You've lost me there completely :) Does music-streamer works without `plug` plugin? What "PCM switch" you are talking about?

onli, 2016-12-18:
> There is a limitation in a current bluealsa version.  …

It should be only one audio element. This seems to be the related output in FF:

```
console.log: apply headers on URL : http://stream.paskuda.biz/track/832?supportMP3=maybe&supportOGG=maybe
console.log: cjs    : false
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 00:1E:7C:2A:B1:3B type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
../../../src/asound/bluealsa-pcm.c:103: Getting transport for 00:1E:7C:2A:B1:3B type 1
../../../src/asound/bluealsa-pcm.c:587: Setting constraints
../../../src/asound/bluealsa-pcm.c:356: Closing plugin
```

But I do see the "device busy" line when it got stuck at the audio element and then trying to play a youtube video at the same time. So I think this is another bug (I first I thought "yes, that will be it", but then remembered that I removed the code that generates multiple audio elements at the same time, it is only one single audio element).

But if you want, I can revisit this later when mixing works? Seems like a strange bug that defies logic, maybe doesn't need to have highest priority.

> What "PCM switch" you are talking about?

I worded that complicated. I meant emptying the .asoundrc, so FF uses the normal audio system again. Like, switching away from bluealsa.

tfontoura, 2018-03-14:
+1 for dmix

commit 7319c75c (2016-12-18): Check buffer availability while processing event

Fixes #12