arkq/bluez-alsa #13: bluealsa: Unsupported RTP payload type: 1

guillaumezin, 2016-12-11:
Hello,

I'm trying to use a raspberry as bluetooth speaker receiver. Currently, the sender is a Kubuntu 16.10 PC.

I launch on the RPi bluealsa then arecord -f cd -D bluealsa:HCI=hci0,DEV=00:21:4F:4E:08:46 toto.wav

I got this:

pi@Sejour ~/bluez-alsa $ sudo bluealsa 
bluealsa: ctl.c:484: Starting controller loop
bluealsa: bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: bluez.c:924: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: bluez.c:924: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: bluez.c:924: Registering profile: 0000111E-0000-1000-8000-00805F9B34FB: /Profile/HFPHandsFree
bluealsa: bluez.c:924: Registering profile: 0000111F-0000-1000-8000-00805F9B34FB: /Profile/HFPAudioGateway
bluealsa: main.c:212: Starting main dispatching loop
bluealsa: bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: bluez.c:427: A2DP Sink (SBC) configured for device 00:21:4F:4E:08:46
bluealsa: bluez.c:429: Configuration: channels: 2, sampling: 44100
bluealsa: transport.c:548: State transition: 0 -> 0
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 0 -> 1
bluealsa: transport.c:632: New transport: 8 (MTU: R:672 W:672)
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 1 -> 2
bluealsa: transport.c:94: Created new IO thread: A2DP Sink (SBC)
bluealsa: io.c:330: Starting IO loop: A2DP Sink (SBC)
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 2 -> 0
bluealsa: transport.c:658: Releasing transport: A2DP Sink (SBC)
bluealsa: transport.c:686: Closing BT: 8
bluealsa: io.c:62: Exiting IO thread
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 0 -> 1
bluealsa: transport.c:632: New transport: 10 (MTU: R:672 W:672)
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 1 -> 2
bluealsa: transport.c:94: Created new IO thread: A2DP Sink (SBC)
bluealsa: io.c:330: Starting IO loop: A2DP Sink (SBC)
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 2 -> 0
bluealsa: transport.c:658: Releasing transport: A2DP Sink (SBC)
bluealsa: transport.c:686: Closing BT: 10
bluealsa: io.c:62: Exiting IO thread
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 0 -> 1
bluealsa: transport.c:632: New transport: 10 (MTU: R:672 W:672)
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:548: State transition: 1 -> 2
bluealsa: transport.c:94: Created new IO thread: A2DP Sink (SBC)
bluealsa: io.c:330: Starting IO loop: A2DP Sink (SBC)
bluealsa: ctl.c:544: New client accepted: 8
bluealsa: ctl.c:547: +-+-
bluealsa: ctl.c:547: +-+-
bluealsa: ctl.c:326: PCM requested for 00:21:4F:4E:08:46 type 1 stream 1
bluealsa: ctl.c:547: +-+-
bluealsa: io.c:93: Opening FIFO for writing: /var/run/bluealsa/hci0-00:21:4F:4E:08:46-1-1
bluealsa: Unsupported RTP payload type: 1
bluealsa: Unsupported RTP payload type: 1
bluealsa: Unsupported RTP payload type: 1
bluealsa: Unsupported RTP payload type: 1
...

and toto.wav is empty (44 bytes).

Any idea?

Thank you

Arkq, 2016-12-11:
Hi,

What application you are using to stream BT audio from the Kubuntu -
pulseaudio, bluealsa, or something else? Because is seems that the RPT
packed is out of "standard" (if anything regarding Bluetooth A2DP audio
transfer can be called a standard). If you've got a smartphone with the BT
capabilities, try to use it and tell me if it works.

guillaumezin, 2016-12-11:
I stream using Pulseaudio (Amarok plays a MP3 and I displace its output to RPi seen as bluetooth speaker). I guess it is using A2DP audio.

I tried using my Android tablet, it seems that it worked a bit, then I wasn't able to reconnect streams, it seems related to the crap Android firmware of the chinese tablet this time...

My goal is to connect a videoprojector (I will open for Christmas...), I'm confident regarding the almost working test with the Android tablet.

Arkq, 2016-12-11:
Try to compile bluealsa from the [pulseaudio](https://github.com/Arkq/bluez-alsa/tree/pulseaudio) branch. I've removed payload type check, so now recording should work. I will not merge this commit into the master branch, though. This commit is a workaround for a "bug" in a PulseAudio.

guillaumezin, 2016-12-11:
Thank you, it works this way.

By the way, may I suggest you to indicate dependencies in you readme? It would be easier to compile (I had to compile on raspbian jessie bcutil, bctoolbox, mbedtls-2.4.0, ortp, from https://github.com/BelledonneCommunications, fdk-aac-0.1.4 from https://sourceforge.net/projects/opencore-amr/files/fdk-aac for --enable-aac option, and I needed libtool, autoconf,  libglib2.0-dev, libasound2-dev, libbluetooth-dev and libsbc-dev).

closed 2016-12-11 by guillaumezin

Arkq, 2016-12-13:
I'm glad, that it works for you. Also, I've added the dependency list to the readme file, following your suggestion. However, I've used original library names, because it seems that every distribution is trying to use its own naming convention...

G10DRAS, 2017-04-18:
Hello @guillaumezin just saw this post, you did a great job.
Is it possible for you to upload compiled bluez-alsa project for jessie somewhere?
It will save time and will help many others who need it on Jessie.
Thanks in advance.

guillaumezin, 2017-04-19:
Hello, look at https://github.com/guillaumezin/bluez-alsa/releases/tag/v1.3.0

G10DRAS, 2017-04-20:
@guillaumezin Thanks a lot !!

guillaumezin, 2017-04-22:
@G10DRAS I did as you mentionned, a package with checkinstall

https://github.com/guillaumezin/bluez-alsa/releases/tag/v1.2.0a

And I put my scripts as example to autoconnect my RPi as A2DP sink

https://github.com/guillaumezin/bluez-alsa/tree/master/scripts

G10DRAS, 2017-04-23:
@guillaumezin Thanks once again !!! You save my day. oRTP compilation on Jessie is a real mess .

fernando-rodriguez, 2017-10-14:
@Arkq I'm getting the same error but from an Android phone (old Samsung Galaxy Victory). Building with --disable-payloadcheck fixes the problem. So since it's not just a pulseaudio issue may I suggest that this will be turned into a command switch rather than a compile-time option? Perhaps it should be the default since most devices don't do it (this phone works with every other device I've tried) or at least include a hint in the error message. Thanks for the good work!

Arkq, 2017-10-22:
Hi @fernando-rodriguez

> Building with --disable-payloadcheck fixes the problem.

Could you please, paste here the payload type which is used by your Android version (from the warning message)? A2DP spec does not specify which payload type shall be used exactly or at least I couldn't find it. It only states, that the dynamic range shall be used, which is 96-127. In the case of PulseAudio the usage of payload type 1 is unforgivable :D (https://tools.ietf.org/html/rfc3551#section-6) If your phone uses the value from this range, I will have to loosen a little bit this payload type checking.

> So since it's not just a pulseaudio issue may I suggest that this will be turned into a command switch rather than a compile-time option? Perhaps it should be the default since most devices don't do [...]

I'll think about it :D.

> or at least include a hint in the error message.

Good point. I'll add some hint.

noamelf, 2018-11-18:
Hi, I'm having this issue as well :crying_cat_face: .
Just opened a ticket for PulseAudio: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/591

uqs, 2020-01-11:
Hey @Arkq I was wondering whether you could simply remove the check (3x) in src/io.c. I'm asking this as the current ChromeOS seems to be still using the older pulseaudio underneath and trying to stream anything from a Chromebook to bluez-alsa results in the RTP payload type 1 check firing.

It seems a bit strict to me to discard all those payloads < 96. There's type 8 for example which is PCM audio. Why would you want to discard that? Shouldn't bluez-alsa be able to deal with a PCM stream? What if my phone wants to pass through a GSM stream also?

Type 1 is silly, sure, but considering that pulseaudio messed that up and it is widely deployed in many (old) versions, many of which will never see an update, it seems straightforward to me. Postel's Law and all that.

Thanks for considering
uqs

commit c43441ad (2017-04-21): Add configure option for PulseAudio compatibility

PulseAudio uses incorrect (reserved) value for RTP payload type. So, our
payload type check prevents playing audio transfered from PulseAudio. In
order to allow playing audio from PulseAudio server, this constraint has
to be removed.

Fixes #13 and closes #44
commit fd6e91b3 (2018-07-23): Allow dynamic range payload type

Some devices set other value than 96 (e.g. Doogee X5 Pro).

Fixes #13