arkq/bluez-alsa #14: Default audio card

davidgfnet, 2016-12-13:
Hello!

Would you mind providing a bit more info on how making the device default system wide? I was playing around with /etc/asound.conf but setting a default bluetooth environment will lead to alsa trying to load libasound_module_pcm_bluetooth.so which obviously doens't exist in bluez 5.
I was wondering what exactly I should modify in my file in order to make it realize it's a bluealsa device:

> pcm.btheadset {
>    type plug
>    slave {
>        pcm {
>            type bluetooth
>            device B8:BB:AF:E4:D3:61
>            profile "a2dp"
>        }
>    }
>    hint {
>        show on
>        description "[Samsung] Soundbar J-Series"
>    }
> }
> ctl.btheadset {
>   type bluetooth
> }

Thanks for this great project! It's very interesting (I'm using it on the raspberry pi 3 BTW)

davidgfnet, 2016-12-13:
Ha! Nevermind, managed to find it myself:
Just replaced bluetooth with bluealsa and then for the sake of convenience btheadset to default (!default actually).
Now my whole system spits out sound to my soundbar. Only issue though is the buffering being quite big. On my senheiser speakers they will play at a higher sample rate (barely noticeable, like 5-10% oversampling) so that way they ca reduce the samples in the buffer and make sure the latency is low.
Any ideas on that?

closed 2016-12-13 by davidgfnet

Arkq, 2016-12-14:
To be honest I have no idea what might cause oversampling. The first thing which comes to my mind is sampling frequency incompatibility. However, I can see, that in your configuration you are using `plug` plugin, which should autocorrect this issue.

It is possible to detect whether this issue is caused by bluealsa, though. You have to save raw audio from [ALSA PCM plugin](https://github.com/Arkq/bluez-alsa/blob/master/src/asound/bluealsa-pcm.c#L284) (before it is sent to the bluealsa) and/or from the bluealsa [IO thread function](https://github.com/Arkq/bluez-alsa/blob/master/src/io.c#L522) (assuming that SBC was selected as an audio codec, otherwise [here](https://github.com/Arkq/bluez-alsa/blob/master/src/io.c#L945)).

After these lines, add something like:

    write(f, head, len /* or "samples * 2" */); // one sample is 16 bit

and before `while` or `for` loop add:
   int f = open("/tmp/xxx.raw", O_WRONLY|O_CREAT, 0644);

This should save all incoming PCM data. Afterward, you can play these files (providing correct sampling and number of channels).

If oversampling occurs in both cases, then it is a proof, that something is wrong with plug plugin. Oversampling in the data saved from the "bluealsa" server - definitely an issue with bluealsa :)

davidgfnet, 2016-12-15:
Hey! I think I didn't express myself properly before. I'm saying that some
devices, like my senheiser speakers, in order to cope with buffering and
reduce delay, might actually consume audio at a higher speed (essentially
by increasing the sampling playback freq) until the FIFO is close to empty.
But again, that's something device specific. I will give a try to the
senheiser headphones with bluealsa, it might be that the delay disappears
thanks to that.

Also, isn't there any way that we can add the device into the ALSA
registered devices? Some apps will actually pull that list in order to let
you choose which playback device you want and therefore breaking the audio
output (in my case Kodi). I'm pretty sure I can hack the apps, but it would
be much nicer to have a clean solution.

Anyway, great work man! Thanks! Do you think it might get upstreamed at
some point (and maybe maintained by someone from the alsa team?)