arkq/bluez-alsa #16: Crash when starting Saints Row 4

onli, 2016-12-19:
That might be a bit specific, but if it happens in that game it will happen with others as well. When I start this game it crashes. The related output in the console:

```
ALSA lib dlmisc.c:272:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/i386-linux-gnu/alsa-lib/libasound_module_pcm_bluealsa.so
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such device or address
*** BREAKPAD CRASH ***
```

Firefox did work just before, but the youtube tab was of course closed while trying the game. I have the .asoundrc from the FF-issue, plus the defaults.bluealsa.… lines from the readme to be able to control the volume via alsamixer -D bluealsa.

Arkq, 2016-12-19:
> ALSA lib dlmisc.c:272:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/i386-linux-gnu/alsa-lib/libasound_module_pcm_bluealsa.so

This line is interesting. It seems, that alsa library is not able to load plugin... strange. It is possible to debug it further, but you will have to compile alsa-lib by yourself (with small modification). Compilation is pretty straightforward (I've done it by myself) - ./configure and make (without make install).

In the src/dlmisc.c source file you will have to add "debug" line after the `dlopen()` call in the snd_dlopen function (current master line 83 and 87) as follows:

```
diff --git a/src/dlmisc.c b/src/dlmisc.c
index b36c48f..f14afb9 100644
--- a/src/dlmisc.c
+++ b/src/dlmisc.c
@@ -81,10 +81,12 @@ void *snd_dlopen(const char *name, int mode)
                strcat(filename, "/");
                strcat(filename, name);
                handle = dlopen(filename, mode);
+               fprintf(stderr, "ERR: %s\n", dlerror());
                free(filename);
        }
        if (!handle)
                handle = dlopen(name, mode);
+       fprintf(stderr, "ERR: %s\n", dlerror());
        return handle;
 #else
        return NULL;
```

Then run application as follows:

    $ export LD_LIBRARY_PATH=/[... replace with absolute path ...]/alsa-lib/src/.libs/
    $ application-to-run

The cause of the error should pop out.

onli, 2016-12-20:
> Are you using a 32 bit Linux? Or the game you're trying to run is 32 bit and the system is 64 bit?

Yes, possibly. I'm using 64 bit linux, but have multi-arch libs activated. To be able to start Steam I have to start it like this:

```
LIBGL_DRIVERS_PATH="/usr/lib32/dri" LIBGL_DEBUG="verbose" /usr/bin/steam
```

Then for some games I have to reset that path to /usr/lib/dri. Saints Row 4 is not one of them, it should be a 32 bit application.

If I should still compile stuff to help here you tell me, but I would need a pointer what exactly, like from where to get the alsa-lib I'd need. https://www.alsa-project.org/main/index.php/Download (and after tomorrow I'll be offline for a few weeks).

Arkq, 2016-12-20:
I think, there is no need to compile alsa-lib. This case is clear. You are running a multi-arch Linux and bluez-alsa is not ready for that, not yet.

infirit, 2016-12-20:
>bluez-alsa is not ready for that, not yet

I would not say that yet. To run 32bit application on a 64bit native operating system you will need 32bit version of alsa-lib, alsa-plugins *and* bluealsa. I doubt the op has build and installed a 32bit version of bluez-alsa. If this is the case I would consider this a packaging problem not a bluealsa one.

Arkq, 2016-12-20:
It is packaging problem. However, I think (I'm not sure, though), that bluez-alsa build configuration (configure.ac, Makefile.am, etc.) will need to be modified accordingly in order to allow multiarch builds. Bluealsa server does not need to be 32 bit, it is run globally as a system service.

onli, 2016-12-20:
> I doubt the op has build and installed a 32bit version of bluez-alsa

In fact I did not. For #12 I compiled bluealsa manually. You think it is possible that when I install it over https://packages.gentoo.org/packages/media-sound/bluez-alsa, the proepr libs will be built and it would work? I could try to install the 9999-git version (but won't have the time this year anymore).

Arkq, 2016-12-20:
> I could try to install the 9999-git version

It won't help either. You can use this approach as a convenient method for installing latest master snapshot (however, for debugging it is easier to install things manually from the source). This will install (by default) 64bit version. If you want to test this (if it is really an issue), you can crosscompile bluez-alsa as a 32bit and then copy (do not issue `make install`) ALSA plugins to the /usr/lib/i386-linux-gnu/alsa-lib directory. If you've never done this before, it might not be straightforward....

infirit, 2016-12-20:
I am co-author of that ebuild. I'll add multilib suppot to in and post it here. But probably not until next year.

infirit, 2016-12-20:
Got it done just now, try https://gist.github.com/infirit/c115f5ae66da691f490c76345b636e0a

file list
```
# equery f bluez-alsa
 * Searching for bluez-alsa ...
 * Contents of media-sound/bluez-alsa-9999:
/etc
/etc/init.d
/etc/init.d/bluealsa
/usr
/usr/bin
/usr/bin/bluealsa
/usr/lib32
/usr/lib32/alsa-lib
/usr/lib32/alsa-lib/libasound_module_ctl_bluealsa.so
/usr/lib32/alsa-lib/libasound_module_pcm_bluealsa.so
/usr/lib64
/usr/lib64/alsa-lib
/usr/lib64/alsa-lib/libasound_module_ctl_bluealsa.so
/usr/lib64/alsa-lib/libasound_module_pcm_bluealsa.so
/usr/share
/usr/share/alsa
/usr/share/alsa/alsa.conf.d
/usr/share/alsa/alsa.conf.d/20-bluealsa.conf
/usr/share/doc
/usr/share/doc/bluez-alsa-9999
/usr/share/doc/bluez-alsa-9999/README.md.bz2
```

onli, 2016-12-20:
That works! Thanks!

I still had to symlink /usr/lib/i386-linux-gnu to /usr/lib32/ to make it not crash, but that might be really something application specific, assuming a debian environment or something. Only tested till the menu so far, but seems good. Hey, and it's still this year :)

infirit, 2016-12-20:
Good, the ebuild has been updated in the portage tree as well, https://gitweb.gentoo.org/repo/gentoo.git/commit/?id=81434d56ebbd17a4e38fedbaade9e883d1f6d650

Arkq, 2016-12-20:
Wow, so the case is closed? It seems, that I've misjudged the power of gentoo ebuilds :)

onli, 2016-12-21:
I'm not sure whether you want to use that issue to coordinate getting that modified ebuild also for the normal, non-9999 ebuild? Otherwise, for me it is fixed.

infirit, 2016-12-21:
>that I've misjudged the power of gentoo ebuilds

A lot of time was spend on making his work and now its easy. Gentoo is a dream for developers and power users :-)

Arkq, 2016-12-21:
> whether you want to use that issue to coordinate getting that modified ebuild also for the normal, non-9999 ebuild

This issue was a packaging problem. So, I'm marking it as resolved.

@infirit thanks a lot for your assistance!

closed 2016-12-21 by Arkq