arkq/bluez-alsa #18: configure: WARNING: unrecognized options: --disable-thread-safety

raftopoulos, 2016-12-21:
$ ../configure --disable-thread-safety

results in

configure: WARNING: unrecognized options: --disable-thread-safety
configure: WARNING:               *** alsa-lib >= 1.1.2 ***
configure: WARNING: Starting from alsa-lib 1.1.2, it is possible to enable
configure: WARNING: thread-safe API functions. Unfortunately, this feature
configure: WARNING: is not mature enough - software plugins may experience
configure: WARNING: random deadlocks (bluez-alsa PCM plugin is affected).
configure: WARNING: Either compile alsa-lib without this feature or export
configure: WARNING: LIBASOUND_THREAD_SAFE=0 while using bluealsa PCM.
 
Debian Stretch x64, BlueZ 5.43, ALSA 1.1.2

Arkq, 2016-12-21:
Now I see, that this troubleshooting section is misleading. The --disable-thread-safety is a configuration option for alsa-lib >= 1.1.2, not for bluez-alsa. If you are using alsa-lib from a distro repository, the easiest approach is to define LIBASOUND_THREAD_SAFE=0 in you .profile file as follows:

    export LIBASOUND_THREAD_SAFE=0

However, before this, make sure, that you are affected by this deadlock issue. For example, mpv (mplayer) is not affected. So, it depends on what kind of software you are using.

raftopoulos, 2016-12-21:
Thanks for clarification.

commit eb380871 (2016-12-21): Rewording and clarifications

Fixes #18
closed 2016-12-21 by raftopoulos