arkq/bluez-alsa #22: No such device

Vitaliy69, 2016-12-27:
I disabled the default audio device via the BIOS and fully deleted _pulseaudio_ from the system, compile and install from the latest sources _bluez-alsa_ with _alsa-lib_ 1.1.2.

Then start _bluealsa_ and _bluetoothctl_ 5.40:

```
power on
scan on
agent on
default-agent
pair 00:19:7F:98:XX:XX
connect 00:19:7F:98:XX:XX
[CHG] Device 00:19:7F:98:XX:XX Connected: yes
Connection successful
[9xxPlantronics]#
```
But can not set _bluealsa_ as a default audio device. For this, created the file _/etc/alsa.conf_ as below:

```
pcm.!default {
   type plug
   slave {
       pcm {
           type bluealsa
           device 00:19:7F:98:XX:XX
           profile "a2dp"
       }
   }
   hint {
       show on
       description "BT Headset"
   }
}
ctl.!default {
  type bluealsa
}
```

And the _~/.asoundrc_ as below:

```
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "00:19:7F:98:XX:XX"
defaults.bluealsa.profile "a2dp"
```

Then restart _bluealsa_.

```
# cat /proc/asound/cards 
--- no soundcards ---
```

The command _# aplay -D bluealsa:HCI=hci0,DEV=00:19:7F:98:XX:XX,PROFILE=a2dp test.wav_ is finished with error:

```
ALSA lib ../../../git/src/asound/bluealsa-pcm.c:707:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:786: audio open error: No such device
```
Output from _bluealsa_:
```
bluealsa: Unsupported AT command: +BRSF=24
bluealsa: Unsupported AT command: +CIND=?
```

What can I do wrong?

If use _pulseaudio_ instead of _bluealsa_, BT headset works fine.

erikly01, 2016-12-28:
When you say "default audio device", are you attempting to black audio from the bluetooth source -> bluez -> bluez-alsa -> sound card?

Vitaliy69, 2016-12-28:
No, I'm attempting to play wav file via BT headset with help of bluez-alsa.

erikly01, 2016-12-28:
I've played around with the bluez-alsa quite a bit, and was able to get it to work.  I didn't write the code, 'arkq' did so he'd know better than me.  I do see a few issues with your current setup.

1. I recommend using only /etc/asoundrc or ~/.asoundrc for alsa configuration.
2. alsa 1.1.2 has a deadlock bug in it so if you can downgrade to 1.1.1 you'll know it isn't failing because of that issue.
3. I don't think your alsa.conf file is necessary because bluez-alsa creates the 20-bluealsa.conf file places it in the /usr/share/alsa/alsa.conf.d/ directory when you use the command 'sudo make install'

4. If you decide to keep 'your' alsa.conf then there are a few things that need to be corrected in it.
  a. Add the interface parameter.  It is required by the bluealsa type/code etc. 
  b. Add the bluetooth mac address of the destination device (i.e. headset) 
     pcm {
           type bluealsa
-->      interface "hci0"   // Place your hciX that corresponds to the bluetooth device etc.
-->      device 00:19:7F:98:XX:XX   // Place the MAC of the BT headset.
           profile "a2dp"

ctl.!default {
  type bluealsa
--> interface "hci0" // Place your hciX etc.
}

5. If you keep 'your' alsa.conf then ~/.asoundrc isn't necessary.  If you do 'make install' in the bluez-alsa directory, then you should use ~/.asoundrc.  Again, the MAC addr should correspond to the destination device (i.e. your bluetooth headset).

Again, I'm not the author of this code and thus I may have made an error or two.  Hopefully this helps.

Vitaliy69, 2017-01-09:
**erikly01**, thank you for help!

I was forgotten to add the line _interface "hci0"_ to _alsa.conf_ file. However, it does not help.

So I deleted all files and left _~/.asoundrc_ only.

If to change _a2dp_ profile to _sco_, playback begins to work, but only 1 time:

```
wget http://www.kozco.com/tch/piano2.wav
aplay -D bluealsa piano2.wav 
Playing WAVE 'piano2.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```

After some seconds, headset disconnected in _bluetoothctl_. Moreover, can not be connected within several seconds.

The system logs in this moment:

```
Jan 09 10:51:24 pi bluetoothd[300]: Unable to get connect data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
Jan 09 10:53:26 pi bluetoothd[300]: Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jan 09 10:53:57 pi bluetoothd[300]: Unable to get connect data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jan 09 10:54:02 pi bluetoothd[300]: Unable to get connect data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
```

Why does it happen? Moreover, why playback does not work with _a2dp_ profile? What device can not be opened?

By the way, recording with _sco_ profile also not work. The command _arecord -d 3 -D bluealsa test.wav_ is never finished, test.wav is 44 bytes.

Arkq, 2017-01-09:
Hi,

Fist of all make sure you are using current master snapshot of bluez-alsa. Compile it with the --enable-debug option (during configuration). Then, start bluealsa server manually, as follows:

    ./src/bluealsa --disable-hfp

If it complains, about permissions, create /run/bluealsa directory and chmod it to you. Also, make sure, that pulseaudio and other instance of bluealsa (e.g. started by the init.d) is not running.

Then, connect Bluetooth device (e.g. using bluetoothctl). The log from the bluealsa should be as follows (or very similar):

```
bluealsa: ../../src/ctl.c:464: Starting controller loop
bluealsa: ../../src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: ../../src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: ../../src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: ../../src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: ../../src/bluez.c:924: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: ../../src/bluez.c:924: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: ../../src/main.c:217: Starting main dispatching loop
bluealsa: ../../src/bluez.c:846: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: ../../src/transport.c:593: State transition: 0 -> 2
bluealsa: ../../src/transport.c:97: Created new IO thread: HSP Audio Gateway
bluealsa: ../../src/bluez.c:785: HSP Audio Gateway configured for device 1C:48:F9:9D:81:5C
bluealsa: ../../src/transport.c:593: State transition: 0 -> 2
bluealsa: ../../src/io.c:1223: Starting IO loop: HSP Audio Gateway
bluealsa: ../../src/io.c:1074: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: ../../src/transport.c:97: Created new IO thread: HSP Audio Gateway
bluealsa: ../../src/io.c:1134: AT command: +VGS=09
bluealsa: ../../src/io.c:1134: AT command: +XAPL=0000-0000-0100,7
bluealsa: ../../src/io.c:1134: AT command: +IPHONEACCEV=2,1,6,2,0
bluealsa: ../../src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../src/bluez.c:427: A2DP Source (AAC) configured for device 1C:48:F9:9D:81:5C
bluealsa: ../../src/bluez.c:429: Configuration: channels: 2, sampling: 44100
bluealsa: ../../src/transport.c:593: State transition: 0 -> 0
```

The most important line is (if your device does not support AAC, or bluealsa is compiled without AAC supper, you might see SBC instead):

    bluealsa: ../../src/bluez.c:427: A2DP Source (AAC) configured for device 1C:48:F9:9D:81:5C

If the output is somehow "strange", please post it here.

Next, make sure, that the .asoundrc file looks like this:

```
pcm.!default {
      type plug
      slave.pcm {
              type bluealsa
              device "00:19:7F:98:XX:XX"
              profile "a2dp"
      }
      hint {
              show on
              description "BT Headset"
      }
}
```

Then, try to play something using aplay (e.g. `aplay test.wav`). If nothing happens, please send the logs from the bluealsa (from the beginning) and from the aplay.

When the a2dp profile will work, we might try to check the SCO then.

Vitaliy69, 2017-01-10:
Hello, **Arkq**!

Thank you for help!

Here is my logs:

_bluetoothctl_

```
[bluetooth]# connect 00:19:7F:98:FD:29
Attempting to connect to 00:19:7F:98:FD:29
[CHG] Device 00:19:7F:98:FD:29 Connected: yes
Connection successful
[CHG] Device 00:19:7F:98:FD:29 ServicesResolved: yes
[9xxPlantronics]#
```

_bluealsa_

```
bluealsa --disable-hfp
bluealsa: ../../git/src/ctl.c:464: Starting controller loop
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: ../../git/src/bluez.c:924: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: ../../git/src/bluez.c:924: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: ../../git/src/main.c:217: Starting main dispatching loop
bluealsa: ../../git/src/bluez.c:846: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/bluez.c:785: HSP Audio Gateway configured for device 00:19:7F:98:FD:29
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1068: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1217: Starting IO loop: HSP Audio Gateway
bluealsa: ../../git/src/ctl.c:524: New client accepted: 8
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:496: Client closed connection: 8
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:524: New client accepted: 8
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:496: Client closed connection: 8
bluealsa: ../../git/src/ctl.c:527: +-+-
```

_aplay_

```
aplay piano2.wav
../../../git/src/asound/bluealsa-pcm.c:103: Getting transport for 00:19:7F:98:FD:29 type 1
ALSA lib ../../../git/src/asound/bluealsa-pcm.c:707:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:786: audio open error: No such device
```

Arkq, 2017-01-10:
OK, so from the logs I can see, that the A2DP profile is not connected. You can try to issue connect command once more and see if the "important" line (`A2DP Source (XXX) configured for device`) will show up. Without this line, device is not available (from the ALSA point of view). Sometimes profile is not connected - due to signal loss, or other strange "Bluetooth" reasons.

However the Audio Gateway profile is connected, so you can try to use this one. Change "a2dp" profile to "sco" in the .asoundrc and try to play. The quality will be very poor (sampling 8kHz), though. But, there is a possibility that this will not work either. However it is out of the BlueALSA scope. Encoding and data transfer is performed entirely by the Bluetooth kernel module.

In order to debug this problem even further you can dump Bluetooth transfer from the HCI device. See the EDIT of this comment: https://github.com/Arkq/bluez-alsa/issues/7#issuecomment-259513389 You can read the whole comment section of this issue, because it might be somehow related to your problem.

Vitaliy69, 2017-01-11:
I tried another headset with A2DP profile (Jabra instead of 9xxPlantronics), but get "no such" device anyway. May be some problems with the configuration of the operation system? But with PulseAudio no any problems on the same system. 

New logs:

```
# bluetoothctl 
[NEW] Controller 00:19:86:00:0A:D8 tpp [default]
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:19:86:00:0A:D8 Discovering: yes
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent 
Default agent request successful
[NEW] Device 1C:48:F9:40:03:FB JABRA TALK
[bluetooth]# pair 1C:48:F9:40:03:FB
Attempting to pair with 1C:48:F9:40:03:FB
[CHG] Device 1C:48:F9:40:03:FB Connected: yes
[CHG] Device 1C:48:F9:40:03:FB UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:48:F9:40:03:FB UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:48:F9:40:03:FB UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 1C:48:F9:40:03:FB ServicesResolved: yes
[CHG] Device 1C:48:F9:40:03:FB Paired: yes
Pairing successful
[CHG] Device 1C:48:F9:40:03:FB ServicesResolved: no
[CHG] Device 1C:48:F9:40:03:FB Connected: no
[CHG] Controller 00:19:86:00:0A:D8 Class: 0x080000
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 Class: 0x0c0000
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 00:19:86:00:0A:D8 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[bluetooth]# connect 1C:48:F9:40:03:FB
Attempting to connect to 1C:48:F9:40:03:FB
[CHG] Device 1C:48:F9:40:03:FB Connected: yes
Connection successful
[CHG] Device 1C:48:F9:40:03:FB ServicesResolved: yes
[JABRA TALK]# 

# bluealsa --disable-hfp
bluealsa: ../../git/src/ctl.c:464: Starting controller loop
[... 10 lines cut ...]
bluealsa: ../../git/src/bluez.c:785: HSP Audio Gateway configured for device 1C:48:F9:40:03:FB
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1217: Starting IO loop: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1068: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../git/src/bluez.c:427: A2DP Source (SBC) configured for device 1C:48:F9:40:03:FB
bluealsa: ../../git/src/bluez.c:429: Configuration: channels: 2, sampling: 48000
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 0
bluealsa: ../../git/src/ctl.c:524: New client accepted: 12
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:496: Client closed connection: 12
bluealsa: ../../git/src/ctl.c:527: +-+-

# aplay piano2.wav 
../../../git/src/asound/bluealsa-pcm.c:103: Getting transport for 00:19:7F:98:FD:29 type 1
ALSA lib ../../../git/src/asound/bluealsa-pcm.c:707:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:786: audio open error: No such device
```

Bluetooth dump [here](http://www.filedropper.com/bluetooth_3).

Thank you for help!

Arkq, 2017-01-11:
With JABRA it should work without any troubles - at lest the logs say so. Unfortunately, you are using wrong MAC address when trying to connect.

    # bluealsa
    A2DP Source (SBC) configured for device 1C:48:F9:40:03:FB
    ....
    # aplay
    Getting transport for 00:19:7F:98:FD:29 type 1

Change `device` in your .asoundrc file to "1C:48:F9:40:03:FB" and it should work.

Could you provide bluetooth dump from the "not working" device (9xxPlantronics)? Also, could you try to use SCO profile with JABRA (recording and playing) and let me know if it works?

Vitaliy69, 2017-01-11:
Sorry, forgot to change the MAC address.

The playback works fine with both Jabra and 9xxPlantronics BT headsets on SCO profile.

The recording does not work with both Jabra and 9xxPlantronics on SCO profile (the output file is always 44 bytes, the problem like [here](https://github.com/Arkq/bluez-alsa/issues/7#issuecomment-259505570)).

May be the deadlock bug with the recording (used alsa-lib 1.1.2)?

Some logs.

Jabra SCO playback (ok, 9xxPlantronics the same, but have not got A2DP profile detected):

```
# bluealsa --disable-hfp
bluealsa: ../../git/src/ctl.c:464: Starting controller loop
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: ../../git/src/bluez.c:688: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: ../../git/src/bluez.c:924: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: ../../git/src/bluez.c:924: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: ../../git/src/main.c:217: Starting main dispatching loop
bluealsa: ../../git/src/bluez.c:846: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/bluez.c:785: HSP Audio Gateway configured for device 1C:48:F9:40:03:FB
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1068: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1217: Starting IO loop: HSP Audio Gateway
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../git/src/bluez.c:427: A2DP Source (SBC) configured for device 1C:48:F9:40:03:FB
bluealsa: ../../git/src/bluez.c:429: Configuration: channels: 2, sampling: 48000
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 0
bluealsa: ../../git/src/ctl.c:524: New client accepted: 10
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/ctl.c:306: PCM requested for 1C:48:F9:40:03:FB type 2 stream 0
bluealsa: ../../git/src/io.c:75: Opening FIFO for reading: /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-0
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/transport.c:794: New SCO link: 14 (MTU: R:48 W:48)
bluealsa: ../../git/src/ctl.c:381: PCM close for 1C:48:F9:40:03:FB type 2 stream 0
bluealsa: ../../git/src/transport.c:827: Cleaning PCM FIFO: /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-0
bluealsa: ../../git/src/transport.c:834: Closing PCM: 13
bluealsa: ../../git/src/ctl.c:527: +-+-
bluealsa: ../../git/src/io.c:75: Opening FIFO for reading: (null)
bluealsa: ../../git/src/transport.c:804: Closing SCO: 14
bluealsa: ../../git/src/ctl.c:496: Client closed connection: 10
bluealsa: ../../git/src/ctl.c:527: +-+-

# aplay -D bluealsa piano2.wav 
../../../git/src/asound/bluealsa-pcm.c:103: Getting transport for 1C:48:F9:40:03:FB type 2
../../../git/src/asound/bluealsa-pcm.c:587: Setting constraints
Playing WAVE 'piano2.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
../../../git/src/asound/bluealsa-pcm.c:366: Initializing HW
../../../git/src/asound/bluealsa-pcm.c:140: Requesting PCM open for 1C:48:F9:40:03:FB
../../../git/src/asound/bluealsa-pcm.c:157: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-0
../../../git/src/asound/bluealsa-pcm.c:389: FIFO buffer size: 4096
../../../git/src/asound/bluealsa-pcm.c:394: Selected HW buffer: 8000 ?= 4 periods x 2000 bytes
../../../git/src/asound/bluealsa-pcm.c:425: Prepared
../../../git/src/asound/bluealsa-pcm.c:319: Starting
../../../git/src/asound/bluealsa-pcm.c:243: Starting IO loop
../../../git/src/asound/bluealsa-pcm.c:340: Stopping
../../../git/src/asound/bluealsa-pcm.c:340: Stopping
../../../git/src/asound/bluealsa-pcm.c:401: Freeing HW
../../../git/src/asound/bluealsa-pcm.c:186: Closing PCM for 1C:48:F9:40:03:FB
../../../git/src/asound/bluealsa-pcm.c:356: Closing plugin
```
Jabra SCO recording (failed with 44 bytes, 9xxPlantronics the same, but have not got A2DP profile detected):
[... 104 lines cut ...]
bluealsa: ../../git/src/bluez.c:785: HSP Audio Gateway configured for device 1C:48:F9:40:03:FB
bluealsa: ../../git/src/transport.c:590: State transition: 0 -> 2
bluealsa: ../../git/src/transport.c:94: Created new IO thread: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1068: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: ../../git/src/io.c:1217: Starting IO loop: HSP Audio Gateway
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../git/src/bluez.c:493: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../git/src/bluez.c:427: A2DP Source (SBC) configured for device 1C:48:F9:40:03:FB
[... cut ...]
bluealsa: ../../git/src/ctl.c:527: +-+-

root@tpp:~# arecord -d 5 test.wav
../../../git/src/asound/bluealsa-pcm.c:103: Getting transport for 1C:48:F9:40:03:FB type 1
ALSA lib ../../../git/src/asound/bluealsa-pcm.c:707:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
arecord: main:786: audio open error: No such device
```

On the 9xxPlantronics with A2DP profile - recording and playback the error "no such device" (the logs above).

Full dump for 9xxPlantronics: [here](http://www.filedropper.com/bluetooth_4).

Arkq, 2017-01-11:
OK, so it seems there is a progress :) that's great!

SCO playback - checked.

SCO recording - It's sad that it's not working... It would require some debugging, however it might be hard to do it on a distance. If I put my hands on a device with a similar problem, I'll try to fix it. But for now I have to put in on hold.

A2DP playback - It looks like a deadlock. Try to use: `LIBASOUND_THREAD_SAFE=0 aplay -D bluealsa piano2.wav`. However, I'm not able to reproduce this deadlock with aplay... I will have to look at this a little bit longer.

A2DP recording - not supported. A2DP profile is a one-way transport. If you're connecting headset, it won't be possible to capture stream from the build-in microphone. You have to connect a device with A2DP playback capabilities (e.g. smartphone), then it'll be possible to record with A2DP profile. In order to check what "action" is possible with which profile, issue: `amixer -D bluealsa`. In my case it is:

```
Simple mixer control 'Jabra MOVE v2.3.0 - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [on]
  Front Right: Playback 127 [100%] [on]
Simple mixer control 'Jabra MOVE v2.3.0 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 10 [67%] [on] Capture 15 [100%] [on]
Simple mixer control 'Jabra MOVE v2.3.0 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 9
  Mono: Playback 6 [67%]
```

With A2DP it is possible only to playback, with SCO it is possible to capture and playback. And there is a special control, which shows battery level. It is under the "playback" section, though.

I will have some spare time during this weekend (I suppose), so this issue has to wait until then.

Vitaliy69, 2017-01-12:
Good news! A2DP playback works fine with _LIBASOUND_THREAD_SAFE=0_ flag.

Vitaliy69, 2017-01-12:
Good news! An A2DP playback works fine with the _LIBASOUND_THREAD_SAFE=0_ flag.

So it remains to solve the two problems - thread issue on SCO recording and determine the cause of the absence A2DP profile for 9xxPlantronics BT headset.

1. I see function _bluealsa_start_ called from the logs, but _io_thread_ thread not started. The wrong condition on 324 string (is variable _io_started_ init?) or some thread created call issue? May be bad cast in _io_thread_ function? Have not got any ideas. [Here](http://www.filedropper.com/stracetar) system calls logs, perhaps they can be useful.
2. Can't to find the datasheet for Plantronics Discovery 925. Not sure if it A2DP support.

Arkq, 2017-01-12:
Ad. 1
I will check this, but not until the weekend.

Ad. 2
Any info I've found on the net, is that this earpiece supports: Headset v 1.1 (HSP), Hands-free v. 1.5 (HFP). And this is consistent with the bluetooth.dump you've provided. There is no AVDTP communication. So I guess, that A2DP is not supported.

Vitaliy69, 2017-01-23:
Hello, **Arkq**!

Any good news?

I debugged the eternal cycle on the SCO recording (44 bytes output file).

See the problem in _static void *io_thread(void *arg)_ function - the _poll_ function never ending. If to add the timeout 10 seconds, it finished after 10 second with logs from _arecord_ as below (string number are different from git, because some debug logs added):

```
../../../src/asound/bluealsa-pcm.c:252: Starting IO loop
../../../src/asound/bluealsa-pcm.c:321: Exiting IO thread
arecord: pcm_read:2096: read error: No such device
```

What device can not be found? May be some additional logs is required?

Arkq, 2017-01-23:
Lately I'm rather busy, so I've got no time to tinker around bluealsa. But your problem might be kernel or bluez related. You can try to check two things. First, after connecting to the capture stream of the SCO, check if the hcitop (included with this project) shows incoming data (RX/s). If there is none, you might check with tcpdump and see if the dump file is "big" after some recording. Because, if the IO thread hangs on the poll function, it seems that there is no incoming data and this is beyond bluealsa control. My suggestion is to try to use your headset with other PC (other hardware) or other Linux version.

Vitaliy69, 2017-01-23:
Yes, no any activity in RX/s or TX/s when recording and about 15000-17000 bytes/s when playing (both on SCO). It's strange, but with PulseAudio recording works fine. May be some dbus configuration issue. I will continue to research.

Arkq, 2017-01-23:
Hmm... So, if with PA it works, it definitely has to be something wrong
with bluealsa... It might be worth to look at the function responsible
for acquiring SCO socket in the transport.c (transport_acquire_bt_sco).
Maybe, I'm doing something wrong, or not doing something...

If you've got the nerve to research this, I'd be very glad. Like I said
before, with my setup everything works fine, so it is hard for me to
debug it.

Arkq, 2017-01-23:
> no any activity in RX/s or TX/s when recording and about
> 15000-17000 bytes/s when playing

One more thing. Is this 15 kbytes/s during playback in both directions
or only in the TX indicator?

Vitaliy69, 2017-01-23:
It was about 15 kbytes/s for TX and the same for RX during playback.

Vitaliy69, 2017-01-24:
By the way. Is blue-alsa required some Linux kernel specific build flags? I have a light weight kernel for the equipment with low performance.

Arkq, 2017-01-24:
> It was about 15 kbytes/s for TX and the same for RX during playback.

That's strange, but I've got an idea. Could you try to play and capture
at the same time? Because, it might be required to stream audio in
order to receive it from the headset - e.g. headset might not start to
stream audio from the microphone until it receives something to play
(some kind of "optimization"). If it works, try to stop playback and
see if recording still captures sound (e.g. using hcitop).

> Is blue-alsa required some Linux kernel specific build flags?

No, or at least nothing I know about. However, bluealsa requires some
CPU power if working as a A2DP source - power required for audio
encoding.

Vitaliy69, 2017-01-24:
**Arkq**, thanks in advance for a good idea!

Start *arecord* without *aplay*, get a string _Starting_ in the log and waiting for a file descriptor in the poll function.

Start *aplay*, in *arecord* log appears _Starting IO loop_ and recording works fine... 

My be some *AT* commands needed in blue-alsa?

Arkq, 2017-01-31:
What happens when you stop aplay? Does arecord still work? Because I might have some idea how to "fix" this issue :)

It is not AT command issue (I think). The problem is, that I'm using HSP/HFP profile in a non-standard way. This profile should be used to stream audio to the headset and receive one from it - standard usage as a handsfree setup for a phone. However, I thought, that receiving audio and not streaming one should be also possible - save CPU and bandwidth. But it might not be as easy as it seems.

Vitaliy69, 2017-02-01:
Yes... :) I just made tests... When play astop, arecord continues to record the sound while I don not interrupt the recording manually.

Arkq, 2017-02-01:
Check if code from the [sco-hotstart](https://github.com/Arkq/bluez-alsa/tree/sco-hotstart) branch resolves this issue.

Vitaliy69, 2017-02-02:
Recording does not work yet. :-(

The debug string "Hot-starting SCO for reading" not appear.

If to start arecord only, t->sco.spk_pcm.fd == -1 and t->sco.mic_pcm.fd == -1.

If to start aplay only, t->sco.spk_pcm.fd > 0 and t->sco.mic_pcm.fd == -1.

If to start arecord (at this moment t->sco.spk_pcm.fd == -1 and t->sco.mic_pcm.fd == -1), then aplay - t->sco.spk_pcm.fd > 0 and t->sco.mic_pcm.fd > 0.

Arkq, 2017-02-02:
I took me a while, but I might know what's going on. I've pushed one debug line to the sco-hotstart branch. Please, check if this line is printed ("FIFO not connected yet") when using arecord only. If yes, then remove the O_NONBLOCK flag from the open call from [this](https://github.com/Arkq/bluez-alsa/blob/sco-hotstart/src/io.c#L97) line and remove [this](https://github.com/Arkq/bluez-alsa/blob/sco-hotstart/src/io.c#L107) one. Then try again. (After this changes, bluealsa might be not useable for A2DP recoding, but it will reveal the problem).

Vitaliy69, 2017-02-02:
Yes, I get the line "FIFO not connected yet". And... SCO recording works fine with removed O_NONBLOCK flag from 97 line and deleted 107 line. :-)

Arkq, 2017-02-02:
OK, thanks a lot for your help with this bug. Now I know exactly
what causes this issue - it is not caused by the out of standard HSP
profile usage...

I will fix this in a next few days.

Arkq, 2017-02-02:
@Vitaliy69, please check if the current master snapshot works for you.

closed 2017-02-02 by Arkq

Vitaliy69, 2017-02-03:
Works fine! Thank you!

keyoffecka, 2017-11-09:
Hi Arkq!

First of all, I really appreciate the work you've done so far. This is really a very good project unlike pa which "... breaks your audio" (sic)

Unfortunately I've ran into a similar issue with my bluetooth device. The playback works just fine, aplay works, mplayer works, chromium works, I can listen to YouTunbe audio. But the recording doesn't. And unfortunately I have no more ideas nor clues how to proceed.

I have alsa-1.1.2
bluz 5.47
bluealsa build against the commit #9736a05

Please take a look at the logs.

asound.conf:

defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "04:5D:4B:53:8F:1D"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 0

pcm.duplex {
  type asym;
  playback.pcm "bluz";
  capture.pcm "bluzmic"
}

pcm.!default {
  type plug;
  slave.pcm "duplex";
}

pcm.bluzmic {
  type bluealsa
  device "04:5D:4B:53:8F:1D"
  profile "sco"
  "delay" 0
}

pcm.bluz {
      type plug
      slave.pcm {
              type bluealsa
              device "04:5D:4B:53:8F:1D"
              profile "a2dp"
              "delay" 0
      }
      hint {
              show on
              description "BT Headset"
      }
}

=====
sudo bluetoothd -d -n

bluetoothd[18792]: Bluetooth daemon 5.47
bluetoothd[18792]: src/adapter.c:adapter_init() sending read version command
bluetoothd[18792]: Starting SDP server
bluetoothd[18792]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:052f
bluetoothd[18792]: src/plugin.c:plugin_init() Loading builtin plugins
bluetoothd[18792]: src/plugin.c:add_plugin() Loading hostname plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading wiimote plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading autopair plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading policy plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading a2dp plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading avrcp plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading network plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading input plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading hog plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading gap plugin
bluetoothd[18792]: src/plugin.c:add_plugin() Loading scanparam plugin
[... 465 lines cut ...]
../shared/ctl-client.c:321: Opening PCM FIFO (mode: RO): /var/run/bluealsa/hci0-04:5D:4B:53:8F:1D-2-1
bluealsa-pcm.c:301: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:340: Prepared
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:205: Starting
../shared/ctl-client.c:390: Requesting PCM resume for 04:5D:4B:53:8F:1D
^CAborted by signal Interrupción...
arecord: pcm_read:2096: read error: Llamada al sistema interrumpida
bluealsa-pcm.c:245: Stopping
bluealsa-pcm.c:308: Freeing HW
../shared/ctl-client.c:364: Closing PCM for 04:5D:4B:53:8F:1D
bluealsa-pcm.c:261: Closing plugin

====

ls -la test.wav 
-rw-r--r-- 1 echo users 44 nov  8 23:34 test.wav

As you see I have that 44-byte issue, meaning that the .wav is empty. Even more, despite I say to record the audio only for 3 seconds, it doesn't stop it. Mind the

^CAborted by signal Interrupción...

If you could only give me a hint where to dig further that would be really great. Thanks!

keyoffecka, 2017-11-10:
I am sorry for the mess here. I hope you haven't lost your time trying to analyse the issue reported. It turns out the issue has nothing to do with BlueAlsa. I found a solution here:

https://forums.gentoo.org/viewtopic-p-7846088.html?sid=e055c02e2ca9e8edb99b75d9e24c97b8#7846088

The guy mentions another thread with the solution

https://forums.gentoo.org/viewtopic-p-6910768.html#6910768

which comes from a long and a more general discussion on the Arch forum

https://bbs.archlinux.org/viewtopic.php?id=91496

So, just for the record, just in case if somebody else struggling with the same issue. I had to enable CONFIG_USB_EHCI_TT_NEWSCHED in the kernel.

But anyway, thanks for a good software.

ageorgios, 2018-09-02:
@keyoffecka 
I am following the above configuration you got
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "00:1D:DF:F8:95:85"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

pcm.duplex {
type asym;
playback.pcm "bluz2";
capture.pcm "bluzmic"
}

pcm.!default {
type plug;
slave.pcm "duplex";
}

pcm.bluzmic {
type bluealsa
device "00:1D:DF:F8:95:85"
profile "sco"
"delay" 1000
}

pcm.bluz {
type plug
slave.pcm {
type bluealsa
device "00:1D:DF:F8:95:85"
profile "sco"
"delay" 1000
}
hint {
show on
description "BT Headset"
}
}

to use it with google assistant that is a full duplex audio application.

But although it can record being on profile sco, it does not automatically turn to a2dp when playing audio and nothing is heard

commit 04f5dec2 (2017-02-02): Synchronize PCM FIFO opening

POSIX-compatible FIFO opening in the non-blocking mode might cause a
headache (blocking mode is a nightmare too). In order to synchronize
this process, we will use signal (message) sent via the communication
socket. The purpose of this signal is to indicate, that the reading
side is ready, so now we can open the writing side.

During certain conditions, this lack of synchronization causes the IO
thread associated with the SCO to fail during FIFO opening, and then
to block in the poll system call until some event occurs.

Fixes #22