arkq/bluez-alsa #23: bad sound with VLC

Ruhrmolch, 2016-12-29:
Hi,
i got that nice thing working. But with VLC, there is a problem with the sound, its very bad, constantly interrupted. If i take a look at the VLC-console-output, i know why:
```
bluealsa-pcm.c:243: Starting IO loop
bluealsa-pcm.c:340: Stopping
bluealsa-pcm.c:425: Prepared
bluealsa-pcm.c:319: Starting
bluealsa-pcm.c:243: Starting IO loop
bluealsa-pcm.c:340: Stopping
bluealsa-pcm.c:425: Prepared
bluealsa-pcm.c:340: Stopping
bluealsa-pcm.c:340: Stopping
```

The plugin seems to be constantly started and stopped by vlc. VLC is using the default-alsa device.
That is my .asoundrc:

```
pcm.btspeaker {
 type plug
  slave {
    pcm {
      type bluealsa
      interface hci0
      device 88:C6:26:C2:76:35
      profile "a2dp"
    }
  }
  hint {
    show on
    description "BT Headset"
  }
}

pcm.!default {
    type plug
    slave.pcm "btspeaker"
}
ctl.!default {
	type hw
	card 0
}
```

Arkq, 2017-01-07:
Hi,

Finally, I've got some time to dig around this issue and it seems that the problem is beyond my understanding....

The problem is (it seems it is) with the VLC audio synchronization code. For some strange reasons there is a build-in constraint, which determines audio-video maximal physical desynchronization. If the audio needs to be sent to the device in an advance (due to device internal delay, which is the case here), there is a maximal value for this advance time - and bluealsa exceeds this time approximately 30x. I've try to adjust these constrains, but the result was not satisfactory - there is no glitching (due to device start and stop), but the audio is not synced properly.

For the record, this is the change I've made (current master snapshot of VLC) to increase sync tolerance:

```
--- a/include/vlc_aout.h
+++ b/include/vlc_aout.h
@@ -50,7 +50,7 @@
 
 /** Maximum delay of actual audio playback time from coded PTS,
  * above which upsampling will be performed */
-#define AOUT_MAX_PTS_DELAY              (3 * CLOCK_FREQ / 50)
+#define AOUT_MAX_PTS_DELAY              (100 * CLOCK_FREQ / 50)
 
 /* Max acceptable resampling (in %) */
 #define AOUT_MAX_RESAMPLING             10
```

Also, other related files:

- [src/audio_output/dec.c:255](https://github.com/videolan/vlc/blob/master/src/audio_output/dec.c#L255)
- [modules/audio_output/alsa.c](https://github.com/videolan/vlc/blob/master/modules/audio_output/alsa.c)

I will not investigate this further. All reference implementations (imho) work correctly, e.g. aplay, mplayer (even firefox seems to work). If anyone with VLC code experience would like to contribute, I'd be very happy. Btw. I'm not saying, that it is not a bluealsa bug, I just don't know where to look for clues.

Ruhrmolch, 2017-01-09:
Do you have the same problem with vlc?

Arkq, 2017-01-09:
Yes, I was able to reproduce this bug. Bluealsa and VLC do not come along, unfortunately. There is a small chance, that when I figure out how to fix #15 , this issue will be less severe. But I wouldn't bet on it.

In the meantime, you can use mpv/mplayer as a replacement for VLC. I'm using it as a reference video player.

Arkq, 2017-04-10:
Since commit e7a4c40 it is possible to customize (overwrite) PCM delay. Using this functionality, I've managed to force VLC to work without patching the source code. One can use something like this:

```
pcm.headset {
	type plug
	slave.pcm {
		type bluealsa
		device "88:C6:26:C2:76:35"
		profile "a2dp"
		delay -20000
	}
	hint {
		show on
		description "BT Headset"
	}
}
```

The key part is `delay -20000`. This value might vary between setups. If VLC is still stopping and starting PCM, increase or decrease this value, e.g.: -25000, -30000, -40000, -10000.

The result is, that audio is audible, but out of sync... by a second or two. So, use the keyboard shortcut `J` in order to "fix" this out of sync audio (in my case it was -1000 ms). Done.

Arkq, 2020-04-05:
I think that this issue was resolved in #231

closed 2020-04-05 by Arkq