arkq/bluez-alsa #24: bluealsa.service failed to load because of soft or hardblock

Ruhrmolch, 2016-12-29:
If, for example, the wireless switch on my notebook is turned off, the service fails to start at boot.
If i turn it on later, the service is not ready.

nheir, 2017-01-01:
Does the bluetooth service succeed to start ?
(assuming you use the .service from AUR)

Ruhrmolch, 2017-01-01:
yes, it fails. so, in case of switching bluetooth on, after the boot, you have to restart the service manually.
```
● bluealsa.service - BluezAlsa proxy
   Loaded: loaded (/usr/lib/systemd/system/bluealsa.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Thu 2016-12-29 09:53:43 CET; 7min ago
  Process: 511 ExecStart=/usr/bin/bluealsa (code=exited, status=1/FAILURE)
 Main PID: 511 (code=exited, status=1/FAILURE)

Dez 29 09:53:43 Django systemd[1]: Started BluezAlsa proxy.
Dez 29 09:53:43 Django bluealsa[511]: /usr/bin/bluealsa: No HCI device available
Dez 29 09:53:43 Django systemd[1]: bluealsa.service: Main process exited, code=exited, status=1/FAILURE
Dez 29 09:53:43 Django systemd[1]: bluealsa.service: Unit entered failed state.
Dez 29 09:53:43 Django systemd[1]: bluealsa.service: Failed with result 'exit-code'.
```

nheir, 2017-01-01:
You may try a udev rule like : 

    ACTION=="add", KERNEL=="hci0", ENV{SYSTEMD_WANTS}+="bluealsa.service"

It should run bluealsa when the hci0 interface comes up

Ruhrmolch, 2017-01-03:
That does the job. Thank you!!!

But do you have any idea, why this does not work:

```
[Unit]
Description=BluezAlsa proxy (modified)
Requires=bluetooth.service
After=bluetooth.target bluetooth.service

[Service]
Type=simple
User=bluealsa
Group=audio
ExecStart=/usr/bin/bluealsa

[Install]
WantedBy=multi-user.target

```

nheir, 2017-01-03:
What do you mean it does not work ? 
Do you expect the service to start whenever bluetooth.target is reach ?

Ruhrmolch, 2017-01-07:
yes, indeed. freedesktop.org:

>     This target is started automatically as soon as a Bluetooth controller is plugged in or becomes available at boot.
> 
>     This may be used to pull in Bluetooth management daemons dynamically when Bluetooth hardware is found.

tomatopeel, 2018-11-17:
I would also be interested to hear why this is. I've also had to add the udev rules file so thanks for that @nheir 

But it does feel hacky, why doesn't the unit file work like @Ruhrmolch is asking?

@Arkq sorry to highlight, seems like this issue might have gotten lost under the pile though

Arkq, 2019-03-11:
With the current master this issue should no longer occur. Right now, HCI device is fetched from the BlueZ via the D-Bus API. Also, it is possible to start bluealsa even though there is no HCI device present in the system - it will be automatically added to the bluealsa when plugged-in.

closed 2019-03-11 by Arkq