arkq/bluez-alsa #25: input sound issues

roizcorp, 2017-01-13:
hi,

I've connected phone in Raspi via BT (bluetoothctl) while bluez-alsa service is running (great stuff!)
However, when I play music I cannot hear it in the Pi, nor I could find anyway to route the BT input into my audio card
Appreciate the help

Arkq, 2017-01-13:
See this issue: #11

roizcorp, 2017-01-13:
i execute the following command `arecord -f dat -D bluealsa | aplay` and it worked!!

Can't wait to make it an all-in-one hands free system for my car!

roizcorp, 2017-01-14:
sorry for re-open this...I realize now that I'm stuck...
yes.. the arecord -f dat... works, but it will not share...the output alsa card should play mpd hi-fi music (that's why I chose alsa) AND the BT input...
**Scenario**: I drive my car with the lovely music I have in my car pi mpd, and the navigation app on the phone wants to inform me about something...I expect the sound channeled from the BT to kick in while the music is playing and say whatever it has to say.

I have this solution now with all male aux cable from my phone to my car pi but I would like to spare the cable and spare the static elecric sound from the phone whenever I do something with it or just charge it

I know you are not supposed to accept these kind of issues but will indeed help me to achieve the use case...

onli, 2017-01-17:
I think that is because mixing is not yet possible with bluealsa. See https://github.com/Arkq/bluez-alsa/issues/12#issuecomment-267835252, I don't think that has changed yet, though I don't see an open issue for it.

roizcorp, 2017-01-17:
I thought to overcome this by using alsa's dmix and the alsaequal
[](https://wiki.archlinux.org/index.php/Advanced_Linux_Sound_Architecture#System-wide_equalizer)
can you list how the bluealsa (device,plugin,card,subcard - god damn it ALSA!) would show in the configuration

Arkq, 2017-01-17:
> yes.. the arecord -f dat... works, but it will not share

If the problem is with physical audio device (not bluealsa) you might
try to use dmix (however, it is strange, that it is not used by default
for your soundcard - if it natively does not support mixing). However,
if the problem is with the bluealsa, hmmmm onli is right, currently it
is not possible (and dmix won't work with bluealsa).

And be sure, that when you are referring to the physical card, you are
using "default" PCM device (or something like that), not "hw:X". Using
hardware directly omits ALSA plugins which provide e.g. mixing.

On Tue, 17 Jan 2017 05:36:15 -0800
roizcorp <notifications@github.com> wrote:
> I thought to overcome this by using alsa's dmix and the alsaequal
> [](https://wiki.archlinux.org/index.php/Advanced_Linux_Sound_Architecture#System-wide_equalizer)
> can you list how the bluealsa (device,plugin,card,subcard - god damn it ALSA!) would show in the configuration
>

roizcorp, 2017-01-17:
if I understand the issue correctly (and will report back in case solved or different analysis), is that bluealsa consumes the default audio card and will not allow any other feed, and on the other hand if the default audio card is already used by other feed then bluealsa cannot share as quoted in one of your commend "resource is busy"

Arkq, 2017-01-17:
> bluealsa consumes the default audio card and will not allow any other feed

Bluealsa is a software ALSA plugin which implements PCM device.
Currently it is not possible to connect more than one client to the
bluealsa PCM (one client for reading - recodring, and one for playing -
playback). It has nothing to do with other audio PCM devices.

> if the default audio card is already used by other feed then bluealsa cannot share

I don't know if I understand you correctly, but in such a case,
bluealsa is not sharing - the other PCM device is sharing. And it is
possible that natively that device is not able to share connection -
mix playbacks (it is common). So, ALSA guys have created dmix plugin.

Can you outline your whole setup (including .asoundrc configuration
file)?

roizcorp, 2017-01-20:
ok, so as much as I read in the Alsa documentation it gets worst...
I use a raspberry pi, which has a crappy sound card, that why I bought an external GPIO based sound card. I use MPD which allows me to specify which card to use. Every time the Pi starts the MPD kicks in and starts to play HiFi music (24/192). HiFi music is important to me and this is why I keep use Alsa rather than PulseAudio.

Now, this is a car based Pi, so I want that my navigation app in the phone will still be able to tell me stuff while I listen to the music. Currently I connect the phone into the soundcard's input but as a result I static noise from the phone as well.

So the idea is to connect the phone to the Pi via BT, and whenever it would have something to say I could hear it, this is why I need both BlueAlsa and MPD streaming music/sound in the same time.

configuration is as follows (specially made for the blueAlsa, I did not need to make one where only the MPD was in play):
define the default sound card in /etc/asound.conf
`defaults.pcm.!card 1
defaults.ctl.!card 1`

declare BlueAlsa's interface, device and profile
`defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "XX:XX:XX:XX:XX:Xx"
defaults.bluealsa.profile "a2dp"
`
problems:

1. even if BT is connected and phone plays music, I cannot hear anything unless I use the `arecord -f dat -D bluealsa | aplay` (ecasound command is correct but it does not gets to stream the music for some reason, all I get is the "Prepared")
2. the solution in section 1 does not work if MPD already started to play music and by that made the device "busy", if MPD is stopped then this will work.

I need to maintain the HiFi music while allow the phone to kick in whenever it needs to say something.
Thanks

roizcorp, 2017-02-14:
sorted it out with a physical mixer in the sound card, in order to maintain high fidelity.
MPD is playing via hifi card, bluealsa is playing via the pi's default audio card `arecord -f dat -D bluealsa |  aplay`. then simply connecting physically between the pi lineout and the hifi card's input.

closed 2017-02-14 by roizcorp