arkq/bluez-alsa #27: Loss of frames with pjsip

Vitaliy69, 2017-02-02:
Hello and TIA for good work!

I try to use _pjsua_ (SIP application) from _pjsip_ library with latest version of bluez-alsa (SCO profile).

The sound always interrupt, the conversation example [here](http://www.filedropper.com/output_19). I have some suspicions with incorrect buffer length of the buffer or drop the buffer before it finish.

The CPU load for _pjsua_ is about 10% and 5% for _bluez-alsa_. So the problem is not performance.

_aplay_ works fine on the same system with _bluez-alsa_, _pjsua_ also works fine on the same system with _PulseAudio_.

How to debug and fix this error?

If to set _pcm->frame_size_ to 15 bytes (calculated as 16) in _bluealsa_hw_params_ function, the sound is almost continuous, but cracks appear.

Vitaliy69, 2017-02-02:
The buffer size for aplay (works) fine:
```

../../../git/src/asound/bluealsa-pcm.c:366: Initializing HW
../../../git/src/asound/bluealsa-pcm.c:140: Requesting PCM open for 1C:48:F9:40:03:FB
../../../git/src/asound/bluealsa-pcm.c:157: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-0
../../../git/src/asound/bluealsa-pcm.c:389: FIFO buffer size: 4096
../../../git/src/asound/bluealsa-pcm.c:394: Selected HW buffer: 8000 ?= 4 periods x 2000 bytes
```

And for pjsua:

```
../../../git/src/asound/bluealsa-pcm.c:140: Requesting PCM open for 1C:48:F9:40:03:FB
../../../git/src/asound/bluealsa-pcm.c:157: Opening PCM FIFO (mode: RO): /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-1
../../../git/src/asound/bluealsa-pcm.c:394: Selected HW buffer: 1600 ?= 5 periods x 320 bytes
../../../git/src/asound/bluealsa-pcm.c:425: Prepared
../../../git/src/asound/bluealsa-pcm.c:366: Initializing HW
../../../git/src/asound/bluealsa-pcm.c:140: Requesting PCM open for 1C:48:F9:40:03:FB
../../../git/src/asound/bluealsa-pcm.c:157: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-1C:48:F9:40:03:FB-2-0
../../../git/src/asound/bluealsa-pcm.c:389: FIFO buffer size: 4096
../../../git/src/asound/bluealsa-pcm.c:394: Selected HW buffer: 2240 ?= 7 periods x 320 bytes

```

Arkq, 2017-02-05:
> If to set pcm->frame_size to 15 bytes (calculated as 16)

Are you sure, that the frame_size is calculated as 16 bytes? Because it seems ridiculously wrong. It should be 2 bytes for mono and 4 bytes for stereo. If however it is calculated as 16 bytes, then it might a cause of trouble.

> The CPU load for pjsua is about 10% and 5% for bluez-alsa. So the problem is not performance.

Actually this issue might be performance related. But the bottleneck in not a CPU, but an overall bluealsa performance/timing. The problem might be too small period. I've tried to set the minimum requirement for period size to 1024 bytes, but it is not respected... I remember that I've seen such an interruption in the audio by myself, but I can't remember with which application.

I'd like to reproduce this issue with the pjsua, however I don't know how to set it up :D - I've got no (zero, null, zilch) experience with VoIP. Could you give me a hint how to setup pjsua. Some "howto" article, or simple configuration. I was able to compile pjproject and that's all :).

Vitaliy69, 2017-02-06:
> Are you sure, that the frame_size is calculated as 16 bytes?

Sorry, forgot divided into 8 (_snd_pcm_format_physical_width(io->format)_ = 16, _io->channels_ = 1). So it 2 bytes. But changing this variable affects the sound interruption.

Ok, to reproduce this issue need to install VoIP server and compile _pjproject_.

I use [AsteriskNOW](http://www.asterisk.org/downloads/asterisknow) Linux distro with Asterisk 11. It is free and based on Cent OS. Install on the any virtual machine (use free Oracle VM VirtualBox). 

When it's done, FreePBX web interface will available by [address](http://192.168.75.236/admin) (change to own IP address). It is need to add 2 users (_Applications -> Extensions -> Add an Extension -> Generic CHAN SIP Device_, set any value for _User Extension_ field (for example _600_), _Display Name_ (for example _User 600_) and password in the _Secret_ field (for example _ab1234_), press _Submit_ and then _Apply Config_ at the top of the page). When configure the network in _Settings -> Asterisk SIP Settings tab_ (in _Local Networks_ field, press _Submit_ and _Apply Settings_). 

The VoIP SIP server is ready. :-)

Then compile the _PJSIP_ (I try 2.4.5 and 2.5.5 versions). 

```
wget http://www.pjsip.org/release/2.6/pjproject-2.6.tar.bz2
tar xvjf pjproject-2.6.tar.bz2
cd ./pjproject-2.6/
./configure
make dep
make
```

To run _PJSUA_ just run the command _./pjsip-apps/bin/pjsua-x86_64-unknown-linux-gnu_ after successful compilation.

When select _ (+a)    Add new accnt_:

```
>>> +a
Your SIP URL: (empty to cancel): sip:600@192.168.75.236
URL of the registrar: (empty to cancel): sip:192.168.75.236
Auth Realm: (empty to cancel): *
Auth Username: (empty to cancel): 600
Auth Password: (empty to cancel): ab1234
```

Just change credentials to own. 

Repeat that on the second PC (add the second account):

```
>>> +a
Your SIP URL: (empty to cancel): sip:601@192.168.75.236
URL of the registrar: (empty to cancel): sip:192.168.75.236
Auth Realm: (empty to cancel): *
Auth Username: (empty to cancel): 601
Auth Password: (empty to cancel): ab1234
```

And make the call from first PC to the second:

```
m
(You currently have 0 calls)
Buddy list:
 -none-

Choices:
   0         For current dialog.
  -1         All 0 buddies in buddy list
  [1 - 0]    Select from buddy list
  URL        An URL
  <Enter>    Empty input (or 'q') to cancel
Make call: sip:601@192.168.75.236
```

To answer the call, press _a_ hot button and send _200_ reply code on the second PC.

With _bluez-alsa_ this issue reproduced, with _pulseaudio_ no any problems. :-)

Vitaliy69, 2017-02-11:
Did you manage to find out anything? Be may need my help?

Arkq, 2017-02-13:
Unfortunately not yet. When something pops out, or I will need some
assistance, I will write immediately :)

Vitaliy69, 2017-03-03:
Hello! Are there any news?

Arkq, 2017-03-04:
Hi! I'm constantly working on it, and it seems, that it is a little bit harder than I thought. There is no much info on the net about the ALSA PCM I/O plugin architecture.... I'm making some progress, though :] However, it might take another week or two.

Arkq, 2017-04-09:
Hi.

If you're still using bluez-alsa try fresh master snapshot. I think, that finally I've made things right (commit [ca72c1e](https://github.com/Arkq/bluez-alsa/commit/ca72c1e9d8f7be773066a11776342df8fcab5bd9)). I wasn't able to test it with the PJSIP (I have failed to set it up right....), though.

Vitaliy69, 2017-04-09:
Of course, I like bluez-alsa and still using it. Thank you very much for the fix! I will check it within a week.

Vitaliy69, 2017-04-12:
Thanks in advance! No any problems with PJSIP with the latest master snapshot.

closed 2017-04-12 by Vitaliy69

Arkq, 2017-04-12:
That's great! :tada:

vs1785, 2020-10-31:
How were you able to get PJSIP detect bluealsa devices?

It never detects for me and gives me an error of no default device exists. PJMEDIA_EAUD_NODEFDEV