arkq/bluez-alsa #28: Some videos in FF don't play

onli, 2017-02-02:
This one is strange. https://www.youtube.com/watch?v=g0fqYeT8u9A is an example. It plays just fine when not using bluealsa, which shows it is not a codec issue. But as soon as FF outputs sounds via bluealsa, that video has no sound at all. Other videos *do* work, https://www.youtube.com/watch?v=v0nmHymgM7Y&list=PL-I91CUhz4mhHyTz5p3rrqVtUEwYbTFyD as example.

I'll add log output in a separate comment, will have to restart FF and play the video again.

onli, 2017-02-02:
Nope, I see nothing sound-related in the logs. bluealsa started in the terminal, firefox in terminal, syslog. Is there a other place I should look?

I hope you can reproduce this also without, sorry.

Arkq, 2017-02-03:
Yes, I'm able to reproduce it. It is really strange :/ My first though is, that the audio settings are incompatible with supported ones. However, the plug plugin should correct everything.

The output is:
```
../../../src/asound/bluealsa-pcm.c:123: Getting transport for 1C:48:F9:9D:81:5C type 1
../../../src/asound/bluealsa-pcm.c:609: Setting constraints
../../../src/asound/bluealsa-pcm.c:377: Closing plugin
```

so the HW parameters are not even initialized. I will have to look in the FF code (once more :D).

onli, 2017-04-28:
https://www.youtube.com/watch?v=g0fqYeT8u9A is working by now for me, with the current master of bluealsa. Seems like one of the other changes (or a FF update) has fixed that one. I'll keep my eyes out for new broken videos, but maybe this is already solved.

Arkq, 2017-04-28:
It is possible, but not likely... When you've reported this issue, I've checked it on my setup (I do not remember exact versions of FF and its dependencies) and video did not play. Some time later, though, I've check that again (in order to investigate it further - lurk into FF source code) and the video worked fine (without any upgrades). So, I think, that this issue was "fixed" by Youtube - video was reencoded or some other audio format is now available. But the fact remains, that there is some issue.

I'm not using FF on a daily basis, so it is hard for me to spot anything. So, if you find some troubles, please, report immediately :)

Arkq, 2019-11-25:
Closing stale issue due to lack of information.

closed 2019-11-25 by Arkq