arkq/bluez-alsa #30: Possible to have 2 headsets in .asoundrc?

emugel, 2017-02-09:
This is not a bug bug a mere question.
I have 2 different pairs of bluetooth loudspeakers, They both work independently, but right now my `.asoundrc` looks like this:
```
# defaults.bluealsa {
#     interface "hci0"
#     device "43:25:03:FF:D5:BF"
#     profile "a2dp"
# }
defaults.bluealsa {
    interface "hci0"
    device "00:E0:4C:E3:44:4F"
    profile "a2dp"
}
```
As of now, it's one or the other. I can use a script to change/move the file, but was wondering if there was another way (e.g. maybe have one as hci1?)

But the way, thanks a lot for bluez-alsa!

Arkq, 2017-02-09:
Yes, it is possible. However, you have to define PCMs "statically" (to be honest I don't know how to call it in an ALSA way). See the example below:

```
pcm.myname1 {
        type plug
        slave.pcm {
                type bluealsa
                device "43:25:03:FF:D5:BF"
                profile "a2dp"
        }
        hint {
                show on
                description "Bluetooth something something"
        }
}
pcm.myname2 {
        type plug
        slave.pcm {
                type bluealsa
                device "00:E0:4C:E3:44:4F"
                profile "a2dp"
        }
}
pcm.myname2-sco {
        type plug
        slave.pcm {
                type bluealsa
                device "00:E0:4C:E3:44:4F"
                profile "sco"
        }
}
```

emugel, 2017-02-10:
Alright! It works :)
Right now I can use 
```
echo "connect 43:25:03:FF:D5:BF" | bluetoothclt
mpg123 --au - mp3/AstralProjection-Anything_can_happen.mp3 | aplay -D bt0
``` 
or
```
echo "connect 00:E0:4C:E3:44:4F" | bluetoothclt
mpg123 --au - mp3/AstralProjection-Anything_can_happen.mp3 | aplay -D bt1
``` 
To connect and play an mp3 file to either one.

closed 2017-02-10 by emugel