arkq/bluez-alsa #31: Workaround for missing libasound_module_pcm_bluealsa.so and other..

kiwi-ed, 2017-02-10:
Hi,

I have cross-compiled bluealsa for a NXP imx7 and it sort-of runs... I have a couple of questions.
1) You mentioned in one reply that bluealsa acts as another sound card, is this correct?
2) This being the case if I want to use my imx7 as a BT speaker  I have to link basically link two sound cards together via arecord of similar  ( bluealsa <=> aplay )?
3) If the last answer is yes is there any way to modify blueasla itself to use hw:0,0 directly? 
4) Tying arecord as suggested by another post I get an error message of a missing library: libasound_module_pcm_bluealsa.so. The Yocto version and Bluez5 version I use no longer builds or includes this library. Any suggestions?

So far with other applications on this processor I have got away from using pulseaudio and getting BT to use only alsa via bluez5 would be great. My BT audio ( my board as sink ) works well with pulseaudio, but is has limited resources hence wanting to use bluealse.

Thanks.

Arkq, 2017-02-12:
> bluealsa acts as another sound card, is this correct?

In some extends yes, but BlueALSA is not a physical sound card. It is a PCM device (but not sound card).

> I have to link basically link two sound cards together via arecord of similar?

Yes.

> is there any way to modify blueasla itself to use hw:0,0 directly?

It is a way, but I won't do it. Try to use the `bluealsa-aplay` (from the current master snapshot). It omits ALSA PCM plugins, so there should be no problem with missing library. Building this functionality into the bluealsa itself wouldn't save much CPU cycles (compared with the bluealsa-aplay), but it would definitely make a mess in the code :D.

> I get an error message of a missing library: libasound_module_pcm_bluealsa.so.

This library is a PCM plugin for BlueALSA (shipped with bluez-alsa). However, it seems that it was installed in the wrong place. This file, has to be placed in the ALSA modules directory (on standard linux distribution, this file is copied to the `/usr/lib/alsa-lib/`.

kiwi-ed, 2017-03-08:
Hello,

Thanks for the information so far. After messing around with pulse and bt on my custom board, pulse is a real pain trying to get it to play nicely with all my other ALSA based music renderers. So I'm going to have another crack at bluez-alsa.

You mentioned that bluez-alsa acts as a pcm device, what would be the best way to connect this pcm to the default audio hardware or will it render to my sound card as it's a pcm anyway [ using  embedded Linux [?  ( based on your reply:  I have to link basically link two sound cards together via arecord of similar? Yes. )

Do I have to include this pcm in asound.conf etc?

The library "libasound_module_pcm_bluealsa" doesn't seem to be part of bluez any more, if bluez5 has dumped supporting ALSA it makes sense that this library would no longer be included in their bluez5 package. Any suggestions on what to do here? Could the bluez4 libasound_module_pcm_bluealsa work with bluez5?

Finally, all I require is a2dp over bt, so would I loose much reverting back to bluez4 & ALSA?

Kind regards,

Ed.

kiwi-ed, 2017-03-08:
Hi,

Adding to the email below, is it possible to use gstreamer, playbin with bluez-alsa?

________________________________
From: Edward Tyrrell
Sent: 08 March 2017 07:44:39
To: Arkq/bluez-alsa; Arkq/bluez-alsa
Cc: Author
Subject: Re: [Arkq/bluez-alsa] Workaround for missing libasound_module_pcm_bluealsa.so and other.. (#31)

Hello,

Thanks for the information so far. After messing around with pulse and bt on my custom board, pulse is a real pain trying to get it to play nicely with all my other ALSA based music renderers. So I'm going to have another crack at bluez-alsa.

You mentioned that bluez-alsa acts as a pcm device, what would be the best way to connect this pcm to the default audio hardware or will it render to my sound card as it's a pcm anyway [ using  embedded Linux [?  ( based on your reply:  I have to link basically link two sound cards together via arecord of similar? Yes. )

Do I have to include this pcm in asound.conf etc?

The library "libasound_module_pcm_bluealsa" doesn't seem to be part of bluez any more, if bluez5 has dumped supporting ALSA it makes sense that this library would no longer be included in their bluez5 package. Any suggestions on what to do here? Could the bluez4 libasound_module_pcm_bluealsa work with bluez5?

Finally, all I require is a2dp over bt, so would I loose much reverting back to bluez4 & ALSA?

Kind regards,

Ed.

mill1000, 2017-12-07:
A note regarding the missing library libasound_module_pcm_bluealsa.so

It appears the some Debian-based systems (e.g. Raspbian) install alsa-plugins to an architecture specific folder (e.g. /usr/lib/arm-linux/gnueabihf) instead of /usr/lib/.

A symbolic link can solve this issue pretty easily

Arkq, 2017-12-07:
Yes, symlink would be fine, however I'd like to go for a better detection for ALSA directory. Right now it is kind of naive approach: https://github.com/Arkq/bluez-alsa/blob/99d9d72a5aed22dd755485a2661e209ded697079/configure.ac#L93

However, I have no idea how to make it better (if it is even possible at the first place). After all, distribution specific configuration should be configured by a packager (person responsible for creating distribution specific package). Anyway, if someone knows how to correct configure.ac file, I'd be very happy to see some PR :)

infirit, 2017-12-07:
@mill1000 what is the output of the below command run from a terminal.

```
pkg-config --variable=libdir alsa
```

If it points to the the correct libdir it is very easy to fix.

mill1000, 2017-12-07:
@Arkq 
Yes it appears the Debian package overrides the plugindir with the following
`--with-plugindir=/usr/lib/$(DEB_HOST_MULTIARCH)/alsa-lib`
I was not aware that the plugindir was an available argument to the config script.

@infirit 
Yes! Looks like pkg-config has the right path to the alsa-lib

Arkq, 2017-12-07:
@infirit and @mill1000 Many thanks for your help! I hope, that there will be no more issues regarding ALSA modules not being found :)

closed 2017-12-07 by Arkq

mill1000, 2017-12-07:
Aweome. Fresh build worked like a charm on Raspbian Stretch

commit 0ef747e5 (2017-12-07): Get ALSA libdir using pkg-config

Fixes #31