arkq/bluez-alsa #32: bluez-alsa as audio sink

humarf, 2017-02-11:
Hello,

It's more a question than a real issue. I tried to configure my computer to work as audio sink. After having paired computer and mobile I could stream audio to the computer - great. It took a while until I simply tried the `arecord ... |aplay` construct because I was confused of not seeing the bluealsa device in a `arecord -L`. But it works anyway. So my question is: Alsatools not showing the device in their lists (tested aplay, arecord and alsamixer) is this an expected behavior of bluez-alsa?

Thanks and best regards
humarf

Arkq, 2017-02-12:
Hi,

Yes, this behaviour is correct - however might be annoying :). BlueALSA uses software PCM plugin, which is not listed in the kernel ALSA module, so aplay or arecord does not list BlueALSA as an audio card (BlueALSA is not an audio card, it is only a PCM device - a software one). However, one might use ALSA configuration file (e.g. ~/.asoundrc) to mark PCM device to be shown in the aplay listing, e.g.:

```
pcm.myname {

       [ .... PCM configuration .... ]

        hint {
                show on
                description "My PCM device"
        }
}
```

PS.
Since a lot of ppl asks about the PC-based Bluetooth audio sink, I've just created simple player for BlueALSA. Instead of `arecord ... |aplay`, you might use `bluealsa-aplay XX:XX:XX:XX:XX:XX`. Right now it is only a proof of concept player, but with time it might get better :D.

cookj273, 2017-02-13:
So if i am using bluealsa-aplay and getting tons of underruns, what would the problem/solution?  Any idea?

Arkq, 2017-02-14:
@cookj273, the problem is (I think) related to this #27 issue. Currently I'm working on in, so stay tuned.

roizcorp, 2017-02-14:
Hi,
relating to the bluealsa-aplay....ummm I've just reinstalled bluealsa from git and I cannot find such a process `bluealsa-aplay` to launch

Thanks for this

humarf, 2017-02-14:
Did the same on my system. Here is bluealsa-aplay  installed in `/usr/bin`. In git repository it's in `utils`.

roizcorp, 2017-02-14:
yes, apparently I update the local git wrong so I erased, cloned and compiled it again
I can see it now

sorry

cookj273, 2017-02-14:
@Arkq alright thanks for the info!  Thanks for all the great work by the way : ).  I do have some enforced period_size on my output pcm device so sounds like maybe that could be the issue.  I will play around some : )

humarf, 2017-02-15:
Hi Arkq,

Thanks for your advice. Tool bluealsa-aplay is working good.

There is one thing I am wondering about and I am totally unsure if this is a bug. Good news is that I got `bluealsa-aplay` working on a first generation raspberry pi when applying hints from issue #21. Output of `bluealsa-aplay` says that playback is 16bitLE and 44,1kHz. When I try the same thing on the raspberry pi using `arecord .. | aplay` construct it does not work using preset `-f cd` in `arecord`. Instead of this (found out by trial and error) I have to use `-f dat` that is nearly the same but using 48kHz. On my PC preset `-f cd` was working fine. Do you have any idea what this could be?

Arkq, 2017-02-15:
The available sampling rate depends on the "bluealsa ALSA configuration" (I'll just explain).

Bluealsa ALSA plugin can be used with the `plug` plugin, or without. If one is using bluealsa without the `plug` plugin, then one has to set correct number of channels and sampling rate - correct values are enforced by the connected BT device. If one is using `plug` plugin, then the correction is done by this plugin (using host CPU). On the other hand, bluealsa-aplay does not perform correction for the incoming signal (analogical to the `arecord`), but uses correct values (for channels and sampling) automatically. Than, this signal is written to the desired output PCM device (which might be a `plug` plugin - so the signal will be corrected if needed).

```
bluealsa-server -> bluealsa -> [plug] -> [plug] -> hw
-------------------------------------    ------------
              arecord                        aplay

bluealsa-server -> [plug] -> hw
-------------------------------
      bluealsa-aplay

```

humarf, 2017-02-15:
Not sure that I get you right. Command that I issued was `arecord -f cd -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp | aplay`. From what I understood preset cd is a shortcut for 2 channels, 16bitLE and 44,1kHz. In my command I did not refer to `plug` plugin. What am I doing wrong? Do you mean that my phone in fact sends data at 48kHz and `bluealsa-play` corrects this automatically ?

Arkq, 2017-02-15:
It's most likely. 44.1kHz and 48kHz are mandatory for A2DP sink (in this case bluealsa) and at least one of them is mandatory for source (phone). Most phones support both. So if 44.1kHz is not enforced in the bluealsa (via the command line argument), 48kHz will be selected (selection is performed by bluealsa).

humarf, 2017-02-15:
I think I understand now a bit better. I'm still a bit curious why this did not come up when doing the same on thing on my PC. However when I get you right there is a way to enforce sampling rate. I presume I have to add something here `bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp`(?) What parameter do I have to apply, please?

Arkq, 2017-02-15:
You have to pass `--a2dp-force-audio-cd` argument to the bluealsa server. Note, that if the source does not provide 44.1kHz sampling rate, the 48kHz will be used instead. So this option actually means "force if available" :).

```
$ bluealsa -h
usage: bluealsa [OPTION]...

  -h, --help		print this help and exit
  -i, --device=hciX	HCI device to use
  --disable-a2dp	disable A2DP support
  --disable-hsp		disable HSP support
  --disable-hfp		disable HFP support
  --a2dp-force-mono	force monophonic sound
  --a2dp-force-audio-cd	force 44.1 kHz sampling
  --a2dp-volume		control volume natively
  --aac-afterburner	enable afterburner
  --aac-vbr-mode=NB	set VBR mode to NB
```

cookj273, 2017-05-02:
@Arkq Just wanted to let you know the issue i was having with underruns when using bluealsa as an alsasrc was being caused by bluetooth scanning being enabled.  Once i disabled scanning no more issues, so my bluetooth just couldn't handle the throughput of the both the scan and the recording.

Arkq, 2017-05-02:
OK, thanks for letting me know. I did not think that scanning is so expensive. I'm experiencing very similar issue while streaming to more than one sink.

infirit, 2017-05-02:
It is, with my reasonable recent BT4 usb dongle scanning makes my mouse misbehave.

leandrotlz, 2017-09-02:
I installed Raspbian Stretch today and both bluealsa-aplay and arecord|aplay result in severe underruns that make it pretty much unusable for sound. Bluetooth scanning is disabled on my phone location settings, and neither the phone nor the pi are in discoverable mode. Any troubleshooting steps to try to figure this out would be appreciated.

Arkq, 2017-09-02:
@leandrotlz check the CPU usage on the machine where the bluealsa is running, and post the results here. Check which codec was selected: SBC or AAC. If AAC, disable it during compilation (recently I've experienced some strange behaviour with the AAC while streaming from Android).

Compile bluez-alsa with the debug enabled `--enable-debug` and post here logs from the `bluealsa` and `bluealsa-aplay -vv`.

leandrotlz, 2017-09-03:
@Arkq CPU usage for bluealsa-aplay is 0% to 1% in the task manager. Overall CPU usage for the system is likewise minimal while testing this.

I don't know how to check what codec is selected, please advise with a command.

That said, I used `../configure --disable-aac --enable-debug` so if it was AAC before, it shouldn't be now. The problem persists.

I don't know where bluealsa stores logs, please advise. Trying to run bluealsa results in the message "bluealsa: Couldn't initialize controller thread: Address already in use" which I assume is normal because the server is running.

Console output from bluealsa-aplay:

`pi@arachnos:~ $ bluealsa-aplay -vv F0:D7:AA:C1:24:5C
Selected configuration:
  HCI device: hci0
  PCM device: default
  PCM buffer time: 500000 us
  PCM period time: 100000 us
  Bluetooth device(s): F0:D7:AA:C1:24:5C
  Profile: A2DP
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:125: Subscribing for notifications: true
bluealsa-aplay: ../../utils/aplay.c:458: Starting main loop
bluealsa-aplay: ../../utils/aplay.c:480: Fetching available transports
bluealsa-aplay: ../../utils/aplay.c:537: Creating PCM worker F0:D7:AA:C1:24:5C
Used configuration for F0:D7:AA:C1:24:5C:
  PCM buffer time: 500000 us (88200 bytes)
  PCM period time: 100045 us (17648 bytes)
  Sampling rate: 44100 Hz
  Channels: 2
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:302: Requesting PCM open for F0:D7:AA:C1:24:5C
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:321: Opening PCM FIFO (mode: RO): /var/run/bluealsa/hci0-F0:D7:AA:C1:24:5C-1-1
bluealsa-aplay: ../../utils/aplay.c:253: Starting PCM loop
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
bluealsa-aplay: ../../utils/aplay.c:286: An underrun has occurred
`

Arkq, 2017-09-04:
@leandrotlz there is separate issue #60 where this problem will be tracked.

> I don't know where bluealsa stores logs, please advise. Trying to run bluealsa results in the message "bluealsa: Couldn't initialize controller thread: Address already in use" which I assume is normal because the server is running.

Try the `pkill` hack described in the #60 and please post everything related to the "jittery audio" in there as well.

Arkq, 2019-11-25:
Closing stale issue.

closed 2019-11-25 by Arkq