arkq/bluez-alsa #34: bluez-alsa with Volumio2

phweyland, 2017-02-14:
Hi Arkq
I'm not sure this is right place but I've not seen any other way for asking questions...
I've installed Volumio2 on Raspberry PI3 + Hifiberry.
I wanted to add the bluetooth renderer feature.
I've added bluetooth to the system and succeeded in connecting some bluetooth devices.
Then I've tried to installed bluez-alsa.
Configure told me which package weres still missing and I've added alsa-lib.
Now Configure tells me Bluez >= 5.0 is missing. But Bluez is installed (along with bluetooth), version 5.4 if I'm not mistaken... As a newbie, I'm stuck.
Do you know if bluez-alsa is compatible with Volumio 2 ? Could you indicate any stuff or information which could help me making progress ?
Thanks in advance
Philippe

humarf, 2017-02-15:
I think it's a distro dependent question. I'm on Archlinux and had a similar problem. Header files configure is looking for were located in package bluez-libs not in package bluez. This had to be installed too. Don't know on what distro volumio is based on but maybe there is a similar package.

phweyland, 2017-02-21:
Volumio is based on http://archive.raspberrypi.org/debian/.
I've tried with the Raspbian distribution ( http://archive.raspbian.org/raspbian/). Bluez is already installed (version 5.23) but not recognized either by configure (No package 'bluez' found ). I tried to install again bluez from http://www.bluez.org/. Bluez seems to work but remains not visible for bluez-alsa configure.
I've not found the package bluez-libs for bluez>5.0. The others seem to be deprecated.

infirit, 2017-02-21:
You probably need ``libbluetooth-dev``.

phweyland, 2017-02-24:
Thank you Infirit, that did it.
I've now installed bluez-alsa on Raspbian with error I think. And MPD & upmpdcli still work.
I've now to make the BT connexion work...

Arkq, 2017-02-25:
I have no experience with RP, nor Volumio, but I guess, that you have to add (somehow) bluealsa to the "runlevel" you are using (in order to make it start automatically during the OS boot). There is no startup script provided within the bluez-alsa repo, though (there is too much distributions to provide startup configuration for them all).

phweyland, 2017-02-26:
Thank you for the answer.
So I have to try to start bluealsa first. I get this:
$  bluealsa
bluealsa: error while loading shared libraries: libsbc.so.1: cannot open shared object file: No such file or directory
This library seems to exist:
$ sudo find -name "libsbc.so.1"
./usr/local/lib/libsbc.so.1

infirit, 2017-02-26:
>This library seems to exist:
>$ sudo find -name "libsbc.so.1"
>./usr/local/lib/libsbc.so.1

Yes but shared libs or not looked for in ``/usr/local`` by default. You need to tell the lib loader where to find them by exporting ```LD_LIBRARY_PATH``` before running ``bluealsa``. For example,

```
export LD_LIBRARY_PATH="/usr/local/lib/:$LD_LIBRARY_PATH"\
/usr/local/bin/bluealsa
```

Another option is to add the path to ``/etc/ld.so.conf``` or if available create a conf file in ```/etc/ld.so.conf.d/``` which contains also contains the path.

However first try from the command line in a terminal.

humarf, 2017-02-27:
Did you install sbc from source? If yes, this might cause the problem. You could install it from repository - I don't use raspbian but I think package is called `libsbc1` there. If you want to stick with sbc from source you could rebuild using `configure --prefix=/usr`. Then `make install` would copy libraries to `/usr/lib`.

phweyland, 2017-02-27:
Thank you for the help. I've made good progress. With:

> export LD_LIBRARY_PATH="/usr/local/lib/:$LD_LIBRARY_PATH"
> sudo bluealsa

On another ssh session I can connect some devices like an iphone (bluetoothctl)
One the last session I launch:

> bluealsa-aplay XX:XX:XX:XX:XX:XX

and I get the sound on the local sound device (hifiberry). Great !

Some issues or questions.
On the bluealsa terminal I get some messages (several of each one):

> bluealsa: SBC decoding error: No such file or directory
> bluealsa: SBC decoding error: No such process
> bluealsa: Couldn't release transport: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist

The command `sudo systemctl status -l bluetooth` gives the following error after having connected the bluetooth device:

> raspberrypi bluetoothd[586]: Invalid folder length
> raspberrypi bluetoothd[586]: Unable to get io data for Hands-Free unit: getpeername: Transport endpoint is not connected (107)

The sound is very choppy and the bluetooth connexion fails after 10 or 20 seconds (despite I had one case with clear sound and stable bluetooth connexion).

Trying to connect an android device, the sound is horrible and the cpu hangs after few seconds. I've read that Android v6 upsamples (and that's not configurable). That may not help.

I've also tried to manage the volume with the command (but I've not understood what control name is):

> amixer -D bluealsa sset 'my iphone name' 70%

and I've got the message:

> ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'defaults.bluealsa.interface'
> ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
> ALSA lib conf.c:4727:(snd_config_expand) Args evaluate error: No such file or directory
> ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL bluealsa
> amixer: Mixer attach bluealsa error: No such file or directory

Arkq, 2017-02-28:
> On the bluealsa terminal I get some messages (several of each one):
>    bluealsa: SBC decoding error: No such file or directory

It looks like the device has been disconnected.... strange, though.

> The sound is very choppy

Try to use current master snapshot (bluealsa-aplay). I've just added [0.5s buffering](https://github.com/Arkq/bluez-alsa/blob/76d928e8b667af3a6fea0bc5c5e132c37a60e12f/utils/aplay.c#L56). If it's not enough you might try to increase it. It's not a silver bullet, though. Note, that if you stop the audio (on the phone), and then start it again (without terminating bluealsa-aplay), you will see the "An underrun has occurred" message - this is OK (you've stopped audio, so the underrun has indeed occured). You should not see this message during the playback, though.

> I've not understood what control name is.

Use `amixer -D bluealsa`. This will list available controls, e.g.

```
$ amixer -D bluealsa
Simple mixer control 'MyPhone - A2DP',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 127
  Front Left: Capture 127 [100%] [on]
  Front Right: Capture 127 [100%] [on]
```

then use:

   $ amixer -D bluealsa sset 'MyPhone - A2DP' 70%

> and I've got the message:
>    ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'defaults.bluealsa.interface'

Current master should fix it as well - remember to `make install` it. The "problem" was, that in the ALSA configuration file, there was no defaults for variable definitions. So in order to work, one had to specify them on the command line, or in the local asoundrc file. Now, I've added some sane default values. For most cases it should work right away.

phweyland, 2017-03-01:
I've tried the `apt-get install libsbc1` but bluealsa `configure` doesn't find SBC. SO I've reinstalled it from git with `configure --prefix=/usr`. I can now start bluealsa with error not exporting the Library.

I assume that `Current Master` means that I reinstall bluealsa from git: https://github.com/Arkq/bluez-alsa.git.

Choppy sound when I start to play. I get some messages on the bluealsa terminal like:
```
bluealsa: Missing RTP packet: 985 != 984
bluealsa: Missing RTP packet: 997 != 995
bluealsa: Missing RTP packet: 1002 != 1001
```
Generally, it's enough to stop, wait a little while, and start again to get a stable good sound and the above messages stop.

`$ amixer -D bluealsa` gives me the message:
```
ALSA lib dlmisc.c:252:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/arm-linux-gnueabihf/alsa-lib/libasound_module_ctl_bluealsa.so
amixer: Mixer attach bluealsa error: No such device or address
```
I set the asoundrc file:
```
$ sudo ~/.asoundrc
defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "68:FB:7E:24:25:52"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000
```
but I get the same message. Idem for `aplay -D bluealsa`.  The file `libasound_module_ctl_bluealsa.so` is not there:
```
$ ls /usr/lib/arm-linux-gnueabihf/alsa-lib/
smixer
```

Arkq, 2017-03-02:
> I've tried the apt-get install libsbc1 but bluealsa configure doesn't find SBC

Use `apt-cache search sbc`, and if there is a dev package for sbc (e.g. libsbc-dev) install it.

> Choppy sound when I start to play. I get some messages on the bluealsa terminal like:
bluealsa: Missing RTP packet: 985 != 984

It might be a performance/transmission issue. It seems that your host is not able to receive all packets from the Bluetooth device. It also might mean, that the connection link is weak - packets are lost (e.g. Bluetooth devices are too far apart). In order to better understand what causes this package lost some debugging would be required, and not particularly with the bluealsa itself.

> Cannot open shared library /usr/lib/arm-linux-gnueabihf/als [...] _bluealsa.so

Use `--with-alsaplugindir=/usr/lib/arm-linux-gnueabihf/alsa-lib` during bluez-alsa configuration.

phweyland, 2017-03-02:
Thank you for your patience Arkq ! :-)
With the switch on blue-alsa configuration amixer works and the glitches are away.
I see the that the iPhone controls the volume:
```
$ amixer -D bluealsa
Simple mixer control 'Philippe iPhone - A2DP',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 127
  Front Left: Capture 44 [35%] [on]
  Front Right: Capture 44 [35%] [on]
```
But this volume is not applied, except for volume = 0. For MPD (/etc/mpd.conf), to use the hardware control of the sound card, I have to set this:
```
mixer_control   "Digital"
audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional
#       mixer_type      "hardware"      # optional
#       mixer_device    "default"       # optional
        mixer_control   "Digital"       # optional
#       mixer_index     "0"             # optional
}
```
Is there any equivalent setting for bluealsa ?

I've tried to launch bluealsa as a service:
```
sudo systemctl start bluealsa
Failed to start bluealsa.service: Unit bluealsa.service failed to load: No such file or directory.
```
Is that possible ? How should I set the bluealsa.service file ? Would you have an example ?

infirit, 2017-03-02:
>I've tried to launch bluealsa as a service:

bluealsa does not provide a init or service file, it is not really a daemon. However I have not experienced any problems running as a service on gentoo as long as you make sure you keep bluealsa requirements in mind.

I do not know (and have no intention of learn) how unit files work. But you may get the necessary inspiration from the gentoo init file, https://gitweb.gentoo.org/repo/gentoo.git/tree/media-sound/bluez-alsa/files/bluealsa-init.d

infirit, 2017-03-02:
Oh, arch linux has a service file for it, https://aur.archlinux.org/packages/bluez-alsa-git/

phweyland, 2017-03-03:
Thanks Infirit, the arch linux service file works !

phweyland, 2017-03-03:
> Use apt-cache search sbc, and if there is a dev package for sbc (e.g. libsbc-dev) install it.

That works.

> $ sudo apt-get install libsbc1 libsbc-dev

Bluealsa configuration accepts this.

phweyland, 2017-03-03:
Note: I can switch between `iPhone-BT-bluealsa `and `bubbleupnp-upmpdcli-MPD` back and forth. When I'm back to bubbleupnp I can still play and control volume but I've lost the feedback (start/hold and volume). As if the Alsa feedback was kept by bluealsa (?)

phweyland, 2017-03-04:
Trying to check if BT USB dongle could be more performant than the onboard Raspberry BT I'm stuck  here...

> $ amixer -D bluealsa
> ALSA lib ../../../src/asound/bluealsa-ctl.c:570:(_snd_ctl_bluealsa_open) BlueALSA connection failed: No such file or directory
> amixer: Mixer attach bluealsa error: No such file or directory

with bluealsa running and

> $ bluealsa-aplay 68:FB:7E:24:25:52
> bluealsa-aplay: BlueALSA connection failed: No such file or directory

While the device is connected.

>  $ bluetoothctl
> [NEW] Controller 5C:F3:70:7F:A5:1E raspberrypi #1 [default]
> [NEW] Device 68:FB:7E:24:25:52 Philippe iPhone
> [NEW] Controller B8:27:EB:38:C1:5B raspberrypi

What I did miss ?

infirit, 2017-03-04:
Did you make sure the usb adapter is working? And did you update your asoundrc file with the name of the new adapter?

You can use hcitool to list the adapters ```hcitool dev```

phweyland, 2017-03-05:
> You can use hcitool to list the adapters
 ```
$ hcitool dev
Devices:
        hci1    B8:27:EB:38:C1:5B
        hci0    5C:F3:70:7F:A5:1E
$ hcitool inq
Inquiring ...
        68:FB:7E:24:25:52       clock offset: 0x5241    class: 0x7a020c
$ hcitool con
Connections:
        > ACL 68:FB:7E:24:25:52 handle 11 state 1 lm MASTER AUTH ENCRYPT
```
What could I do to better check ?
> did you update your asoundrc file with the name of the new adapter?

... I've no asoundrc file. I thought it was set to configure Alsa to send the audio stream to BT, while I'm using it the other way (BT -> Alsa). However if I'm not mistaken, in the former case hci0 represented the on board BT ("raspberry") while in the new config hci0 represents the BT adapter ("raspberry #1"). So the bluealsa default settings should still work. Is that correct ?

phweyland, 2017-03-09:
> I can switch between iPhone-BT-bluealsa and bubbleupnp-upmpdcli-MPD back and forth. When I'm back to bubbleupnp I can still play and control volume but I've lost the feedback (start/hold and volume).

If I restart bubbleupnp the feedback is back. A bubbleupnp issue.

phweyland, 2017-03-10:
Based on an [pulseaudio example](https://gist.github.com/oleq/24e09112b07464acbda1) I've setup the logic to auto start and stop bluealsa-aplay on connection (and deconnection) of (already trusted) bluetooth source (telephone). I'll share it if anybody wants.

The main thing which is still missing is the control of volume. bluealsa-aplay already controls volume on PCM plug bluealsa. When I change the volume on the telephone, amixer shows the change. ([35%] or any other value):

> $ amixer -D bluealsa
> Simple mixer control 'Philippe iPhone - A2DP',0
>   Capabilities: cvolume cswitch
>   Capture channels: Front Left - Front Right
>   Limits: Capture 0 - 127
>   Front Left: Capture 44 [35%] [on]
>   Front Right: Capture 44 [35%] [on]

but the actual default sound device doesn't receive it:

> $ amixer sget 'Digital'
> Simple mixer control 'Digital',0
>   Capabilities: pvolume pswitch
>   Playback channels: Front Left - Front Right
>   Limits: Playback 0 - 207
>   Mono:
>   Front Left: Playback 207 [100%] [0.00dB] [on]
>   Front Right: Playback 207 [100%] [0.00dB] [on]

Any idea to complete the mapping ?

Arkq, 2017-03-11:
> The main thing which is still missing is the control of volume.

Try the latest master.

Long story short, my phone does not interact very well with BlueZ (volume keys do not work), so I forgot to implement it... x_x

phweyland, 2017-03-11:
> Try the latest master.

Fantastic ! it has the alsa logarithm behaviour but that works. 
Thank you again Arkq

Arkq, 2017-03-11:
In the future, it might be possible, however not right now. One key component is missing - event notifications from the bluealsa server, so it would be possible to notify client (e.g. bluealsa-aplay), that the volume has been changed. It is not implemented yet, and I have scheduled it for future release (it's not a critical component right now).

phweyland, 2017-03-12:
Back to the intial case, I've reproduced the raspbian installation on Volumio distrib.
I've described the corresponding steps there : https://volumio.org/forum/volumio-bluez-alsa-a2dp-bluetooth-support-t6130.html

Caravanvolumio, 2017-04-18:
Hi there
I have installed Volumio 2 on A Pi Zero WiFi, With HiFiBerry Dac Zero
I followed pweylands build
https://volumio.org/forum/volumio-bluez-alsa-a2dp-bluetooth-support-t6130.html
and can play music from my iPad to the PI wired from the DAC to the Xplorer.
However I would like to play audio from volumio to the Xplorer using only bluetooth.
I can then take the speaker outside the Van
I have spoken to pweyland about this and he kindly suggested the path I want to follow is more like:
Volumio -> alsa -> bluez 5 -> Bluetooth controller (on pi) -> Bluetooth Headset.
On https://github.com/Arkq/bluez-alsa i would find instruction for that.

I have tried to follow the Arkq build by doing the following
Install Dependencies
sudo apt-get update
sudo apt-get install libasound2-dev
sudo apt-get install dh-autoreconf
sudo apt-get install libortp-dev
sudo apt-get install bluez pi-bluetooth bluez-tools libbluetooth-dev
sudo apt-get install libusb-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev
sudo apt-get install libsbc1 libsbc-dev

Plus – for Arku –Bluez-Alsa
sudo apt-get install libbsd-dev
sudo apt-get install libncurses5-dev

Build Bluez-Alsa
sudo git clone https://github.com/Arkq/bluez-alsa.git
cd bluez-alsa
Go to root
sudo su

mkdir -p m4

autoreconf –install
mkdir build && cd build
../configure --enable-debug
make && make install

mkdir -p /var/run/bluealsa
chown volumio /var/run/bluealsa

Using bluetoothctl, I can get the StudioSeries Xplorer paired & trusted, but not connected.
# info returns
Device 00:01:01:16:80:C7
	Name: Xplorer
	Alias: Xplorer
	Class: 0x240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

# connect returns
Attempting to connect to 00:01:01:16:80:C7
Failed to connect: org.bluez.Error.Failed
Can either of you suggest what I should do next?
Thanks for looking

popsUlfr, 2017-04-29:
@Caravanvolumio : maybe the dbus policy is missing?
https://wiki.archlinux.org/index.php/Bluetooth_headset#Headset_via_Bluez5.2Fbluez-alsa

`/etc/dbus-1/system.d/bluetooth.conf`
```
<policy user="bluealsa">
  <allow send_destination="org.bluez"/>
</policy>
```
Just before the closing `</busconfig>`

So I got the bluetooth speaker working. I can connect with my phone and play audio but it is so soooo choppy :(
I'm using the latest git version of bluez-alsa, is there any way make it better?

Arkq, 2021-01-16:
Closing stale issue.

PS.
With current master it is possible to disable software volume by using `bluealsa --a2dp-volume`.

closed 2021-01-16 by Arkq