arkq/bluez-alsa #35: Music Player Daemon and bluealsa sink

skidoo23, 2017-02-19:
Try to play music from my mobile to MPD. Sound is ok. But high load on a single cpu core. Debian Strech on x64,  ALSA 1.1.3, MPD and bluealsa from git.

How to reproduce

bluealsa --a2dp-force-audio-cd
bluetoothctl connect 11:22:33:44:55:66
mpc add alsa://bluealsa:HCI=hci0,DEV=11:22:33:44:55:66,PROFILE=a2dp

Please note: https://forum.musicpd.org/viewtopic.php?f=7&t=4055#p6617

> I traced the problem to the fact that the bluealsa PCM *requires* the client code to call snd_pcm_poll_descriptors_revents() after poll() in order to clear events from an internal file descriptor. Without this function, all subsequent calls to poll() return immediately resulting in a busy loop in the I/O thread. Unfortunately I find the alsa documentation ambiguous on whether snd_pcm_poll_descriptors_revents() is meant to be mandatory so it is arguable whether this is a bug in bluealsa or mpd.

Any help?

kuikka, 2017-02-22:
If the application is doing select() or poll() manually it _has_ to call snd_pcm_poll_descriptors_revents() after poll/select to give alsa the chance to serve the socket(s).

That is, unless it is using some of the API that does that internally already, such as snd_pcm_wait().

kuikka, 2017-02-22:
Well, turns out there is more. snd_pcm_writei() will perform the wait _unless_ the audio device is in non-blocking mode, which is the case with MPD, is likely the cause here.

skidoo23, 2017-02-25:
MPDs input plugin opens ALSA caputure device with SND_PCM_STREAM_CAPTURE, SND_PCM_NONBLOCK options. Using snd_pcm_readi with bluealsa results most of the time in -EAGAIN. Using snd_pcm_readi with Onboard or USB capture device returns positive values only.

Arkq, 2017-02-25:
Hi, sorry for such a late response, I've been really busy lately...

It seems, that this issue is caused by the MPD ALSA input plugin implementation. I've reproduced this issue, however it seems that the bluealsa_poll_revents() function is not called at all - and it HAS TO be! Not calling this function (essentially snd_pcm_poll_descriptors_revents()) might cause the effect we can see - poll will not block any more.

> Unfortunately I find the alsa documentation ambiguous on whether snd_pcm_poll_descriptors_revents() is meant to be mandatory so it is arguable whether this is a bug in bluealsa or mpd.

If one is using poll/select in the code, then the call to the snd_pcm_poll_descriptors_revents() is mandatory. It might work with other ALSA input devices, because it is more likely, that other input devices do not perform anything fancy inside the descriptors_revents() callback - and bluealsa does...

Arkq, 2019-03-11:
Issue moved to #174

closed 2019-03-11 by Arkq