arkq/bluez-alsa #36: Unknown field delay

skidoo23, 2017-02-27:
arecord -D bluealsa:HCI=hci0,DEV=11:22:33:44:55:66,PROFILE=a2dp
ALSA lib ../../../src/asound/bluealsa-pcm.c:660:(_snd_pcm_bluealsa_open) Unknown field delay

Any help?

Arkq, 2017-02-27:
Hi,

Of course, I'll help. It's caused by my recent (last) commit. :)

It seems, that you've installed bluez-alsa, but somehow not every file was copied during the installation process. Make sure, that the file /usr/lib/alsa-lib/libasound_module_pcm_bluealsa.so is up to date.

skidoo23, 2017-02-27:
Lemon fresh with enzymes :)

-rw-r--r--  1 root root 102814 Feb 27 22:09 libasound_module_ctl_bluealsa.a
-rwxr-xr-x  1 root root   1102 Feb 27 22:09 libasound_module_ctl_bluealsa.la
-rwxr-xr-x  1 root root  59888 Feb 27 22:09 libasound_module_ctl_bluealsa.so
-rw-r--r--  1 root root 112434 Feb 27 22:09 libasound_module_pcm_bluealsa.a
-rwxr-xr-x  1 root root   1114 Feb 27 22:09 libasound_module_pcm_bluealsa.la
-rwxr-xr-x  1 root root  68784 Feb 27 22:09 libasound_module_pcm_bluealsa.so

Arkq, 2017-02-27:
So, if you would have been using the latest master branch (without custom patches) and these files were up to date (freshly compiled), this error would have been on the line 529, not 660.

Line which throws error:
https://github.com/Arkq/bluez-alsa/blob/master/src/asound/bluealsa-pcm.c#L529

So a quick search shows, that your file was compiled from this commit (it can be even older, though):
https://github.com/Arkq/bluez-alsa/blob/a7faa576d3ec23a4d1d48bd3c132d67b3c1f4eed/src/asound/bluealsa-pcm.c#L660

The `delay` field was added in the latest commit. And it seems, you are using the ALSA configuration file from this commit, because before that, there was no `delay` field.
https://github.com/Arkq/bluez-alsa/blob/e7a4c40e374b4e7419cd8f5bcafc17bc2e518969/src/asound/20-bluealsa.conf#L56

Sooooooooo, something is VERY not right :D

Try to use a fresh git clone. Compile it and install, then we'll see.

skidoo23, 2017-02-27:
D'oh! Some old stuff in library search path. Sorry for bothering you.

Arkq, 2017-02-28:
OK, so is this case closed?

raftopoulos, 2017-02-28:
Yes. Please close case.

closed 2017-02-28 by Arkq