arkq/bluez-alsa #38: buffer_size and period_size options for bluealsa-aplay

guillaumezin, 2017-03-07:
Hello,

I'm using bluez-alsa to use my Raspberry PI 3 with sound card as headless bluetooth speaker for my videprojector.

Currently, I redirect sound of bluetooth to sound card with a command combination looking like  arecord ... | aplay ... --buffer-time=100000

I tried to replace it with bluealsa-aplay but the buffer is bigger and I feel latency between video and sound. Would it be possible to add option like --buffer-time / --buffer-size / --period-time / --period-size to bluealsa-aplay ?

Or do you know how to put this value on hw PCM with ALSA ? I see in the code that you inherits value from PCM parameters, but I didn't manage to change them in my /etc/asound.conf.

Thank you.

Guillaume

closed 2017-03-07 by Arkq

Arkq, 2017-03-07:
I was to lazy to make these parameters configurable via command line :)

Also, I'd like to ask you for one thing. Can you test the CPU usage (on Raspberry PI of course) for the arecord|aplay and bluealsa-aplay, and post the results here. Because, I wonder, if there is any detectable performance boost. On PC the difference is negligible, at least on mine, so I don't know if bluealsa-aplay is worth the effort :|.

guillaumezin, 2017-03-07:
Hello,

Thank you for your quick modifications.

On Raspbian Jessie (32 bits), both methods give me 6% CPU with top command:

arecord -f cd -D bluealsa:HCI=hci0,DEV=$NUM,PROFILE=a2dp | aplay -D hw:CARD=sndrpihifiberry,DEV=0 --buffer-time=100000

/usr/bin/bluealsa-aplay --hci=hci0 --pcm=hw:CARD=sndrpihifiberry,DEV=0 --pcm-buffer-time=100000 ${NUM}

commit b7ba1805 (2017-03-07): Specify buffer and period time via command line

Fixes #38