arkq/bluez-alsa #39: Address-less BT speaker sink & volume control via phone

George-ou812, 2017-03-10:
Hello,
Using an Arm cross-compiled bluesalsa build things are working on my platform as expected. However as it is a BT speaker changing the address to suit the client sending music each time is not practical. 
1) Is there a way to automatically connect to any address? I need to get this to act the same as a normal BT speaker, pair, connect from any device => render music without any intervention on the speaker side.

2) This may be a long-shot but is there any way for the volume to be adjusted via the phone/tablet?
Currently I'm using: arecord -f cd -D bluealsa | aplay.

Cheers.

Arkq, 2017-03-10:
Hi,

> Is there a way to automatically connect to any address?

Currently it is not possible. It has crossed my mind, though :) It will be possible (in the future) for sure, with the bluealsa-aplay.

> Is there any way for the volume to be adjusted via the phone/tablet?

To be honest, I don't know (I'm not able to check it right now). I think, that it should be possible. If not via the bluealsa itself, it could be done via some scripting. It is possible, that upon connecting BT device, BlueZ registers new input device (if BT device supports such a feature), which could be used to control volume.

George-ou812, 2017-03-10:
Rats! Would have been perfect solution and finally rid of pa which eats 32% of my resources. I imagine a lot of the RPI people would like this feature too.

Arkq, 2017-03-11:
Checkout fresh master snapshot. If you are lucky (your phone/tablet is correctly recognized by the BlueZ) you should be able to control volume.

George-ou812, 2017-03-13:
Just on the DBus monitoring idea, do you have the DBus interface address org.x.x ?
Any additional information of the specifics of listening for the A2DP connection establishment?

infirit, 2017-03-13:
You can, the name is ``org.bluez`` then you have several options. Either watch for new devices with interface ``org.bluez.Device``` or proxy ``org.freedesktop.DBus.ObjectManager`` on path ``/`` and listen to the ``InterfaceAdded`` signal. Gio (GLib) has nice high level classes for this which we use in blueman.

Then inspect the ```UUIDs``` property on interface ```org.bluez.Device1`` and look for the uuid that corresponds to A2DP.

George-ou812, 2017-03-13:
Yea, could add that to simple-agent and run bluealsa-play with the appropriate connected mac perhaps.

infirit, 2017-03-13:
``simple-agent`` is bluez4 and replaced with ``bluetoothctl`` in version 5.

If you are up for some hacking, you could create your own agent and have it handle pin codes for a select set of macs. For inspiration look https://github.com/blueman-project/blueman/blob/master/blueman/main/applet/BluezAgent.py but skip making it the default.

George-ou812, 2017-03-14:
Bluez5-5.43 is shipped with simple-agent and works well with pulseaudio. On my embedded Linux I can automatically pair, trust ( I pin functionality added ) any mobile / tablet and play music exactly like an off-the-shelf BT speaker does with no user intervention speaker side. My issues are pulse eats 32% of my resources and pulse also plays games with my other music renders that need non-pcm access to the sound card, hence why blue-alsa is great. 
Look at this all last night I'm wondering if it's going to be possible to do the same with blue-alsa?
For example, blue-alsa needs a hci0 mac in order to start, For a BT a2dp device to show on my iphone I need bluetoothd, blue-alsa and a python agent. A chicken-n-egg situation, blue-alsa needs a hci0 mac to start and tell bluez its capabilities BUT I need to be able to change the mac to the current connected device automatically to play music. 

@infirit, my set-up already does this, I can see all the trust devices within the BT directories in my file system ( with PA ). My problem is changing blue-alsa's mac on the fly with one of those stored and still keeping blue-alsa alive so a phone/tablet can see an a2dp device in the first place. Hope this makes sense?

Arkq, 2017-03-14:
@George-ou812 You do not have to kill/restart bluealsa server, in fact it is not advised. The only thing you have to do, it is to start bluealsa-aplay (or arecord|aplay) with a correct MAC address.

George-ou812, 2017-03-14:
Yea, that's the problem! Passing the mac to bluealsa without human intervention...
If I start bluealsa my phone will find an a2dp device, connect and render via arecord. If I pair another phone I have to change the HCI device in asound.conf by hand otherwide it complains about no transport.

infirit, 2017-03-14:
>Bluez5-5.43 is shipped with simple-agent

Oh, you're right, I thought they dropped it.

Maybe you can post the asoundrc content and what you change and related error messages.

George-ou812, 2017-03-15:
Latest master build has fixed the volume on iphone4! I'll investigate using bluealsa-aplay and see if I can associate a mac dynamically somehow after connection but before "play". Does bluealsa-aplay actually render directly to hardware or has it just hidden "arecord | aplay" internally?

George-ou812, 2017-03-15:
Excellent! Great job. This issue's done for me now. Thanks.

commit 6091611b (2017-07-12): Automatically open PCM connection

This commit provides a functionality of a Bluetooth Speaker. From now
on, it is possible to connect any BT device and play music instantly.

Fixes #39 and fixes #41
closed 2017-07-12 by Arkq