arkq/bluez-alsa #40: bluealsa-aplay - no transport: invalid argument

George-ou812, 2017-03-14:
Hi 

Based on fresh download todfay. Using:

arecord -f cd -D bluealsa | aplay
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Playing WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo

Works fine. However 
 
bluealsa-aplay E4:CE:8F:92:67:80 -d hw:0,0 -v           // or any variant of
Selected configuration:
  HCI device: hci0
  PCM device: hw:0,0
  PCM buffer time: 500000 us
  PCM period time: 100000 us
  Bluetooth device: E4:CE:8F:92:67:80
  Profile: A2DP
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:197: Getting transport for E4:CE:8F:92:67:80 type 1
bluealsa-aplay: Couldn't get BlueALSA transport: Invalid argument

E4:CE:8F:92:67:80 is the iphone I was just using with the previous command. I've also tried this command before the above, restarted things in various sequences still the same the error.     
Any idea what the "invalid" argument is?

George-ou812, 2017-03-14:
Comment BTW  "// or any variant of" , certainly not my invalid argument :)

George-ou812, 2017-03-15:
Works! Turns out I using a slightly older bluealsa daemon not the latest that came with the newer build. Out-of-tree building for you...

Arkq, 2017-03-15:
OK, I'm closing this issue then.

closed 2017-03-15 by Arkq