arkq/bluez-alsa #41: bluealsa-aplay not releasing hw audio card unless bluealsa-aplay killed

George-ou812, 2017-03-15:
Phone connected and streams song - all works well. But if the phone stops streaming or the BT has been turned off on the phone ( i.e total BT disconnection ); unless I kill bluealsa-aplay xx.. I cannot use my sound card for any other purpose. This is an issue If I choose to stream via another phone app like dnla or airplay to my hardware.

Debug: 
bluealsa-aplay E4:CE:8F:92:67:80
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa: ../../src/ctl.c:534: New client accepted: 10
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:197: Getting transport for E4:CE:8F:92:67:80 type 1
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:286: Requesting PCM open for E4:CE:8F:92:67:80
bluealsa: ../../src/ctl.c:313: PCM requested for E4:CE:8F:92:67:80 type 1 stream 1
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa: ../../src/io.c:103: Opening FIFO for writing: /var/run/bluealsa/hci0-E4:CE:8F:92:67:80-1-1
bluealsa: ../../src/io.c:105: FIFO reading endpoint is not connected yet
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:305: Opening PCM FIFO (mode: RO): /var/run/bluealsa/hci0-E4:CE:8F:92:67:80-1-1
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa-aplay: ../../utils/aplay.c:245: Starting main loop
bluealsa: ../../src/io.c:103: Opening FIFO for writing: /var/run/bluealsa/hci0-E4:CE:8F:92:67:80-1-1
bluealsa: ../../src/bluez.c:1039: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: ../../src/transport.c:601: State transition: 2 -> 0
bluealsa: ../../src/transport.c:717: Releasing transport: A2DP Sink (SBC)
bluealsa: ../../src/transport.c:745: Closing BT: 9
bluealsa: ../../src/io.c:72: Exiting IO thread
bluealsa: ../../src/bluez.c:504: Endpoint method call: org.bluez.MediaEndpoint1.ClearConfiguration()
bluealsa: ../../src/transport.c:324: Freeing transport: A2DP Sink (SBC)
bluealsa: ../../src/transport.c:844: Cleaning PCM FIFO: /var/run/bluealsa/hci0-E4:CE:8F:92:67:80-1-1
bluealsa: ../../src/transport.c:851: Closing PCM: 11

At this point the streamed song has stopped and the phone has been switched off; bluealsa debug states its released / freed the sink. But as stated previously the hw:0,0  card is not released for use until bluealsa-aplay is killed. Help!

George-ou812, 2017-03-16:
OK, based on Arkq's comment "This disconnection "feature" will be possible in the future, alongside with the "real bluetooth speaker" functionality (accepting any MAC address).", ultimately the "real bluetooth speaker" is actually what I'm after and trying to do. Based on that I'll wait this #issue is closed for me. 
Thanks for your help. 
Cheers.

Arkq, 2017-03-16:
@George-ou812 Bare in mind, that the "future" means few months (not days). This project is not a "Bluetooth speaker", but more likely pulseaudio alternative for Bluetooth Audio on Linux. Hence, this feature has very low priority for me. After all, it is possible to accomplish it already with a little bit of scripting (e.g. Python code provide by @infirit).

George-ou812, 2017-03-16:
@Arkq, appreciate your timescale's, "pulseaudio alternative for Bluetooth Audio on Linux" is perfect for me, pa is taking up far too much resources on my board just for BT. Look forward to see this sometime, keep up the good work!

commit 6091611b (2017-07-12): Automatically open PCM connection

This commit provides a functionality of a Bluetooth Speaker. From now
on, it is possible to connect any BT device and play music instantly.

Fixes #39 and fixes #41
closed 2017-07-12 by Arkq