arkq/bluez-alsa #43: Record from bluetooth device(android-phone) via a2dp

mirandw, 2017-04-18:
I managed to porting bluez-alsa to my platform compiled by uclibc, and have fixed several errors.
can now play music through my bluetooth headset with **aplay** via a2dp using bluez-5.28 and alsa-lib1.0.28, but **arecord** still gives me the same error when connecting with android-phone via a2dp.

I tried to provide a full log.

My /etc/asound.conf contains:
`defaults.bluealsa.interface "hci0"`
`defaults.bluealsa.device "C4:07:2F:96:FD:4D"`
`defaults.bluealsa.profile "a2dp"`

then start bluealsa and connect android phone through bluetoothctl,
the output so far
`ctl.c:483: Starting controller loop`
 `bluez.c:699: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource`
`bluez.c:699: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink`
`main.c:218: Starting main dispatching loop`
`bluez.c:504: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()`
`bluez.c:504: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()`
`bluez.c:438: A2DP Sink (SBC) configured for device C4:07:2F:96:FD:4D`
`bluez.c:440: Configuration: channels: 1, sampling: 44100`
`transport.c:603: State transition: 0 -> 0`

amixer -D bluealsa

`Simple mixer control 'HUAWEI MT7-TL10 - A2DP',0`
 ` Capabilities: cvolume cvolume-joined cswitch cswitch-joined`
 ` Capture channels: Mono`
 ` Limits: Capture 0 - 127`
  `Mono: Capture 127 [100%] [on]`

arecord -f S16_LE -c 1 -r 44100 -D bluealsa:HCI=hci0,DEV=C4:07:2F:96:FD:4D,P
ROFILE=a2dp /media/sdcard/test.wav

`/bluez.c:1039: Signal: PropertiesChanged: org.bluez.MediaTransport1`
`transport.c:603: State transition: 0 -> 1`
`transport.c:695: New transport: 9 (MTU: R:672 W:1008)`
`bluez.c:1039: Signal: PropertiesChanged: org.bluez.MediaTransport1`
`transport.c:603: State transition: 1 -> 2`
`transport.c:95: Created new IO thread: A2DP Sink (SBC)`
`src/io.c:287: Starting IO loop: A2DP Sink (SBC)`
`src/io.c:290: io_thread_a2dp_sink_sbc`
`src/io.c:290: io_thread_a2dp_sink_sbc`
`src/io.c:307: before read`
`src/io.c:324: befor io_thread_open_pcm_write`
`io.c:86: Opening FIFO for writing: /var/run/bluealsa/hci0-C4:07:2F:96:FD:4D-1-1`
`bluealsa-pcm.c:104: Starting IO loop`
`io.c:290: io_thread_a2dp_sink_sbc`
`io.c:307: before read`
`io.c:324: befor io_thread_open_pcm_write`
`io.c:290: io_thread_a2dp_sink_sbc`
`io.c:307: before read`
`io.c:324: befor io_thread_open_pcm_write`
`io.c:290: io_thread_a2dp_sink_sbc`
`io.c:307: before read`
`io.c:324: befor io_thread_open_pcm_write`
`io.c:290: io_thread_a2dp_sink_sbc`
`io.c:307: before read`
`io.c:324: befor io_thread_open_pcm_write`
`arecord: pcm_read:2031: read error: No such device`
`bluealsa-pcm.c:249: Stopping`
`bluealsa-pcm.c:314: Freeing HW`
`ctl-client.c:348: Closing PCM for C4:07:2F:96:FD:4D`
`ctl.c:392: PCM close for C4:07:2F:96:FD:4D type 1 stream 1`
`src/transport.c:848: Cleaning PCM FIFO: /var/run/bluealsa/hci0-C4:07:2F:96:FD:4D-1-1`
`transport.c:855: Closing PCM: 12`
`bluealsa-pcm.c:265: Closing plugin`
`bluez.c:1039: Signal: PropertiesChanged: org.bluez.MediaTransport1`
`transport.c:603: State transition: 2 -> 0`
`transport.c:721: Releasing transport: A2DP Sink (SBC)`
`transport.c:749: Closing BT: 9`
`io.c:55: Exiting IO thread`

At this point test.wav is only 44Bytes. I am stopped at "`arecord: pcm_read:2031: read error: No such device`"
Can I try or post anything else to help?

mirandw, 2017-04-18:
Also I tried bluealsa-aplay, still cannot work properly, lost
bluealsa-aplay C4:07:2F:96:FD:4D --verbose

`Selected configuration:`
  `HCI device: hci0`
  `PCM device: default`
 ` PCM buffer time: 500000 us`
  `PCM period time: 100000 us`
  `Bluetooth device: C4:07:2F:96:FD:4D`
` Profile: A2DP`
`/usr/bin/bluealsa-aplay: ../../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0`
`bluealsa: ../../../src/ctl.c:544: New client accepted: 10`
`bluealsa: ../../../src/ctl.c:547: +-+-`
`/usr/bin/bluealsa-aplay: ../../../utils/../src/shared/ctl-client.c:197: Getting transport for C4:07:2F:96:FD:4D type 1`
`bluealsa: ../../../src/ctl.c:547: +-+-`
`/usr/bin/bluealsa-aplay: Couldn't set HW parameters: Invalid argument`
`bluealsa: ../../../src/ctl.c:515: Client closed connection: 10`
`bluealsa: ../../../src/ctl.c:547: +-+-`

mirandw, 2017-04-24:
I have found rootcuase of `arecord: pcm_read:2031: read error: No such device`
It is all my fault when modifying eventfd_read/eventfd_write

mirandw, 2017-04-24:
`bluealsa-aplay C4:07:2F:96:FD:4D --verbose`
This issue is because when setting pcm set_hw_params, alsalib-1.0.2 must set period-time first, then buffer-time, otherwse, `Invalid argument` appears.
This issue's done for me now. Thanks.

closed 2017-04-24 by mirandw

Arkq, 2017-04-24:
Thx for the explanation. I'll reverse these hw_params_set_* calls. :)

Arkq, 2017-04-25:
Are you sure, that the cause of the "Invalid argument" error was the order of set_hw_params calls? I've checked this with various versions of alsa-lib (as old as 0.9.8) and in every case the order is of no importance. Maybe the problem was with something else, e.g. unsupported number of channels or some other incompatibility...

I'll change a little bit error reporting, because right now this set HW params error message does not say much :/

mirandw, 2017-04-25:
sorry, I reversed the order in `arecord` built from alsa-utils, everything is ok. But indeed, after reversing order, bluealsa-aplay works fine. And the params is as expected.
`#bluealsa-aplay --hci=hci0 --pcm=hw:0,1 C4:07:2F:96:FD:4D --pcm-period-time=125000 --pcm-buffer-time=500000`
`bluealsa-aplay: channels 1  rate 44100 buffer 500000 period 125000`

Maybe some other incompatibility exists. I'll keep on debuging.

Sorry for the trouble :)