arkq/bluez-alsa #45: Bluez-alsa with Alsaequal

dagf2101, 2017-04-28:
I was trying hard this morning to make Bluez-alsa and Alsaequal to work together and I cant find a working solution.
Alsaequal is a big one for me so if you could try it out or simply explain why it cannot work I would appreciate.

dagf2101, 2017-04-28:
By the way I just made a pull request for a yocto recipe I've done if you want to merge.

Arkq, 2017-05-01:
I do not see any problems with alsaequal, though. The only problem I've got was to compile alsaequal. However, after few modifications (in the alsaequal source code, due to some incompatibility with a newer version of cups library - caps-0.9.24) I've managed to make it work.

My .asoundrc file looks like this:
```
pcm.jabra {
	type plug
	slave.pcm {
		type bluealsa
		device "1C:48:F9:9D:81:5C"
		profile "a2dp"
		delay 20000
	}
	hint {
		show on
		description "Jabra MOVE v2.3.0"
	}
}
ctl.eq {
	type equal;
}
pcm.jabra-eq {
	type equal;
	slave.pcm "jabra"
}
pcm.default-eq {
	type equal;
	slave.pcm "default"
}
```

Make sure, that you are using `plug` plugin with `bluealsa`. Otherwise, it might not work.

dagf2101, 2017-05-01:
Thanks so much, it works now, I guess my config file was wrong.
This alsa config stuff is not trivial hehe.
I am now able to open both headphones and bluetooth at the same time, both
using alsaequal.

closed 2017-05-01 by dagf2101