arkq/bluez-alsa #47: Question

G3NSVRV, 2017-06-04:
hello, I have some question about this bluealsa "device". I don't really understand why "defaults.blu..." and how I can put a nice equal in the connected bluealsa devices. I'm using alsaequal

G3NSVRV, 2017-06-04:
ctl.equal {
    type equal;
}

pcm.plugequal {
    type equal;
    # Modify the line below if you do not
    # want to use sound card 0.
    #slave.pcm "plughw:0,0";
    # by default we want to play from more sources at time:
    slave.pcm "plug:dmix";
}

# pcm.equal {
# If you do not want the equalizer to be your
# default soundcard comment the following
# line and uncomment the above line. (You can
# choose it as the output device by addressing
# it with specific apps,eg mpg123 -a equal 06.Back_In_Black.mp3)
pcm.!default {
    type plug;
    slave.pcm plugequal;
}

defaults.bluealsa {
    interface "hci0";
    device "00:00:00:00:00:00";
    profile "a2dp";
}

Arkq, 2017-06-05:
Check this issue: #45

Arkq, 2017-08-30:
I'm assuming, that this issue is resolved. If it is still valid, feel free to reopen it.

closed 2017-08-30 by Arkq