arkq/bluez-alsa #50: Bluealsa as parallel input

Vitalic66, 2017-07-08:
Hallo,

finally i managed to get your bluealsa working on my Pi3 with raspbian jessie. 

Great program.

The reason why I installed it, on raspbian wheezy with bluez4 i was able to stream music from phone via BT to Pi and also to hear music same time in kodi mediacenter. Doesn't make sense, but so i had not not restart mediacnter or bluez all the time. I guess it was because of ALSA. Now the bluez5 is set to pulse i had the problem i described.

With your program i can use ALSA again. Also the mediacenter is working with ALSA (kodi 15.2).

But at the moment i got the same problem like with pulseaudio. Only one stream is possible. I know that it works somehow because with bluez4 and ALSA it was working too.

Is there a possibility to route the stream of bluealsa to an ALSA plugin like jack ? I'd like to send both streams to this plugin and this will send the combined stream to the output.

Arkq, 2017-07-09:
Hi,

Right now it is not possible to connect more than one app to the bluealsa ALSA plugin. See #48. I haven't experimented with JACK, though. If JACK supports audio mixing, it might be possible to use it instead of ALSA interface. However, I've got no experience with JACK, and I do not know how it works (on a connection level).

Vitalic66, 2017-07-09:
No sorry that's not what i meant:

phone (BT) stream -->bluealsa pcm --> alsa loopback--\
                                                                                       \
                                                                                        |-->real soundcard-->speaker
                                                                                       /
musicplayer e.g. Kodi audio CD-->alsa loopback 2-----/

This is what i'm trying. Creaating 2 virtual soundcards to stream the music to and both can use the hardware device same time.

Arkq, 2017-07-09:
Hmm... I do not quite understand why you need two separate ALSA devices for real soundcard. If real soundcard (hw in the ALSA nomenclature) does not support mixing, you can use dmix plugin. So, you can connect (to the hw) as many apps as you want. Or I'm missing something from the picture...

Vitalic66, 2017-07-09:
So maybe i did something wrong with my setup for bluealsa and asound.conf.
 Kodi is using pcm.!default .

`bluealsa-aplay: Couldn't open PCM: Device or resource busy
`
This happens when music plays in Kodi. Cant even play a test.wav with aplay as soon as pcm.!default is in use.
Thats why I want to stream BT to a virtual card.

Arkq, 2017-07-09:
OK, so it is definitely something wrong with your ALSA configuration. Make sure, that this overwritten default PCM uses dmix (or that dmix is somewhere in the plugin chain). Not all soundcards support mixing internally, so software mixer (dmix) is required. By default, such a sane configuration is delivered by the ALSA itself. However, if you make something like this (as an example):
```
pcm.!default {
	type plug
	slave.pcm "hw:0"
}
```
all ALSA magic is gone. Now you are using bare metal (not quite bare metal, because format conversion is supported via the plug plugin, however it's just an example) hardware, which might lack audio mixing.

Vitalic66, 2017-07-09:
Call me noob but I don't understand that :/

I create via modprobe snd_aloop 2 virtual soundcards. The hardware card is a Hifiberry-DAC.

```
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```
How can i  link the Kodi stream to virtual device and also the bluealsa stream to the other virtual device and mix both and then link this on to default (real sound card).

In dmix i can only set one slave or can i also set two slaves ?

Like so ?
```
pcm.!default {
	type plug
	slave.pcm "hw:0"
}

pcm.mixed {
        type dmix
        ipc_key 321456
        slave {
                pcm "hw:Loopback,0,0"
                format S16_LE
                rate 44100
        }
        slave {
                pcm "hw:Loopback,1,0"
                format S16_LE
                rate 44100
        }

}
```
I guess not. How should i linke the pcm.mixed to the pcm.!default ?

Sorry for that foolish question, but i don't get it...

Vitalic66, 2017-07-09:
And how and where can I tell bluealsa to use virtual device 1 or 2. In Kodi I can set it up under audio setting.

Arkq, 2017-07-09:
If my understanding of the problem is correct:
- you want to play music from kodi and from bluealsa
- you've got one soundcard and want to use it

Then, what I'd have done would be setting up default PCM to use this soundcard (if it's not done already by the ALSA), e.g.:
```
pcm.!default {
        type dmix
        ipc_key 321456
        slave.pcm "hw:0"
}
```

However, firstly check if it's not done already by the ALSA. Just remove your ALSA configuration (comment it out) and see the output from the `aplay -L` command. If there is a default PCM and it uses the right hardware, then everything is done already. If not, use the configuration I've posted above.

If you are using bluealsa-aplay, you can use `-d` option to specify which PCM it should use. By default it uses "default" PCM.

Vitalic66, 2017-07-09:
The right hw soundcard is found by ALSA. Also without extra config. The problem is, like I said, when playing music in Kodi, bluealsa-aplay complains about device in use.

So starting `bluealsa-play -d bluealsa` for example might fix the start problem. But how should the config look like for pcm.bluealsa ? It has to run into the dmix ?

Arkq, 2017-07-09:
OK, so try something like this:
```
pcm.default-mix {
        type dmix
        ipc_key 321456
        slave.pcm "default"
}
```
Then, use `default-mix` in the kodi and use `-d default-mix` for the bluealsa-aplay.

Vitalic66, 2017-07-09:
`pi@raspberrypi:~ $ sudo bluealsa-aplay -d default-mix D8:C4:E9:B6:5C:80
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default-mix
bluealsa-aplay: Couldn't open PCM: No such file or directory
`

Content of .asoundrc is only your code 
`pcm.default-mix {
        type dmix
        ipc_key 321456
        slave.pcm "default"
}`

Vitalic66, 2017-07-09:
with config like this:

`pcm.default-mix {
        type dmix
        ipc_key 321456
        slave.pcm "default"
}

pcm.default {
        type hw card 0
}
`
i get this:

`pi@raspberrypi:~ $ sudo bluealsa-aplay -d default-mix D8:C4:E9:B6:5C:80
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default-mix
`

Arkq, 2017-07-09:
If you're using `sudo`, the config needs to be in the /root directory or in the /etc as a global configuration. However, you should be able to use bluealsa-aplay as a normal user (sudo is not required).

Vitalic66, 2017-07-09:
guess something must be wrong... sorry for that 

`pi@raspberrypi:~ $ bluealsa-aplay -d default-mix D8:C4:E9:B6:5C:80
ALSA lib conf.c:1231:(parse_def) default is not a compound
ALSA lib conf.c:1697:(snd_config_load1) _toplevel_:17:13:Invalid argument
ALSA lib conf.c:3417:(config_file_open) /home/pi/.asoundrc may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3339:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:3788:(snd_config_update_r) hooks failed, removing configuration
bluealsa-aplay: Couldn't open PCM: Invalid argument`

also without sudo many errors. aplay -L also not working

`pi@raspberrypi:~ $ aplay -L
ALSA lib conf.c:1231:(parse_def) default is not a compound
ALSA lib conf.c:1697:(snd_config_load1) _toplevel_:17:13:Invalid argument
ALSA lib conf.c:3417:(config_file_open) /home/pi/.asoundrc may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3339:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:3788:(snd_config_update_r) hooks failed, removing configuration
`

Arkq, 2017-07-09:
Yeah, something is wrong :D

Remove pcm.default entry from the confing. It's not allowed.

Vitalic66, 2017-07-09:
But that brings me back to the error

`pi@raspberrypi:~ $ bluealsa-aplay -d default-mix D8:C4:E9:B6:5C:80
ALSA lib pcm_dmix.c:1027:(snd_pcm_dmix_open) dmix plugin can be only connected to hw plugin
bluealsa-aplay: Couldn't open PCM: Invalid argument
`

now I'm confused....

but when i do something like that:

`pcm.default-mix {
        type dmix
        ipc_key 321456
        slave.pcm "test"
}

pcm.test {
        type hw card 0
}
`
i get a new error type. at least not complaining about pcm:

`pi@raspberrypi:~ $ bluealsa-aplay -d default-mix D8:C4:E9:B6:5C:80
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:197: Getting transport for D8:C4:E9:B6:5C:80 type 1
bluealsa-aplay: Couldn't set HW parameters: Invalid argument`

Arkq, 2017-07-09:
It would be much easer to communicate without delay. IRC: #bluealsa @freenode.oeg

Vitalic66, 2017-07-09:
After nice conversation via IRC here is the solution to make it work as suggested. Configure the .asoundrc like this:

`pcm.!default {
  type plug
  slave.pcm "dmix"
}
`
In case someone else will stumble over this by searching the net :) THX Arkq

Vitalic66, 2017-07-10:
I forget to mention how to use it:

```
bluealsa-aplay -d plugdmix XX:XX:XX:XX:XX:XX
```

Vitalic66, 2017-07-16:
Somehow it's not working with compiled master version. It's working with downloaded not self-compiled version.

Did you change something?

When using onfig with plugdmix, kodi can't find default pcm but bluealsa-aplay works. When using pcm dmix, kodi is finding default and plays sound, but bluealsa-aplay says pcm open error, already in use.

closed 2017-07-16 by Vitalic66

roizcorp, 2019-10-04:
Hi,

Looping back to this,  I cannot get this dmix in my configuration at all, meaning I do put it as stated above but `aplay -L` does not show anything remotely to the output shown above,

Does it work still?

xavi-b, 2020-09-17:
Hi,

I was looking at a similar problem to play both bluetooth sound from my phone and sound from other applications on my RPI4.

I solved it using the following .asoundrc:
```
pcm.!default {
	type asym
	playback.pcm {
		type plug
		slave.pcm "dmixer"
	}
	capture.pcm {
		type plug
		slave.pcm "input"
	}
}

pcm.bluealsadmix {
    type plug
    slave.pcm "dmixer"
}

pcm.dmixer {
    type dmix
    ipc_key 1024
    slave.pcm "output"
}

pcm.output {
	type hw
	card 0
}

ctl.!default {
	type hw
	card 0
}
```
using the following link to configure bluealsa-play as a A2DP Sink:
https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=247892&p=1513571
and changing the aplay.service file to:
```
[Unit]
Description=BlueALSA aplay service
After=bluealsa.service
Requires=bluealsa.service

[Service]
Type=simple
User=pi
ExecStartPre=/bin/sleep 20
ExecStart=/usr/bin/bluealsa-aplay -vv -d bluealsadmix 00:00:00:00:00:00

[Install]
WantedBy=multi-user.target
```