arkq/bluez-alsa #51: How to handle mismatched formats

djeebus, 2017-07-10:
I'm trying to set up a raspberry pi as a bluetooth speaker, but ran into a problem. When running `bluealsa-aplay --profile-a2dp 14:56:8E:3E:13:F2 --pcm=hw:1,0`, I'm getting:

```
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:213: Getting transport for 14:56:8E:3E:13:F2 type 1
bluealsa-aplay: Couldn't set HW parameters: Set format: Invalid argument: S16_LE
```

After some googling around, it turns out my sound card is a bit higher than that:

```
pi@raspberrypi:~/code/bluez-alsa/src $ cat /proc/asound/card1/stream0 
LH Labs Geek Out HD Audio 1V0 at usb-3f980000.usb-1.4, high speed : USB Audio

Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
  Interface 1
    Altset 2
    Format: SPECIAL
    Channels: 2
    Endpoint: 1 OUT (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
```

I think the key part there is `Format: S32_LE`, right? I'm new to alsa, is there a way to convert from one to the other, or is this just a limitation?

Arkq, 2017-07-10:
See this very recent issue:  #50

Basically, what you need is a `plug` plugin which will convert format for you.

djeebus, 2017-07-10:
Oh! Sorry, saw that but didn't understand that it was what I needed. On the plus side, I'm learning a gob about alsa. Thanks!

closed 2017-07-10 by djeebus