arkq/bluez-alsa #52: Turn off HFP

Vitalic66, 2017-07-10:
Hi,

is it possible to turn off the HFP ? Background. I let bluealsa server start via rc.local, ok. Then i wrote a script to start bluealsa-aplay..... when udev recognizes the connection of BT device.

It works like charme, for the first 20 sec or so, but then i can see that udev acts again, because HFP is getting freed by bluealsa server.

This is how my script ( bt.sh in /usr/lib/udev )looks like:

```
#!/bin/bash

 ACTION=$(expr "$ACTION" : "\([a-zA-Z]\+\).*")
 if [ "$ACTION" = "add" ]
 then

        for dev in $(find /sys/devices/virtual/input/ -name input*)
                do
                        if [ -f "$dev/name" ]
                        then

                        mac=$(cat "$dev/name")

                        sudo -u pi bluealsa-aplay -d plugdmix $mac

                        fi
                done
fi
```

This is what the udev.rules ( 99-input.rules  in /etc/udev/rules.d ) looks like:

```
SUBSYSTEM=="input", GROUP="input", MODE="0660"
KERNEL=="input[0-9]*", RUN+="/usr/lib/udev/bt.sh"
```

This happens in udevadm monitor when connecting:

```
KERNEL[750.803547] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/bluetooth/hci0/hci0:71 (bluetooth)
UDEV  [750.832100] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/bluetooth/hci0/hci0:71 (bluetooth)
KERNEL[754.038376] add      /devices/virtual/input/input3 (input)
KERNEL[754.039283] add      /devices/virtual/input/input3/event2 (input)
UDEV  [786.071908] add      /devices/virtual/input/input3 (input)
UDEV  [786.079762] add      /devices/virtual/input/input3/event2 (input)
```

And this happens on bluealsa server:

```
bluealsa: ../../src/bluez.c:438: A2DP Sink (SBC) configured for device D8:C4:E9:B6:5C:80
bluealsa: ../../src/bluez.c:440: Configuration: channels: 2, sampling: 44100
bluealsa: ../../src/transport.c:601: State transition: 0 -> 0
bluealsa: ../../src/ctl.c:534: New client accepted: 11
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa: ../../src/ctl.c:313: PCM requested for D8:C4:E9:B6:5C:80 type 1 stream 1
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa: ../../src/ctl.c:537: +-+-
bluealsa: ../../src/io.c:1088: RFCOMM disconnected: Connection reset by peer
bluealsa: ../../src/transport.c:601: State transition: 2 -> 4
bluealsa: ../../src/transport.c:769: Closing RFCOMM: 8
bluealsa: ../../src/transport.c:324: Freeing transport: HFP Hands-Free
bluealsa: ../../src/transport.c:324: Freeing transport: HFP Hands-Free
bluealsa: ../../src/io.c:55: Exiting IO thread
bluealsa: ../../src/ctl.c:506: Client closed connection: 11
bluealsa: ../../src/transport.c:844: Cleaning PCM FIFO: /var/run/bluealsa/hci0-D8:C4:E9:B6:5C:80-                                                                                                    1-1
bluealsa: ../../src/ctl.c:537: +-+-
```

And thats the problem. Something what ever is going wrong with HFP. Seems to be a phone problem. Anyway, as soon as HFP gets disconnected, udev is seeing another "add" and calls the script again. And this causes the stream to stop. 

Maybe there is another way to stop the second "add" event. Is there a way to read hci0 connection status as for example "connected" ? If yes, i could write another if condition to the script...

Arkq, 2017-07-10:
Yes it is possible to disable HFP and HSP. Just run `bluealsa -h` and you will see a handful of options :).

I can see that you are not using master snapshot of blueaz-alsa. I would recommend (if it is possible for you) to compile a fresh master snapshot. I've recently added HFP support (in your version there is no HFP support at all).

Furthermore, you could test some "new" functionality, which could simplify a little bit your usage of bluealsa. Just compile [bt-speaker](https://github.com/Arkq/bluez-alsa/tree/bt-speaker) branch. And then you could simply use: `bluealsa-aplay 00:00:00:00:00:00` (bt.sh hack is not required any more). Run `bluealsa-aplay -h` for more information. However this feature is not stable yet. I've got only one phone with A2DP capabilities, so it is hard for me to test it properly.... Hence, it is not in the master yet.

Vitalic66, 2017-07-10:
I'd like to compile a fersh version, but got several things to manage first like ortp and therefore other dependencies. 
Or do you have a tutorial for compiling ?

Arkq, 2017-07-10:
No, I've got no tutorial. However, I can give you a hint. oRTP is not required to be compiled. I'm using only a header file for RTP header structure. Just copy ortp/rtp.h file into the include path (e.g. /usr/include/) and that's all.

Vitalic66, 2017-07-10:
That easy :/

ok...

Vitalic66, 2017-07-11:
Still got some problems with compiling. 

Maybe there should realy be something like a tutorial

```
pi@raspberrypi:~/bluez-alsa $ sudo ./configure
./configure: line 2055: syntax error near unexpected token `foreign'
./configure: line 2055: `AM_INIT_AUTOMAKE(foreign subdir-objects -Wall -Werror)'
```
an before:

```
pi@raspberrypi:~/bluez-alsa $ sudo autoconf
configure.ac:6: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:13: error: possibly undefined macro: AM_PROG_AR
configure.ac:14: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:24: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:25: error: possibly undefined macro: AM_COND_IF
configure.ac:26: error: possibly undefined macro: AC_DEFINE
configure.ac:120: error: possibly undefined macro: AC_MSG_WARN
```

Arkq, 2017-07-11:
Compilation from a source always requires a little bit of skills. In the readme file there is sufficient info for someone who has some experience with compilation (IMHO). The problem with a tutorial is where to draw the line. I cannot assume someone's knowledge and experience. In the "real" world, compilation is performed by the package maintainer for a specific linux distro. There is a LOT of not standard linux distributions, for which a simple tutorial might not suffice.

First of all, do not use sudo for compilation process - it doesn't seem right :). Sudo should be used for installation, though. In your case it seems that the automake in not properly installed (or the version is wrong), or some of its components are missing.

Vitalic66, 2017-07-11:
Also tried autoconf and ./configure as user without sudo. Got immediatly permission denied in both cases.

I'll make a try to remove autoconf and reinstall it. Autoconf was working by the way die thw not necessary packages like oRTP and it's dependencies.

Arkq, 2017-07-11:
> Got immediatly permission denied in both cases.

If you've been using `sudo` before, files created by the configure/automake scrip have wrong (root) owner and cannot be overwritten. Remove the whole project directory and clone it once more.

Post your automake version `automake --version`.

Vitalic66, 2017-07-11:
```
automake (GNU automake) 1.14.1
autoconf (GNU Autoconf) 2.69
```
Following your hint with removing former bluez-alsa folder and recloning git and also following the readme brought me to this:
```
checking ortp/rtp.h usability... no
checking ortp/rtp.h presence... no
checking for ortp/rtp.h... no
```
but
```
pi@raspberrypi:~/bluez-alsa/build $ sudo find / -name rtp.h
/home/pi/ort/ortp-0.27.0/include/ortp/rtp.h
/usr/lib/gcc/arm-linux-gnueabihf/4.9/include/ortp/rtp.h
/usr/include/ortp/rtp.h
```

In which directory is configure looking for rtp.h ? Thought in my standart directories for .h files ?

Arkq, 2017-07-11:
The `rtp.h` file might have some dependencies. In my case, this file requires `ortp/ports.h` and `ortp/str_utils.h` - just check for `#include` statements in the `rtp.h` file from the oRTP project.

However, in order to simplify it even more, just copy the definition of the `rtp_header` structure from the `rtp.h` file into the [a2dp-rtp.h](https://github.com/Arkq/bluez-alsa/blob/master/src/a2dp-rtp.h) file and remove the ortp include. Then also remove the `AC_CHECK_HEADERS([ortp/rtp.h]` from the [configure.ac](https://github.com/Arkq/bluez-alsa/blob/master/configure.ac) file. Autoreconf and you are ready to go.

I think, that in the future I will remove this oRTP dependency, if it will not be required for some new features...

Vitalic66, 2017-07-11:
Was looking into config.log meanwhile and saw what you described.

Fixed rtp problems now, and now i got a new problem...

```
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for ALSA... no
configure: error: Package requirements (alsa) were not met:

No package 'alsa' found

```
pi@raspberrypi:~/bluez-alsa/build $ aplay --version
aplay: version 1.0.28 by Jaroslav Kysela <perex@perex.cz>
```

How can i figure out the standard PATH and how do I add the PATH as variable to configure. Remembering something like ../configure -PATH...

Vitalic66, 2017-07-11:
Ok finished compiling actual master branch of bluez-alsa...

Started bluealsa without HFP support via rc.local `sudo bluealsa --disable hfp`

and still

```
pi@raspberrypi:/usr/lib/udev $ udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[416.555144] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/bluetooth/hci0/hci0:71 (bluetooth)
UDEV  [416.584439] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5.2/1-1.5.2:1.0/bluetooth/hci0/hci0:71 (bluetooth)
KERNEL[419.038558] add      /devices/virtual/input/input3 (input)
KERNEL[419.039179] add      /devices/virtual/input/input3/event2 (input)
UDEV  [451.075672] add      /devices/virtual/input/input3 (input)
UDEV  [451.084335] add      /devices/virtual/input/input3/event2 (input)
```

After some seconds of listening to music udev again registers an "add" action and causes my bt.sh to run again. Still HFP is on... ?

Arkq, 2017-07-11:
Try to disable HSP as well: `bluealsa --disable-hfp --disable-hsp`

However, it might not resolve your problem. When BT is connected by the bluez, only a part of connection is handled by the audio handler (it this case bluealsa), the rest is handled internally. If the device supports input events (e.g. buttons), then new input device is registered (which might be your case - `add /devices/virtual/input/input3/event2`).

Try to use `bluealsa-aplay` from the bt-speaker branch (not master) without the bt.sh script. Just run it as a standalone player. It should handle connection and reconnection of BT devices.

Vitalic66, 2017-07-11:
Is the usage the same as with bluealsa-aplay. Cause i have to change the PCM.

Arkq, 2017-07-11:
Yes, it's the same program. Usage is more or less the same. It will be available (someday) in the master branch anyway. The only important change is that it now can automatically reinitialize itself when you connect new BT device. There is no need to restart it any more - at least in theory, because like I've said in some other comment, it may contain some bugs.

Vitalic66, 2017-07-11:
Doesn't work some how (compiled bt-speaker branch). BTW it's a Samsung Galaxy S7 EDGE @ Android Nougat 7.0

started bluealsa via rc.local
connectod phone via BT
started bluealsa-aplay -d plugdmix XX:XX:XX:XX:XX:XX
music streams and sound comes out of speaker
paused music and started again without BT disconnect --> sound comes out of speaker
disconnected phone (BT off)
connected phone (BT on)
started music --> no sound out of the speaker. (can see stream playing on bluealsa display output)

infirit, 2017-07-11:
The best way to handle this is to monitor the bluez dbus api. I think I posted a small python script before on one of the issue for something similar. I can try to find it in my script folder if interested.

Vitalic66, 2017-07-11:
Yeah please go ahead. Thanks in advance.

Arkq, 2017-07-11:
@Vitalic66 It seems, that the 'it may contain some bugs' part is true to the letter :D I will try to fix it tomorrow.

Vitalic66, 2017-07-11:
if bug = true 
then better go to bed 
else get a cigarette and try something else

Arkq, 2017-07-11:
Pull the bt-speaker branch and recompile. It was a quick (but not dirty) fix :)
I might even merge this code with the master, because it seems, that is was the bug which I've seen few months ago.

Vitalic66, 2017-07-11:
nope. compiled new version:
`
bluealsa: Couldn't register endpoint: GDBus.Error:orgfreedesktop.DBus.Error.TimedOut: Activation of org.bluez timed out`

Vitalic66, 2017-07-11:
Sorry my fault. Something went wrong during first try.

infirit, 2017-07-11:
@Vitalic66, This gives you the basic idea how it is done. Of course replace ``00_9E_C8_61_C7_7F`` with the mac matching your device.

```python
from gi.repository import Gio, GLib

def on_properties_changed(proxy, changed, invalidated):
    changed_props = changed.unpack()
    if "Connected" in changed_props and changed_props["Connected"]:
        print("Starting bluealsa-play")

proxy = Gio.DBusProxy.new_for_bus_sync(
    Gio.BusType.SYSTEM,
    Gio.DBusProxyFlags.DO_NOT_AUTO_START,
    None,
    "org.bluez",
    "/org/bluez/hci0/dev_00_9E_C8_61_C7_7F",
    "org.bluez.Device1")

proxy.connect("g-properties-changed", on_properties_changed)

param = GLib.Variant('(ss)', ("org.bluez.Device1", "Connected"))

prop = proxy.call_sync("org.freedesktop.DBus.Properties.Get", param,
                       Gio.DBusCallFlags.NONE, -1)

if not prop.unpack()[0]:
    print("Not connected waiting for connection")
else:
    print("Starting bluealsa-play")
    
GLib.MainLoop().run()
```

Vitalic66, 2017-07-11:
@Arkq unfort. same behavior as before update

started bluealsa via rc.local
connectod phone via BT
started bluealsa-aplay -d plugdmix XX:XX:XX:XX:XX:XX
music streams and sound comes out of speaker
paused music and started again without BT disconnect --> sound comes out of speaker
disconnected phone (BT off)
connected phone (BT on)
started music --> no sound out of the speaker. (can see stream playing on bluealsa display output)

@infirit Thx in advance, will give this a try tomorrow

Vitalic66, 2017-07-13:
Hi @infirit DBUS is a good idea.  I once played around with it to control playback of remote device via AVRCP.
I used dbus messages like this and it was working:
```
dbus-send --system --print-reply --type=method_call --dest=org.bluez /org/bluez/hci0/dev_D8_C4_E9_B6_5C_80/player3 org.bluez.MediaPlayer1.Play
```

And yes, it still works without problems today too.

Now i'd like to read the Connected status witch is also supported.
```
pi@raspberrypi:~ $ qdbus --system org.bluez /org/bluez/hci0/dev_D8_C4_E9_B6_5C_80
method QString org.freedesktop.DBus.Introspectable.Introspect()
property read QDBusObjectPath org.bluez.Device1.Adapter
property read QString org.bluez.Device1.Address
property read QByteArray org.bluez.Device1.AdvertisingFlags
property readwrite QString org.bluez.Device1.Alias
property read ushort org.bluez.Device1.Appearance
property readwrite bool org.bluez.Device1.Blocked
property read uint org.bluez.Device1.Class
property read bool org.bluez.Device1.Connected
property read QString org.bluez.Device1.Icon
property read bool org.bluez.Device1.LegacyPairing
property read QDBusRawType::a{qv} org.bluez.Device1.ManufacturerData
property read QString org.bluez.Device1.Modalias
property read QString org.bluez.Device1.Name
property read bool org.bluez.Device1.Paired
property read short org.bluez.Device1.RSSI
property read QVariantMap org.bluez.Device1.ServiceData
property read bool org.bluez.Device1.ServicesResolved
property readwrite bool org.bluez.Device1.Trusted
property read short org.bluez.Device1.TxPower
property read QStringList org.bluez.Device1.UUIDs
method void org.bluez.Device1.CancelPairing()
method void org.bluez.Device1.Connect()
method void org.bluez.Device1.ConnectProfile(QString UUID)
method void org.bluez.Device1.Disconnect()
method void org.bluez.Device1.DisconnectProfile(QString UUID)
method void org.bluez.Device1.Pair()
method QDBusVariant org.freedesktop.DBus.Properties.Get(QString interface, QString name)
method QVariantMap org.freedesktop.DBus.Properties.GetAll(QString interface)
signal void org.freedesktop.DBus.Properties.PropertiesChanged(QString interface, QVariantMap changed_properties, QStringList invalidated_properties)
method void org.freedesktop.DBus.Properties.Set(QString interface, QString name, QDBusVariant value)
property read bool org.bluez.Network1.Connected
property read QString org.bluez.Network1.Interface
property read QString org.bluez.Network1.UUID
method QString org.bluez.Network1.Connect(QString uuid)
method void org.bluez.Network1.Disconnect()
property read bool org.bluez.MediaControl1.Connected
property read QDBusObjectPath org.bluez.MediaControl1.Player
method void org.bluez.MediaControl1.FastForward()
method void org.bluez.MediaControl1.Next()
method void org.bluez.MediaControl1.Pause()
method void org.bluez.MediaControl1.Play()
method void org.bluez.MediaControl1.Previous()
method void org.bluez.MediaControl1.Rewind()
method void org.bluez.MediaControl1.Stop()
method void org.bluez.MediaControl1.VolumeDown()
method void org.bluez.MediaControl1.VolumeUp()
```

As you can see: ```property read bool org.bluez.Device1.Connected```
The problem for me is now, how do I use this in the command line. Something like this doesn't work:

```
pi@raspberrypi:~ $ dbus-send --system --print-reply --type=method_property --dest=org.bluez /org/bluez/hci0/dev_D8_C4_E9_B6_5C_80 org.bluez.Device1.Connected
```

Also not with --type=property or --type=get.property . So how does the syntax have to look like ?
Looking at your code, I can see that u use something similar ```"org.freedesktop.DBus.Properties.Get"```

infirit, 2017-07-13:
``Get`` is a mehod on interface ``org.freedesktop.DBus.Properties`` so we need to put them together.

In your introspection output you see ``method QDBusVariant org.freedesktop.DBus.Properties.Get(QString interface, QString name)`` this means ``Get`` expects you to provide the interface the property is on and the property name as strings. 

Putting the whole thing together we get,
```
dbus-send --system --type=method_call --print-reply --dest="org.bluez" "/org/bluez/hci0/dev_D8_C4_E9_B6_5C_80" "org.freedesktop.DBus.Properties.Get" string:"org.bluez.Device1" string:"Connected"
```

However I strongly suggest to use something like gdbus or qdbus as these are much more user friendly and provide cleaner output, example.

```
qdbus --system "org.bluez" "/org/bluez/hci0/dev_D8_C4_E9_B6_5C_80" "org.freedesktop.DBus.Properties.Get" "org.bluez.Device1" "Connected"
```

Vitalic66, 2017-07-13:
For information how to use it: **qdbus** does the magic not dbus-send in this case:

For example MAC Address:
```
pi@raspberrypi:~ $ qdbus --system --literal org.bluez /org/bluez/hci0/dev_D8_C4_E9_B6_5C_80 org.freedesktop.DBus.Properties.Get org.bluez.Device1 Address
[Variant(QString): "D8:C4:E9:B6:5C:80"]
```
and whether connected is true:
```
pi@raspberrypi:~ $ qdbus --system --literal org.bluez /org/bluez/hci0/dev_D8_C4_E9_B6_5C_80 org.freedesktop.DBus.Properties.Get org.bluez.Device1 Connected
[Variant(bool): true]
```

Based on this i can write script for bluealsa-aplay.

Arkq, 2018-08-01:
There has been a lot of changes in the bluealsa-aplay since this issue has been created. I'm not able to reproduce the behaviour from [this comment](https://github.com/Arkq/bluez-alsa/issues/52#issuecomment-314582029), so I'm closing this issue. However, if the current master indeed does not work, please reopen it and provide logs from `bluealsa` and `bluealsa-aplay -vv` (compile bluez-alsa with debugging enabled).

closed 2018-08-01 by Arkq