arkq/bluez-alsa #53: configure error: checking for BLUEZ... no

hagi123456, 2017-07-11:
When installing ```bluez-alsa```, we came across the following error even though ```bluez``` was installed at version 5.43 (debian stretch):
```
checking for BLUEZ... no
configure: error: Package requirements (bluez >= 5.0) were not met:

No package 'bluez' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables BLUEZ_CFLAGS
and BLUEZ_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```
After quiet a long search we found out that 
```libbluetooth-dev``` was the package that was missing. Mentioning this in the dependencies list would be great

closed 2017-07-12 by Arkq

MiguelPerezI, 2021-03-21:
Omg thank you!!

MiguelPerezI, 2021-03-21:
For Checking for 'bluez'                     : not found 
I never wouldǘe guessed that libbluetooth-dev would have fixed the problem.

MiguelPerezI, 2021-03-21:
[image]

commit b64bddaa (2017-07-12): Hint for Debian-based distributions

Fixes #53