arkq/bluez-alsa #54: Cannot open shared library

neumad14, 2017-07-12:
After installing bluez-alsa, I encountered a weird error message when trying to play a sound. 
(Like: aplay -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp test.wav)
All i got was the message:

`ALSA lib dlmisc.c:254:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/x86_64-linux-gnu/alsa-lib/libasound_module_pcm_bluealsa.so
aplay: main:788: audio open error: No such device or address
`
Until now I have no clue why I get this error, since I also made sure that all packages are properly installed. 
Many thanks in advance

Arkq, 2017-07-12:
Is this `/usr/lib/x86_64-linux-gnu/alsa-lib/libasound_module_pcm_bluealsa.so` file present? Because it is rather strange location for alsa modules. You might try to copy the `libasound_module_pcm_bluealsa.so` file (from the `./build/src/asound/.libs/` directory) manually to this location.

neumad14, 2017-07-13:
Ok I found out that I had an alsa-lib directory in /usr/lib where it was stored and just had to move it from there. Now he finds the file but gets a further error:
`aplay -D bluealsa:HCI=hci0,DEV=50:C9:71:B9:96:53,PROFILE=a2dp voice.wav
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:213: Getting transport for 50:C9:71:B9:96:53 type 1
ALSA lib ../../../src/asound/bluealsa-pcm.c:660:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:722: audio open error: No such device`
Does this mean that he doesnt find my bluetooth headset or is there something wrong with my Alsa?

Arkq, 2017-07-13:
BlueALSA and PulseAudio does not work together. See [troubleshooting](https://github.com/Arkq/bluez-alsa#troubleshooting) section in the readme.

neumad14, 2017-07-13:
Yeah i know that but why isnt the headset showing up under my alsa devices

Arkq, 2017-07-13:
BlueALSA does not expose BT devices via the kernel ALSA interface. It provides only a PCM software plugin. If you want your PCM to be displayed alongside `aplay -L` PCMs, you have to add hint description in the configuration file.

See:
http://alsa.opensrc.org/Asoundrc
http://alsa.opensrc.org/Plugin_Documentation

neumad14, 2017-07-14:
But still what does this error mean
`ALSA lib ../../../src/asound/bluealsa-pcm.c:660:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:788: audio open error: No such device
`

Arkq, 2017-07-14:
The question is, does bluealsa see any BT device. In order to check it, run
`amixer -D bluealsa`. If there is no devices, then something is wrong with
your setup (if BT device is indeed connected). Some other app might be
connected to the bluez and "steals" BT devices. If there is some device,
check if is input (capturing) or output (playing) device - maybe you're
trying to use wrong device. Also check device profile: A2DP vs SCO.

neumad14, 2017-07-14:
`amixer -D bluealsa
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:125: Subscribing for notifications: true
`
Well I guess there is something wrong

neumad14, 2017-07-17:
Is it even supporting debian 8 or higher?

neumad14, 2017-07-17:
Ok I got the headset working now, only problem is that the headset always disconnects after I played a sound or recorded something. Also the recordings have a really disturbing sough.

closed 2018-12-15 by Arkq

foxupdown, 2020-06-12:
/ # aplay 0.pcm 
aplay: symbol 'eventfd_read': can't resolve symbol
aplay: symbol 'eventfd_write': can't resolve symbol
ALSA lib dlmisc.c:254:(snd1_dlobj_cache_get) Cannot open shared library /lib/alsa-lib/libasound_module_pcm_bluealsa.so
aplay: main:786: audio open error: No such device or address
/ # ls /lib/alsa-lib/libasound_module_pcm_bluealsa.so 
/lib/alsa-lib/libasound_module_pcm_bluealsa.so
alsa-lib 1.1.1 and alsa-util1.1.1