arkq/bluez-alsa #57: HFP usage / no sound

Vitalic66, 2017-08-27:
Hi again.

Can you give some instructions how to use HFP ? BT is connected and bluealsa also starts just like bluealsa-aplay.
BT connection on phone is showing HFP active. No need for A2DP anymore, as I joined a project to get Android Auto to work on Pi3.

When i make a testcall, the ringtone is still on phone, not on speaker connected to pi.

Vitalic66, 2017-08-27:
Rhought about my needs, 
I need to set source (mic input) to externel mic on sound card  (usb card with mic connected) and output to speaker.
Right now sound and mic input is taken from phone... as it is right.

Arkq, 2017-08-30:
Hi,

The HFP support is still in progress. I'm making some progress, however it is not "production ready" yet :) Currently, I'm able to forward audio from phone (Android 6.0.1) with the HFP. However, I haven't tested receiving sound, not yet, so it might not work.

Please tell me what phone you are using (with the OS version, if it is one of common OSes). Also, post the debug output from the bluealsa, preferably with timing (`../configure --enable-debug --enable-debug-time`).

marcoforlin, 2017-11-16:
Hi, 

I have a related issue
[bluealsalog.txt](https://github.com/Arkq/bluez-alsa/files/1479625/bluealsalog.txt)

. I connect my linux hw (wandboard) to a phone (gs6) via bluetooth and try to play audio using bluezalsa with the command:

sudo aplay -D bluealsa:HCI=hci0,DEV=X:X:X:X:X,PROFILE=sco testaudio.wav

I would expect to hear something on the receiving phone, but i get nothing. SCO device is present on the amixer, at 100% volume.  The playback loop is started but with no sound.

Logs from bluezalsa seem ok, I attach them.

Do you have any idea of what could be the issue? Thanks for your help.

kind regards,
Marco

Arkq, 2018-12-15:
Please, use oFono for HFP - #172

closed 2018-12-15 by Arkq