arkq/bluez-alsa #58: Illegal Instruction on Rpi ZeroW

paul-1, 2017-08-28:
bluealsa is working fine on a Rpi3, but on a Rpi ZeroW, bluealsa crashes with an Illegal instruction.

```
$ sudo bluealsa
bluealsa: ctl.c:489: Starting controller loop
bluealsa: bluez.c:699: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Source
bluealsa: bluez.c:699: Registering endpoint: 0000110A-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSource
bluealsa: bluez.c:699: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DP_MPEG24_Sink
bluealsa: bluez.c:699: Registering endpoint: 0000110B-0000-1000-8000-00805F9B34FB: /MediaEndpoint/A2DPSink
bluealsa: bluez.c:935: Registering profile: 00001108-0000-1000-8000-00805F9B34FB: /Profile/HSPHeadset
bluealsa: bluez.c:935: Registering profile: 00001112-0000-1000-8000-00805F9B34FB: /Profile/HSPAudioGateway
bluealsa: bluez.c:935: Registering profile: 0000111E-0000-1000-8000-00805F9B34FB: /Profile/HFPHandsFree
bluealsa: bluez.c:935: Registering profile: 0000111F-0000-1000-8000-00805F9B34FB: /Profile/HFPAudioGateway
bluealsa: main.c:226: Starting main dispatching loop
bluealsa: bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: transport.c:608: State transition: 0 -> 2
bluealsa: transport.c:96: Created new IO thread: HFP Audio Gateway
bluealsa: ctl.c:571: +-+-
bluealsa: bluez.c:796: HFP Audio Gateway configured for device A0:E9:DB:0D:2D:FD
bluealsa: transport.c:608: State transition: 0 -> 2
bluealsa: transport.c:96: Created new IO thread: HFP Audio Gateway
bluealsa: rfcomm.c:389: Starting RFCOMM loop: HFP Audio Gateway
bluealsa: at.c:156: AT message: SET: command:+BRSF, value:59
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:+BRSF, value:992
bluealsa: io.c:997: Starting IO loop: HFP Audio Gateway
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:55: HFP state transition: 0 -> 2
bluealsa: at.c:156: AT message: TEST: command:+CIND, value:(null)
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0,1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:55: HFP state transition: 2 -> 5
bluealsa: at.c:156: AT message: GET: command:+CIND, value:(null)
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:+CIND, value:0,0,1,5,0,5,0
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:55: HFP state transition: 5 -> 7
bluealsa: at.c:156: AT message: SET: command:+CMER, value:3, 0, 0, 1
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: rfcomm.c:55: HFP state transition: 7 -> 8
bluealsa: rfcomm.c:55: HFP state transition: 8 -> 9
bluealsa: rfcomm.c:55: HFP state transition: 9 -> 13
bluealsa: at.c:156: AT message: SET: command:+VGS, value:10
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: ctl.c:571: +-+-
bluealsa: at.c:156: AT message: GET: command:+BTRH, value:(null)
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:OK
bluealsa: at.c:156: AT message: CMD: command:+CLCC, value:(null)
bluealsa: Unsupported AT message: AT+CLCC
bluealsa: rfcomm.c:48: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: bluez.c:504: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: bluez.c:504: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ctl.c:571: +-+-
bluealsa: bluez.c:438: A2DP Source (SBC) configured for device A0:E9:DB:0D:2D:FD
bluealsa: bluez.c:440: Configuration: channels: 2, sampling: 48000
bluealsa: transport.c:608: State transition: 0 -> 0
bluealsa: ctl.c:550: New client accepted: 13
bluealsa: ctl.c:571: +-+-
bluealsa: ctl.c:571: +-+-
bluealsa: ctl.c:521: Client closed connection: 13
bluealsa: ctl.c:571: +-+-
bluealsa: ctl.c:550: New client accepted: 13
bluealsa: ctl.c:571: +-+-
bluealsa: ctl.c:571: +-+-
bluealsa: ctl.c:326: PCM requested for A0:E9:DB:0D:2D:FD type 1 stream 0
bluealsa: transport.c:698: New transport: 16 (MTU: R:672 W:895)
bluealsa: ctl.c:571: +-+-
bluealsa: bluez.c:1039: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:608: State transition: 0 -> 2
bluealsa: transport.c:96: Created new IO thread: A2DP Source (SBC)
Illegal instruction
```

paul-1, 2017-08-28:
Found a problem with libsbc (it was build for armv7)  so I rebuilt it, and I get a few lines further before a segfault.  This happens when playback is started.

```
bluealsa: transport.c:96: Created new IO thread: A2DP Source (SBC)
bluealsa: io.c:50: Opening FIFO for reading: /var/run/bluealsa/hci0-A0:E9:DB:0D:2D:FD-1-0
bluealsa: io.c:408: Starting IO loop: A2DP Source (SBC)
bluealsa: transport.c:608: State transition: 2 -> 2
bluealsa: ctl.c:571: +-+-
Segmentation fault
```

Arkq, 2017-08-30:
Hi,

Could you run bluealsa with gdb? I've got no armv7 device with bluetooth, which I could use for testing. Compile bluealsa with the `-g` option: `CFLAGS=-g ../configure [extra options is any]`, then run bluealsa as follows: `gdb --args bluealsa`. When the segfault occurs, type `bt` and post here the results.

paul-1, 2017-08-31:
Armv7 devices were fine.   This is only a problem with armv6.   I ran the debugger, and it was still libsbc crashing....  Finally got the right search term in google to find this.   https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=856487  I applied the referenced patch to not use inline asm for armv6, and it now runs without crashing.  

At least for the first track.  Anytime I change tracks or skip ahead in the track, the alsa output thread from the player stops sending data.   In this case, there are no errors or crashes from the player or bluealsa.  I'm assuming this is still a libsbc issue.   (For the record, I'm using squeezelite as the player.)

closed 2017-09-03 by paul-1