arkq/bluez-alsa #59: Raspberry Pi Zero W blue-alsa input

jrseriel, 2017-09-01:
Curious to see if anyone has been able to get blue-alsa to provide bluetooth input over something like a bluetooth headset, on the raspberry pi zero W. I have been trying pretty vigurously for the past couple of days to no avail. After I have followed all the steps on issue#22 and issue#7 and still receive a no device found error when I try to arecord. Any input would be appreciated!

Arkq, 2017-09-04:
Hi,

Firstly, you have to find out where the logs for bluealsa are stored (if anywhere) if you using Raspbian Stretch (I'm not a raspberry user, so I do not know where such logs are stored). If you're running bluealsa by yourself, post logs here (make sure that bluez-alsa was compiled with `--enable-debug`).

YOUN3SS, 2017-09-11:
+ Try to do it with Bluetooth-to-USB dongle, using Raspberry Pi Bluetooth doesn't give any success.
http://youness.net/raspberry-pi/bluetooth-headset-raspberry-pi-3-ad2p-hsp

Rob58329, 2018-08-28:
Although this is an old thread, I though the following info might be of interest regarding the Raspberry Pi Zero using Stretch:

Up until mid August 2018, "sudo apt-get install bluealsa" installed: "0.7 100  100 /var/lib/dpkg/status"

Together with "bluez v5.50", this populated "sudo bluetoothctl show" with 11 "UUID" profiles, including the Audio Sink profile, enabling me to connect my mobile telephone to my PiZero fine using "Media Audio"

However, the mid-August "sudo apt-get upgrade" updated bluealsa to: "0.9 500 500 http://archive.raspberrypi.org/debian stretch/main armhf"

This update has removed/disappeared the following 3 UUID profiles from "sudo bluetoothctl show":
	UUID: Audio Sink
	UUID: Handsfree
	UUID: Headset
(and also caused the "sudo systemctl status bluetooth" error: "bluetoothd: Failed to set privacy: Rejected (0x0b)"

Predictably, my mobile telephone will no longer connect to the Raspberry Pi Zero.

The only solution I have found is to re-install an earlier backup version of Stretch with bluealsa 0.7, and then issue a "sudo apt-mark hold bluealsa" command so that "sudo apt-get upgrade" will NOT update/change the bluealsa version number.
(Or use the full version of Stretch 27/6/18 which has the 0.7 version of bluealsa, instead of the Stretch-lite version.)

Arkq, 2018-08-28:
@Rob58329, see issue #137

XECDesign, 2018-08-30:
Sorry about posting here instead of there, but I'm getting 404 on that issue and hovering over it says it's private.

> I think that the problem is cause by recent bluealsa changes (according to this comment). By default not all BT profiles are enabled. In order to enable A2DP Sink, one has to specify this profile on the command line. So, it seems that the startup script has to be changed from bluealsa to bluealsa -p a2dp-source -p a2dp-sink. See bluealsa --help for details.

Would there be any negative effects from using `bluealsa -p a2dp-source -p a2dp-sink` by default?

XECDesign, 2018-08-30:
Alright, thanks. I'll leave it as is for now then.

Rob58329, 2018-08-31:
re. No "UUID: Audio Sink" after mid-Aug18 "sudo apt-get update" of bluealsa to v0.90 (Raspberry Pi Zero W)

@Arkq, as @XECDesign says, issue #137 is marked as "Issue title is private" so I can't view it.

@XECDesign, thanks for repeating the info which I am guessing is in issue #137.

I can confirm that for my Raspberry Pi Zero W, with full "sudo apt-get upgrade" to 31Aug18:
* If I edit "/lib/systemd/system/bluealsa.service" and change the relevant line to "ExecStart=/usr/bin/bluealsa --profile=a2dp-sink"
* Then after reboot the "UUID: Audio Sink" appears again under "sudo bluetoothctl show" (and I can connect my mobile phone again as "Media Audio"!)  Yay!

Many thanks for this info - very much appreciated!! :-)

henry-shadowjet, 2018-09-03:
I tried your suggestion, Arkq and it worked. I finally dares to update bluealsa, and I had to change the /lib/systemd/system/bluealsa.service to activate all the profiles using the -p flags and it all works just like it used to. Thank you

borine, 2021-06-04:
According to the last 2 comments, this issue was resolved some time ago.

closed 2021-06-04 by borine