arkq/bluez-alsa #61: Select profile other than A2DP

YOUN3SS, 2017-09-10:
Hi,

How to select other profiles than A2DP?

I tried:
`defaults.bluealsa.profile "auto"`
and
`defaults.bluealsa.profile "hsp"`

In .asoundrc file, but it doesn't work, the output error is:
ALSA lib bluealsa-pcm.c:636:(_snd_pcm_bluealsa_open) Invalid BT profile [a2dp, sco]: auto
Segmentation fault

Arkq, 2017-09-10:
Hi,

The hint is in the error message :)
> Invalid BT profile [a2dp, sco]: auto

Available profiles are: `a2dp` and `sco`, so the configuration should be as follows:

    defaults.bluealsa.profile "sco"

However, SIGSEGV is my fault. Thanks for spotting it.

closed 2017-09-10 by Arkq

YOUN3SS, 2017-09-11:
Thanks for your prompt reply.

What profile(s) are you referring to with "sco" parameter?

Arkq, 2017-09-11:
> What profile(s) are you referring to with "sco" parameter?

SCO is not a profile per se, it is a transport type (BT protocol):
https://en.wikipedia.org/wiki/List_of_Bluetooth_protocols#Synchronous_Connection-Oriented_.28SCO.29

SCO is used by the HFP and HSP profiles. So if one select SCO for audio transmission, then HFP or HSP is used (of course if any of them is connected). I though, that there is no need to separate HFP and HSP profiles in the bluez-alsa configuration. This project is dedicated for audio transfer only. In the HFP/HSP profiles there is much more features, so in order to omit unintentional misleading, I've chosen SCO as an "audio profile" name.

If you need some more information regarding bluez-alsa usage, feel free to email me directly :)

YOUN3SS, 2017-09-11:
Ah, that is more clear for me now :) You're right to avoid the part of HSP/HFP commands that Bluez isn't supporting since v5.

How to get your email?

Arkq, 2017-09-11:
> How to get your email?

From git repo :D or from here: arkadiusz.bokowy at gmail.com

commit 36cc3017 (2017-09-10): Fix invalid argument handling

Fixes #61
commit dae9e755 (2017-09-10): Fix invalid argument handling

Fixes #61