arkq/bluez-alsa #63: Couldn't get BlueALSA transport: No such device

george-ai, 2017-09-17:
Hi.

Will appreciate your help here.

The installation of blue-alsa and aac was perfect.

I am using a jabber bt headset (speaker and mic - A2DP and HSP).

The speaker and playback is working 100%, however when I want to record as below, I am getting a few issues.

Can you provide any guidance here?

arecord -D bluealsa capture.wav
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:213: Getting transport for 50:1A:A5:2F:F3:47 type 1
ALSA lib ../../../src/asound/bluealsa-pcm.c:658:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
arecord: main:722: audio open error: No such device

**My .asoundrc file**

pcm.!default {
    type plug
    slave.pcm {
        type bluealsa
        interface "hci0"
        device "50:1A:A5:2F:F3:47"
        profile "a2dp"
    }
}
ctl.!default {
    type bluealsa
    interface "hci0"
}

george-ai, 2017-09-18:
Hi. 

After reading through the other issues, I have resolved this by using Profile=sco.

Awesome!

Arkq, 2017-09-19:
Hi,

I'm glad you've managed to resolve this issue by yourself. I was off for few last days.

There is no info in the readme file about the SCO profile, because it's not tested very well (not yet) and I'm still working on it. In the future it will be documented. Sorry for the inconvenience :/

I'm closing this issue as resolved.

closed 2017-09-19 by Arkq

xuewenxu, 2019-05-27:
> Hi.
> 
> After reading through the other issues, I have resolved this by using Profile=sco.
> 
> Awesome!

Hi, 
    @george-ai , you said you have resolved this by using Profile=sco.  Is it  like this "arecord -D bluealsa:HCI=hci0,DEV=1C:52:16:17:A8:F4,PROFILE=sco capture.wav" ?