arkq/bluez-alsa #65: connection reset by peer

ashah976, 2017-10-29:
Installed bluealsa with the stretch repo and seems to work fine, I'm only using the a2dp side.
However after a file the BT connection between my speaker drops.  Is there any way to keep the connection alive.  I'm trusted and paired with bluetoothctl all fine.
If I'm constantly streaming the connection stays alive no problem

I've set the below to 0 using the /etc/bluetooth/main.conf file
# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
PairableTimeout = 0

which as I suspected wont do anything as the default is already 0

I've tried pulseaudio which doesnt suffer any such issues the sink just stays alive although change was needed in the pulseaudio config files.....

Arkq, 2017-10-29:
Hi,

If you are using A2DP only, run bluealsa wih `--disable-hsp` and `--disable-hfp`. This will disable HFP connection negotiation, which might be the case here (my implementation is kind a naive, and when the other side does not respond "properly" connection is terminated).

It might be more informative if you manage to past logs from the disconnection event. Also please tell me, if you are connecting phone or a headset?

ashah976, 2017-10-29:
okay tried that here are the logs
This is a fresh install of stretch lite
Connection is of a normal Luna bluetooth speaker
like this one

https://www.amazon.co.uk/LunaBox-SoundBot-Bluetooth-Dual-Driver-Distortion-Rose-Gold/dp/B01N0NBXKM/ref=sr_1_1?ie=UTF8&qid=1509318301&sr=8-1&keywords=luna+bluetooth+speakers

```
pi@raspberrypi:~ $ ps -ef | grep blue
root       433     1  0 22:43 ?        00:00:00 /usr/lib/bluetooth/bluetoothd
root       565   529  0 22:47 pts/0    00:00:00 sudo bluealsa --disable-hsp --disable-hfp
root       569   565  0 22:47 pts/0    00:00:00 bluealsa --disable-hsp --disable-hfp
pi         628   529  0 22:53 pts/0    00:00:00 grep --color=auto blue

LunaBox SoundBot]# connect 42:15:BD:04:08:55
Attempting to connect to 42:15:BD:04:08:55
[LunaBox SoundBot]# bluealsa: ../../src/bluez.c:539: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: ../../src/bluez.c:539: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: ../../src/bluez.c:471: A2DP Source (SBC) configured for device 42:15:BD:04:08:55
bluealsa: ../../src/bluez.c:473: Configuration: channels: 2, sampling: 48000
bluealsa: ../../src/transport.c:616: State transition: 0 -> 0
bluealsa: ../../src/ctl.c:571: +-+-
bluealsa: ../../src/bluez.c:593: Endpoint already registered: /A2DP/MPEG24/Source/1
bluealsa: ../../src/bluez.c:593: Endpoint already registered: /A2DP/MPEG24/Sink/1
bluealsa: ../../src/bluez.c:610: Registering endpoint: /A2DP/SBC/Source/2
bluealsa: ../../src/bluez.c:593: Endpoint already registered: /A2DP/SBC/Sink/1
Connection successful
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -93
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -55
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -96
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -58
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -96
[CHG] Device 14:BB:6E:30:61:D1 RSSI: -55
[LunaBox SoundBot]# exit

The below turn up on the screen when the device disconnects:

pi@raspberrypi:~ $ bluealsa: ../../src/bluez.c:539: Endpoint method call: org.bluez.MediaEndpoint1.ClearConfiguration()
bluealsa: ../../src/transport.c:3
```

ashah976, 2017-11-05:
Managed some further tests - I have tried 2 other BT speakers utilising only a2dp and the same observations are noted.  The speaker disconnects after 3/4 mins of inactivity even if paired.  If audio is continuously played then the connection is kept alive.  The 3/4 mins of disconnection is not issued by bluealsa but by the peer.  It seems bluealsa perhaps doesn't keep the connection "alive" if not audio is being transmitted???

ashah7, 2017-12-18:
I've added this to 
/etc/dbus-1/system.d/bluetooth.conf:
<policy user="bluealsa"> <allow send_destination="org.bluez"/> </policy>

And since I've had no disconnects once paired....?
bluealsa is started as root but root should have access to bluez so why does this work?

Arkq, 2020-01-25:
Indeed bluealsa does not keep connection alive. Right now the HFP profile should work as expected, so you might try to use new bluealsa version and in order to keep connection alive, you might try to send some RFCOMM command (e.g. invalid command, just send something). Maybe that will keep BT connection alive.

closed 2020-01-25 by Arkq