arkq/bluez-alsa #66: bluez-alsa in mobile

yangte, 2017-10-31:
we use bluez-alsa in mobile ,but we find that music has problem with Caton when we use mgmt start discovery.can you give me some suggestions?

Arkq, 2017-11-30:
Hi,

I'm sorry, but I'm not able to help you I'm afraid. I was not aware, that it is possible to run bluez-alsa on Android (I though, that Android does not use Bluez as a Bluetooth stack) :D.

All I can say, is that the discovery mode (if you mean Bluetooth devices discovery), does not play very well with data transfer (reported in other issues).

yangte, 2017-11-30:
yes,you are right.most android mobile use bluedroid as bluetooth,but we do a system os ,so we choose blueZ as bluetooh stack.now we use bluealsa solve the blueZ compatibility with autdioflinger。we find open transport need to set broadcom QOS high,and stop transport set broadcom QOS low.this operation may solve when  discovery device audio can not play very well.

yangte, 2017-12-01:
but now we find use audioflinger to play music will have Last frame sound when pause and resume. can you give me some suggestions?Thanks again and let me know if you need some help!

Arkq, 2021-01-16:
Closing stale issue.

closed 2021-01-16 by Arkq