arkq/bluez-alsa #67: End of audio being truncated

spl237, 2017-11-01:
When playing back an audio WAV file using aplay, the last ~0.5 seconds of the file are truncated. (This is running on a Raspberry Pi under Raspbian Stretch.) It appears that playback is cut off while there are still samples in a playback FIFO.

As a very kludgey bodge, I've inserted a hard delay of 400ms or so into the function bluealsa_drain in bluealsa-pcm.c; this fixes the problem but also prevents aplay from closing cleanly. I don't know if the solution is to implement some sort of test inside the drain function to ensure that the playback FIFO is empty, as at present this simply returns without checking that the FIFO is empty. I also note that the bluealsa_delay function is never called - I'm wondering if an alternative solution is to set an appropriate delay for the playback pipeline.

Either way, something isn't right somewhere here - this has been reported by several Pi users and I have confirmed it locally.

Arkq, 2017-11-01:
Hi,

Thanks for reporting it. I'll try to make a proper drain callback (right now it is not implemented at all). I've postponed this task, because there were more important (IMHO) issues/features to be implemented. However, I understand, that for someone who is playing short audio files, this issue is important.

> I also note that the bluealsa_delay function is never called

The bluealsa_delay is called if implementer calls it - aplay is a simple playback tool, so there is no need for synchronization. The delay function is used mostly by audio-video applications.

Arkq, 2017-11-02:
I've just created some prototype for the drain callback (waiting for all data being transferred to bluez), however it does not resolve this issue completely. There are still few (~200) milliseconds of audio being truncated (compared to ~400ms without drain callback). I'm just wondering if such a functionality (playing out all the audio) works with PulseAudio or any other Bluetooth audio system (without relying on a delay function)?

Right now, I'll be off for a week and a half, so this issue will have to wait until then.

spl237, 2017-11-03:
I've just checked with a PulseAudio implementation, and there is no truncation at all. 

I have created a test file with a brief constant tone followed by 5 0.1 second beeps separated by 0.1 second silences - by playing it and counting the beeps, I can tell how much audio is being dropped. With your original implementation, I only heard 2 of the 5 beeps, so 500-600ms was being truncated. With a sleep added to the drain function, I heard 4 of the 5 beeps, so this was down to 100-200ms being truncated. But using the same test file, the same Bluetooth speaker and aplay with the Jessie version of Raspbian, which uses PulseAudio and Bluez 5.23, all 5 beeps are audible. Raspbian Stretch uses Bluez 5.43 - it's possible there is a difference in latency etc in the different versions of the Bluez stack, but it's hard for me to test PulseAudio with 5.43.

Arkq, 2017-11-03:
Many tanks for this test!

So, if it IS possible to make it work, I'll try to figure it out. Maybe
a little bit of poking around the PulseAudio code...

spl237, 2017-11-03:
Good luck! Let me know if there's anything else I can do to help in the way of testing.

closed 2017-11-15 by Arkq

Arkq, 2017-11-15:
I've just created some "prosthesis" for this issue. Please, check if the result is satisfactory.

In order to resolve this issue completely, some extensive refactoring is required. I've checked how PulseAudio has managed to overcome this issue (Bluetooth specification lacks drain functionality), and it turns out, that PulseAudio releases Bluetooth audio transport few seconds after playback is finished. I could do something similar (Android does it as well), however it will not be easy... As a workaround, I've added [200ms delay](https://github.com/Arkq/bluez-alsa/blob/99d9d72a5aed22dd755485a2661e209ded697079/src/transport.c#L766).

spl237, 2017-11-15:
Many thanks - I'll give it a try in the next day or so and report back.

spl237, 2017-11-16:
With my test file, which has five 100ms tones separated by 100ms of silence, I only got the first two tones before applying this patch. With the patch, I get all of the the first four tones, and a very slightly truncated final tone. It's not 100% perfect - I'd guess it is dropping the final 50ms or so - but it is significantly better than it was before, and I think very few people would notice a problem with real audio rather than test tones. Good job - many thanks!

Arkq, 2017-11-16:
Great,

Could you increase this 200ms delay to 500ms and see if the end of file is also being truncated? If it helps, I'm afraid that it has to be as it is now - asynchronous transport release will be a solution for this issue, however like I said, it's a lot of work. On the other hand, if this increased delay does not help, I'll try to make it better. The problem might be, that the last 50ms is not encoded properly by the encoder - encoder requires at lest X samples.

spl237, 2017-11-16:
Just tried increasing the delay to 500ms - I can't hear any difference from a delay of 200ms; it does sound as if the final beep is still being slightly truncated.

Arkq, 2017-11-16:
OK, many thanks for testing!

federovsky, 2017-11-23:
Hi,
I had the same issue and the recent update solves it. Thanks!
Unfortunately, I have the first fractions of the audio cut off, meaning, the beginning of the audio is not heard. If the wav is short, like "beep" (less than 1 seconds), it can not be heard at all.

Any ideas?

Many thanks for this project!
Ofer

Arkq, 2017-11-30:
Hi @federovsky 

I've experienced similar issue like you (beginning of the file being truncated), however it was caused by my Bluetooth headset - I've dumped bluetooth data (using `tcpdump -i bluetooth0`) and every PCM frame was there. This happens a lot during transport acquisition for the first time after Bluetooth device connection. To be honest, I don't know how to "fix" this problem without breaking something else - e.g. I could send 1s of silence before sending actual audio, however this will produce unnecessary 1s lag... Anyway, I'm open to any suggestions :).

federovsky, 2017-11-30:
Hi @Arkq,

I tried it with Logitech x100 speaker.
I will try with other speakers/headsets.

Thanks!
Ofer

NGkaltzidis, 2024-04-10:
@arkq Has this been fixed for you ?
I'm experiencing the same issues with alsa (cut off the beginning of the playback file.

arkq, 2024-04-10:
@NGkaltzidis How much of the audio is truncated (in ms) ?

Also, this issue was for truncation of the end of the audio, and I think that this case if more or less resolved already. If your case is only for the beginning of the audio being truncated, please submit a new issues.

commit 99d9d72a (2017-11-15): Wait for internal buffers to be drained

When ALSA client requests PCM drain, wait until our internal buffer
is consumed. Note, that samples might still be dropped (not played by
the Bluetooth device), because there is not possible to drain BlueZ
internal buffers (there is no API for that). One possible solution is
to releasing A2DP transport asynchronously with some arbitrary delay.

Fixes #67