arkq/bluez-alsa #69: Bluetooth speaker with mic

samsunix, 2017-12-07:
Hi and sorry!

I am new with alsa and bluez-alsa. Trying to make google assistance work on my pi3.

Speaker works fine with youtube etc but when launching google assistance I get alsa-related error;

`(env) pi@raspberrypi:~/voice-recognizer-raspi $ python3 src/main.py
[1488:1513:ERROR:audio_input_processor.cc(755)] Input error
/home/pi/voice-recognizer-raspi/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
Say "OK, Google" then speak, or press Ctrl+C to quit...
[1488:1516:ERROR:audio_input_processor.cc(755)] Input error
Segmentation fault
`
my asoundrc:

`cm.JBLClip2 {
        type plug
        slave.pcm {
                type bluealsa
                device "04:FE:A1:AD:30:89"
                profile "a2dp"
        }
}
pcm.JBLClip2-sco {
        type plug
        slave.pcm {
                type bluealsa
                device "04:FE:A1:AD:30:89"
                profile "sco"
        }
}
`

Please assist. This is frustrating.
Thank you in advance. 

- Sam

Arkq, 2017-12-08:
Hi,

It would be nice to see the cause of this nasty SEGFAULT. You will have to run this command via gdb, and then backtrace the stack, e.g.:

    # gdb --args python3 src/main.py
    > run
    (wait for SEGFAULT)
    > bc

If in the trace there is ALSA library, you might try to compile bluez-alsa with debug enabled (maybe there will be some additional output). Then post the stack here with the exact version of ALSA library and other related components (e.g. library with this `audio_input_processor.cc` file).

Your asoundrc file seems to be correct, so the cause of this crash has to be somewhere else.

samsunix, 2017-12-08:
Hi and thank you for reply.

I got this:

`(gdb) run
Starting program: /home/pi/env/bin/python3 src/main.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
bTraceback (most recent call last):
  File "src/main.py", line 33, in <module>
    import aiy.voicehat
  File "/home/pi/voice-recognizer-raspi/src/aiy/voicehat.py", line 17, in <module>
c    import aiy._drivers._button
  File "/home/pi/voice-recognizer-raspi/src/aiy/_drivers/_button.py", line 18, in <module>
    import RPi.GPIO as GPIO
ImportError: No module named 'RPi'
[Inferior 1 (process 2834) exited with code 01]
`

After fixing that little issue I got another one:

```
> Starting program: /home/pi/voice-recognizer-raspi/env/bin/python3 src/main.py
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
> [New Thread 0x74ff9470 (LWP 1658)]
> [New Thread 0x745ff470 (LWP 1659)]
> [New Thread 0x73bff470 (LWP 1660)]
> [New Thread 0x733ff470 (LWP 1704)]
> [New Thread 0x729ff470 (LWP 1705)]
> [New Thread 0x721ff470 (LWP 1706)]
> [New Thread 0x717ff470 (LWP 1707)]
> [New Thread 0x70fff470 (LWP 1708)]
> [New Thread 0x70496470 (LWP 1709)]
> [Thread 0x70496470 (LWP 1709) exited]
> [New Thread 0x6f92d470 (LWP 1710)]
> [Thread 0x6f92d470 (LWP 1710) exited]
> [New Thread 0x70496470 (LWP 1711)]
> [1644:1711:ERROR:audio_input_processor.cc(755)] Input error
> [Thread 0x70496470 (LWP 1711) exited]
> /home/pi/voice-recognizer-raspi/src/aiy/_drivers/_led.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
>   GPIO.setup(channel, GPIO.OUT)
> [New Thread 0x70496470 (LWP 1712)]
> [New Thread 0x6f92d470 (LWP 1713)]
> Say "OK, Google" then speak, or press Ctrl+C to quit...
> [Thread 0x73bff470 (LWP 1660) exited]
> [New Thread 0x73bff470 (LWP 1714)]
> [1644:1714:ERROR:audio_input_processor.cc(755)] Input error
> [Thread 0x73bff470 (LWP 1714) exited]
> [Thread 0x70fff470 (LWP 1708) exited]
> [Thread 0x717ff470 (LWP 1707) exited]
> [Thread 0x721ff470 (LWP 1706) exited]
> [Thread 0x729ff470 (LWP 1705) exited]
> [Thread 0x733ff470 (LWP 1704) exited]
> 
> Thread 2 "Assistant.Contr" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x74ff9470 (LWP 1658)]
> 0x7529d200 in ?? ()
>    from /home/pi/voice-recognizer-raspi/env/lib/python3.5/site-packages/google/assistant/library/libassistant_embedder.so
> 
```

samsunix, 2017-12-08:
https://www.raspberrypi.org/blog/another-update-raspbian/

> There is one issue with the support for Bluetooth audio, however. Due to the way the Bluetooth stack has been written, Bluetooth devices do not appear to the system as standard ALSA audio devices

Arkq, 2017-12-08:
Actually the article you've posted is from the last year and is about PulseAudio :) Anyway, quoted phrase is valid for BlueALSA as well. There is no possibility (technically there is, however it will require to write a kernel module) to add Bluetooth PCMs to the list available via the kernel ALSA API. All exposed PCMs are software PCMs (plug-ins) visible via (properly configured) ALSA library.

The stack trace is... strange. I have NO idea what might be wrong. Unfortunately I'm not able to reproduce it (I do not have Raspbery). It might or might not be related to to bluez-alsa, it's hard to tell. Did you compile bluez-alsa with `--enable-debug` option? Because I don't see any logs from the PCM plug-in, so maybe this issue is not caused by bluez-alsa after all.

> Funny thing on AlsaMixer... in and output is clearly visible (and the damn battery).

These PCM controls are from bluez-alsa. SCO is available, so it should be possible to capture audio. Please test this using arecord or any recording program. Also test SCO playback.

Also, what do you mean by "and the damn battery"? :D This was my idea for exposing battery level of connected Bluetooth device (if battery level is available somehow). Prior to this post, I've had no feedback about this "feature". If you think, that it is silly to show battery level alongside volume controls (which kinda is IMHO :D), maybe I'll add some option to disable it (or maybe an option to enable it, and by default it will be disabled). What do you think?

samsunix, 2017-12-09:
Don't get me wrong, Info 'bout battery is great, don't know would I be able to pull it out any other way. Does it belong to mixer, well... maybe not but would be great to get that data anyway, maybe have it on task-bar like laptop battery healt...  Anyhow during compile I had no errors with alsa / bluez / bluez-alsa (well some missing packages which is fixed now). 

If you want to really get this thing working with raspberry I can loan and even bring it to you since we live in the same city (siema!). optionally I can grant you VNC access over Internet if you want have a look. I get now GDBus.error.org.bluez.Error.Failed: Protocol not avaivable when trying to connect my device. 

My next step most likely would be wipe it and start over.

send me email: samuli.latvala@gmail.com for ip / un / pw

/Sammy

Arkq, 2017-12-10:
I'm closing this issue, as it has turned out that the SIGSEGV is in the Google Assistant library due to missing default capturing PCM (it seems like omitted error checking, aka slopy programming :D).

closed 2017-12-10 by Arkq