arkq/bluez-alsa #71: Using bluetooth speaker outside of aplay?

fancywriter, 2017-12-10:
Hi guys.
It's more like a question, than a real issue, however, I think it could worth adding to the README.md file.

I have a bluetooth speaker (JBL Clip 2) and want to use it, you know, exactly as a speaker. I was surprised that bluez can't do it out of the box, but OK. I found this package **bluez-alsa**, which is awesome and really works. I am stucked after last step where it's said that I can run `aplay -D bluealsa file.wav` It works great, however, I don't want to use `aplay`, I want change speaker device to the bluetooth one instead of default one in laptop for all programs using sound, including smplayer (with mpv backend) and firefox (for youtube videos). And when bluetooth is off (or when speaker is off) I want to fallback from laptop speakers again. How to achieve this?

Arkq, 2017-12-11:
Hi!

I don't know if such a feature is provided by ALSA configuration. It is however possible to implement it, e.g. [PulseAudio PCM plug-in](http://git.alsa-project.org/?p=alsa-plugins.git;a=blob;f=pulse/pcm_pulse.c;h=283174357e8b2aa55076fd784c97aebb8b1b2bde;hb=HEAD#l1160) implements such a feature.

Anyway, I'm not sure if implementing it in the bluez-alsa is a good idea. It will be bluez-alsa-specific feature with very restrictive usage. IMHO, more generic solution should be found/developed. What if someone wants to make fallback as follows:

    BT-speaker -> USB-speaker -> build-in-card

USB speaker PCM is accessed via `hw` plug-in, which lacks `fallback` feature.

fancywriter, 2017-12-11:
I agree with you, it doesn't matter really what sound device I connect to my laptop, bluetooth, USB, fire-wire or with astral - I need some generic solution. Perhaps it's already somewhere, but no one know, no one use it...

Arkq, 2018-08-01:
Since this issue is somehow generic (not strictly bound to bluez-alsa project), I'm closing it. However, a solution to this issue will be greatly appreciated :)

closed 2018-08-01 by Arkq

vinnyA3, 2018-08-05:
Hi guys.  I was looking for a native solution to this problem as well; unfortunately, I did not find one.  What I'm doing instead is using a script to swap *asound* config files whenever **bluealsa** is running:  I have *.asoundrc*--which contains the hardware sound card config--and *.asoundrc.bak*--which contains the bluetooth speaker config (I only have one bluetooth device that I connect to really).

I am using [dmenu](https://tools.suckless.org/dmenu/) to run the scripts that manage bluealsa and swap default *.asound* config files.  Here is a [gist](https://gist.github.com/vinnyA3/cdc11b747af58f9e9810e46290e99d9a) that contains the small scripts I'm using to make this work. 

This general solution is actually working quite well for me.  I hope this helps someone :+1: