arkq/bluez-alsa #72: Couldn't get BlueALSA transport: No such device

graham8, 2017-12-11:
I get this just trying to aplay -D bluealsa foo.wav. I think I've followed the instructions - the only tweak was to change the group permissions on /run/bluealsa.

ran bluealsa as root
.asoundrc as per the README
20-bluealsa.conf is there

How to troubleshoot, please?

baruchsiach, 2017-12-14:
I'm experiencing the same issue.

```
ALSA lib bluealsa-pcm.c:659:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:788: audio open error: No such device
```

`bluealsa` under `strace -fx` show these lines

```
[pid 28763] recvfrom(10, "\x04\x00\x00\x00\x5b\xdd\xdc\x26\xc6\x88\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00", 24, MSG_DONTWAIT, NULL, NULL) = 24
[pid 28763] sendto(10, "\x02", 1, MSG_NOSIGNAL, NULL, 0) = 1
```

when running this `aplay` command:

```
aplay -D bluealsa:HCI=hci0,DEV=88:C6:26:DC:DD:5B,PROFILE=a2dp click.wav
```

`bluetoothctl` indicates that the sound device is connected:

```
Device 88:C6:26:DC:DD:5B
	Name: Jaybird X3
	Alias: Jaybird X3
	Class: 0x00240418
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
 	Modalias: bluetooth:v01DAp0003d0122
```

I am on current Debian testing with the following libraries:

libbluetooth3: 5.47
GLib: 2.54.1
sbc: 1.3
alsa (libasound2): 1.1.3

Arkq, 2017-12-14:
Does `amixer -D bluealsa` show anything?

baruchsiach, 2017-12-14:
Yes:

```
Simple mixer control 'Jaybird X3 - A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [on]
  Front Right: Playback 127 [100%] [on]
Simple mixer control 'Jaybird X3 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 15 [100%] [on] Capture 15 [100%] [on]
Simple mixer control 'Jaybird X3 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 0 [0%]
```

Following that command `aplay` magically started to work. But then I stopped and started `bluealsa`, and now `amixer` reports something else:

```
Simple mixer control 'Jaybird X3 - SCO',0
  Capabilities: pvolume pvolume-joined cvolume cvolume-joined pswitch pswitch-joined cswitch cswitch-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: Playback 0 - 15 Capture 0 - 15
  Mono: Playback 15 [100%] [on] Capture 15 [100%] [on]
Simple mixer control 'Jaybird X3 | Battery',0
  Capabilities: pvolume pvolume-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 0 [0%]
```

And `aplay` fails as before.

I should mention that pulseaudio is also running on my system.

Arkq, 2017-12-14:
> I should mention that pulseaudio is also running on my system.

See paragraph 1: https://github.com/Arkq/bluez-alsa#troubleshooting
The explanation how Bluez works is not 100% correct in this paragraph (I was writing it when my knowledge wasn't so extensive as it is now), however using PulseAudio and BlueALSA at the same time isn't the best idea there. In short, if you want to use BlueALSA instead of PulseAudio, you have to unload Bluetooth-realted modules.

@graham8, if you are also running PulseAudio and BluezALSA together, then `aplay -D bluealsa` won't work as well.

### Explanation with some more details

In order to connect Bluetooth device using A2DP profile, one has to register A2DP handler in the Bluetooth stack (Bluez). This handler is used for incoming connection. Once the connection is established, handler is used and no more A2DP connections is allowed. In order to connect more devices (using A2DP), one has to register more handlers. PulseAudio registers **one** A2DP handler for Sink and **one** for Source (I'm not 100% sure, so someone might confirm this). If PulseAudio service starts before BlueALSA, then the handler from the PulseAudio is used (if it matches codec settings) - handlers are used in the registration order. Next device though is connected with the BluezALSA (next free handler). When, the handler is used, BluezALSA registers one more handler, so one can connect third device, then again one more handler is registered, and so on... So, if one wants to use BlueALSA, PulseAudio (and any other application which registers A2DP profile in Bluez) has to be properly configured - Bluetooth support has to be disabled.

baruchsiach, 2017-12-14:
Thanks. Killing pulseaudio and restarting bluetoothd fixed this issue for me. I still can't hear the sound, but this is most likely a different problem.

graham8, 2017-12-14:
No, I'm not running pulse although libpulse is there for some reason.  After a bit of digging, I added this to /etc/dbus-1/system.d/bluetooth.conf:
<policy user="bluealsa">
  <allow send_destination="org.bluez"/>
</policy>
Now, aplay seems to start up but I get no sound:

osmc@vero:~$ aplay -D bluealsa:DEV=FC:A8:9A:B3:42:4F TestFiles/blind1full.WAV
../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../shared/ctl-client.c:213: Getting transport for FC:A8:9A:B3:42:4F type 1bluealsa: ctl.c:554: New client accepted: 10

bluealsa: ctl.c:575: +-+-
bluealsa: ctl.c:575: +-+-
bluealsa-pcm.c:527: Setting constraints
Playing WAVE 'TestFiles/blind1full.WAV' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
bluealsa-pcm.c:272: Initializing HW
../shared/ctl-client.c:302: Requesting PCM open for FC:A8:9A:B3:42:4F
bluealsa: ctl.c:326: PCM requested for FC:A8:9A:B3:42:4F type 1 stream 0
bluealsa: transport.c:812: New transport: 11 (MTU: R:672 W:672)
bluealsa: ctl.c:575: +-+-
../shared/ctl-client.c:321: Opening PCM FIFO (mode: WR): /var/run/bluealsa/hci0-FC:A8:9A:B3:42:4F-1-0
bluealsa: bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: transport.c:668: State transition: 0 -> 2
bluealsa: io.c:55: Opening FIFO for reading: /var/run/bluealsa/hci0-FC:A8:9A:B3:42:4F-1-0
bluealsa: transport.c:104: Created new IO thread: A2DP Source (SBC)
bluealsa: io.c:409: Starting IO loop: A2DP Source (SBC)
bluealsa-pcm.c:295: FIFO buffer size: 4096
bluealsa-pcm.c:301: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:340: Prepared
bluealsa-pcm.c:324: Initializing SW
bluealsa-pcm.c:205: Starting
../shared/ctl-client.c:390: Requesting PCM resume for FC:A8:9A:B3:42:4F
bluealsa: transport.c:668: State transition: 2 -> 2
bluealsa: ctl.c:575: +-+-
bluealsa-pcm.c:103: Starting IO loop

BT is now disconnected, and I can't re-connect.

Arkq, 2017-12-15:
> Segmentation fault sudo bluealsa

Yikes.... It would be very nice to see some backtrace...

> bluealsa: Couldn't initialize controller thread: Address already in use

You have to remove all nodes from the `/run/bluealsa/` directory.

graham8, 2017-12-15:
Sorry, I'll need instructions on how to do a backtrace.

graham8, 2017-12-31:
Back to it.  I've re-compiled/installed bluez-alsa and now I can't get past this message:

osmc@vero:~/TestFiles$ amixer -D bluealsa
amixer: Mixer bluealsa load error: Cannot allocate memory

While on the bluealsa thread, I get:

bluealsa: ctl.c:554: New client accepted: 13
bluealsa: ctl.c:575: +-+-
bluealsa: ctl.c:575: +-+-
bluealsa: ctl.c:527: Invalid request length: 4 != 20
bluealsa: ctl.c:575: +-+-

Then trying to play a file:

osmc@vero:~/TestFiles$ aplay -D bluealsa:DEV=FC:A8:9A:B3:42:4F,PROFILE=a2dp,HCI=hci0 blind1full.WAV
Playing WAVE 'blind1full.WAV' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (92879 92880)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 5
BUFFER_TIME: (464399 464400)
BUFFER_SIZE: 20480
BUFFER_BYTES: 81920
TICK_TIME: 0

with a similar error message from bluealsa about request length.

Arkq, 2019-11-25:
Closing, if the problem still occurs, please reopen with more info.

closed 2019-11-25 by Arkq

hboetes, 2020-05-26:
What fixed the problem for me was: `sudo apt autoremove --purge pulseaudio-module-bluetooth`

which removes the module described in https://github.com/Arkq/bluez-alsa/issues/72#issuecomment-351769985