arkq/bluez-alsa #73: Couldn't initialize controller thread: Address already in use

SuperSam56, 2017-12-29:
Hi @Arkq,
I just installed a fresh new install of Raspbian Stretch, on which i would like to stream Audio from my Android Phone via Bluetooth A2DP.
By the way on https://www.raspberrypi.org/blog/raspbian-stretch/ paragraph "Bluetooth audio": they say "...For Stretch, we are using the bluez-alsa package to make Bluetooth audio work with ALSA itself" 
So i though there is nothing more to do after having paired my bluetooth with my raspberry pi (which seems to be done succesfully)
Unfortunately when i launch the bluealsa command i have only a error message return "Couldn't initialize controller thread: Address already in use"
And when I try to restart bluealsa after pkill bluealsa, it never launches.
Could you please help me understand and resolve this issue ?
thanks

mill1000, 2017-12-29:
When installed on Raspbian Stretch bluez-alsa automatically runs on boot via systemd. Manually executing the bluealsa command is not necessary.

If the Android device recognizes the Raspberry Pi as an audio device when pairing then bluez-alsa is running correctly.

To play audio from the phone, on the Raspberry Pi you should use the bluealsa-aplay command as documented.

SuperSam56, 2017-12-29:
Indeed this is working with either bluealsa-aplay or arecord -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp commands (XX:XX:XX:XX:XX:XX is my bluetooth android phone address).
But this works only when i disable Wifi, otherwise i have sound jumps / jitters effects.
So is there a reason for this particular behavior of sound streaming when Wifi is on? Is there a fix for this?
Furthermore i have a final question: I want to use darkice package to catch a2dp bluetooth sound in a .mp3" mounted file. How should i configure darkice.cfg in its [input] section.

Because when i set the following parameter in darkice.cfg file:
device          = bluealsa:HCI=hci0,DEV=AD:EF:9E:3B:54:EA,profile=a2dp

and when i launch Darkice, i get the following error message:

DarkIce 1.3 live audio streamer, http://code.google.com/p/darkice/
Copyright (c) 2000-2007, Tyrell Hungary, http://tyrell.hu/
Copyright (c) 2008-2013, Akos Maroy and Rafael Diniz
This is free software, and you are welcome to redistribute it 
under the terms of The GNU General Public License version 3 or
any later version.
Using config file: /home/pi/darkice.cfg
Using ALSA DSP input device: bluealsa:HCI=hci0,DEV=AD:EF:9E:3B:54:EA,profile=a2dp
Using POSIX real-time scheduling, priority 4
ALSA lib conf.c:4858:(parse_args) Unknown parameter profile
ALSA lib conf.c:4991:(snd_config_expand) Parse arguments error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa:HCI=hci0,DEV=AD:EF:9E:3B:54:EA,profile=a2dp
DarkIce: DarkIce.cpp:1273: can't open connector [0]

Many thanks for your help

mill1000, 2017-12-30:
Is this on a Pi3? There is a [known issue](https://github.com/raspberrypi/linux/issues/1402) when operating the on-board WiFi and Bluetooth together.

I opt'd for a cheap USB Bluetooth dongle to resolve the problem.

SuperSam56, 2017-12-30:
Yes this is a Pi3B. Do you know if this known issue is also affecting the Pi Zero W which is also both bluetooth and Wifi ?
And could you also please give me some indications on where i could find help to configure darkice to work with bluealsa a2dp ? Because i only find tutorial to make it work with pulseaudio.
many thanks

closed 2018-01-02 by SuperSam56