arkq/bluez-alsa #75: Can't get aptX working on OSX

andrewbasterfield, 2018-01-02:
Hi

First off - great project, looking forwards to getting aptX working.

I have compiled with --enable-aptx however my high sierra macbook client (just because it is easy to see which codec is in use) is still reporting SBC as the codec in use, even after forcing aptX with the bluetooth explorer tool. I tested with my Oneplus 5T phone (tried both aptX and aptX-HD as the bluetooth codec) and got similar results to below. Are there some capability flags not being set? Do I need to set them myself in bluez config?

I would assume the macbook would pick up AAC codec (compiled in) even if apt-X was nonfunctional

```
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:490: Starting controller loop
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/MPEG24/Source/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/MPEG24/Sink/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/APTX/Source/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/SBC/Source/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/SBC/Sink/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/main.c:225: Starting main dispatching loop

/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/../src/shared/ctl-client.c:126: Subscribing for events: 11
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:551: New client accepted: 10
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/aplay.c:575: Starting main loop
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/aplay.c:597: Fetching available transports
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
2018-01-02 12:23:30 [9500] INFO: Registering ourselves (/org/abasterfield/bluez/agent) at /org/bluez [org.bluez.AgentManager1]
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:609: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:541: A2DP Sink (SBC) configured for device F4:0F:24:35:81:BB
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:543: Configuration: channels: 2, sampling: 44100
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/transport.c:669: State transition: 0 -> 0
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:566: Emitting notification: 1 => 10
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:663: Endpoint already registered: /A2DP/MPEG24/Source/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:663: Endpoint already registered: /A2DP/MPEG24/Sink/1
/usr/local/bluez-alsa/bin/bluealsa: /usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/src/bluez.c:663: Endpoint already registered: /A2DP/APTX/Source/1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/utils/aplay.c:597: Fetching available transports../../../src/bluez-alsa/src/bluez.c:663: Endpoint already registered: /A2DP/SBC/Source/1

/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:680: Registering endpoint: /A2DP/SBC/Sink/2
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/aplay.c:661: Creating PCM worker F4:0F:24:35:81:BB
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/../src/shared/ctl-client.c:303: Requesting PCM open for F4:0F:24:35:81:BB
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:551: New client accepted: 12
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:324: PCM requested for F4:0F:24:35:81:BB type 1 stream 1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
/usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/utils/../src/shared/ctl-client.c:322: Opening PCM FIFO (mode: RO): /var/run/bluealsa/hci0-F4:0F:24:35:81:BB-1-1
/usr/local/bluez-alsa/bin/bluealsa: /usr/local/bluez-alsa/bin/bluealsa-aplay: ../../../src/bluez-alsa/src/ctl.c:572: +-+-
../../../src/bluez-alsa/utils/aplay.c:328: Starting PCM loop
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/transport.c:669: State transition: 0 -> 1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/transport.c:813: New transport: 13 (MTU: R:672 W:65535)
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/transport.c:669: State transition: 1 -> 2
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/transport.c:105: Created new IO thread: A2DP Sink (SBC)
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/io.c:236: Starting IO loop: A2DP Sink (SBC)
/usr/local/bluez-alsa/bin/bluealsa: ../../../src/bluez-alsa/src/io.c:82: Opening FIFO for writing: /var/run/bluealsa/hci0-F4:0F:24:35:81:BB-1-1
```

Arkq, 2018-01-02:
Hi,

Now I can see, that your issue is partly my fault :) In the README, aptX appears only in the dependency list, without any further explanation what it does and how to use it.

Short explanation about the `--enable-aptx` option:

This option might be useful mostly for companies (not individual persons, however it is not 100% true), which would like to use bluez-alsa in their commercial products. This options enables integration with the Qualcomm aptX compatible encoding library only. In the future I might add support for decoder as well, however right now it is not available. Hence, with this option enabled, one might use bluez-alsa A2DP source profile (not sink) in order to stream music to the speaker/headset with the aptX capabilities.

However, when one enables `--enable-aptx`, bluez-alsa will require [openaptx](https://github.com/Arkq/openaptx) library. This library (for now) is a [library stub](https://github.com/Arkq/openaptx/blob/master/src/bt-aptx-stub.c#L28) only. It does not perform any encoding (in the future, when I finish my reverse-engineering work, it will perform encoding, however using it beyond research purpose will be illegal). Anyway, it might be used to test aptX integration (it contains aptX encoded test sound) and what's more, it implements Qualcomm libbt-aptx API. So if one owns (legally) libbt-aptx library is should be possible to use it instead of the stub one, which will "enable" aptX encoding in bluez-alsa.

I hope, this explanation is clear enough.

And now, back to your issue.

As you can see in the logs, there is a line `Registering endpoint: /A2DP/APTX/Source/1`, which states that that the aptX source profile has been registered, however there is no corresponding line for sink profile. So, receiving aptX encoded streams is not possible.

> I would assume the macbook would pick up AAC codec (compiled in) even if apt-X was nonfunctional

That's true. AAC will be selected (both source and sink is available) IF connected device supports it. I'm not sure that OSX supports AAC. AAC is supported by iOS (I think) and hi-end Android-based smartphones. In order to be check (it will work only if connection is established by BlueZ, which will ask for A2DP capabilities) if device supports particular codec, one can dump bluetooth trafic with `tcpdump -pn -i bluetooth0  -w /tmp/dump` and use [wireshark](https://www.wireshark.org/) to inspect collected logs.

andrewbasterfield, 2018-01-02:
Thanks for the clarification

commit 913a07bd (2018-01-14): Add warning about aptX support

Closes #75
closed 2018-01-02 by andrewbasterfield