arkq/bluez-alsa #77: Recording but not audio

clustor, 2018-01-06:
I am trying to get bluetooth souce to work with Raspberry Pi zero. After many attempts, I got the playing to work with A2DP with the following .asoundrc as provided in the README. 

defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000

For some reason, .asoundrc was getting overwritten messing up the stack. I disabled write permission so the overwrite doesn't happen anymore and now I am able to "list-cards", "list-sources", etc with pacmd.  

I am also able to paplay just fine.  The only challenge is with "parecord".  'parecord capture.wav' seems to record the content as the file size grows but there is no audio/sound in the file. I can't figure out if SCO transport is operational.  I bumped into multiple PulseAudio default.pa settings to enable remixing etc but they seem to have no effect. I would like to confirm that bluez-alsa is actually receiving the content from microphone. 

Any thoughts on how to troubleshoot this issue?

clustor, 2018-01-07:
More experiments..

I have disabled pulseaudio completely now and ran more experiments.

aplay -D bluealsa sample.wav doesn't work   # I am guessing this is expected
aplay -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp sample.wav # this works

# I changed the defaults.bluealsa.profile to "sco" ~/.asoundrc and with a2dp record was getting the following error. 

_ALSA lib bluealsa-pcm.c:659:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
arecord: main:788: audio open error: No such device
arecord_ 

#With SCO profile
arecord -D bluealsa  produces 44 byte wav file that doesn't change.

Hope I am not missing something.  It's not clear to me how the device knows that recording should commence.

karlkar, 2018-02-03:
I have the same issue...

Arkq, 2019-11-04:
Please test with current [master](https://github.com/Arkq/bluez-alsa/tree/55c8aec1296f0756de8ae6ac334961308e9b119d), because HFP/HSP bluealsa internal support has been extensively improved. Also, see other issues related to RPi. If the problem still occurs please reopen.

closed 2019-11-04 by Arkq